coolsnowwolf/lede #11784: 添加了qemu-ga 后， 编译失败

keivenliao, 2024-01-15:
### 详细叙述

默认的config，只是添加了 qemu-ga

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

../contrib/vhost-user-blk/vhost-user-blk.c:535:5: error: unknown type name 'off64_t'; did you mean 'off_t'?
  535 |     off64_t capacity;
      |     ^~~~~~~
      |     off_t
../contrib/vhost-user-blk/vhost-user-blk.c:537:16: warning: implicit declaration of function 'lseek64'; did you mean 'lseek'? [-Wimplicit-function-declaration]
  537 |     capacity = lseek64(fd, 0, SEEK_END);
      |                ^~~~~~~
      |                lseek
../contrib/vhost-user-blk/vhost-user-blk.c:537:16: warning: nested extern declaration of 'lseek64' [-Wnested-externs]
ninja: build stopped: subcommand failed.
make[5]: *** [Makefile:165: run-ninja] Error 1

closed 2024-01-15 by coolsnowwolf

WYC-2020, 2024-01-15:
又是这个问题，修的想吐 
[image]

GreatMichaelLee, 2024-01-15:
一样的问题

FAILED: contrib/vhost-user-blk/vhost-user-blk.p/vhost-user-blk.c.o
x86_64-openwrt-linux-musl-gcc -m64 -mcx16 -Icontrib/vhost-user-blk/vhost-user-blk.p -Icontrib/vhost-user-blk -I../contrib/vhost-user-blk -Isubprojects/libvhost-user -I../subprojects/libvhost-user -I. -Iqapi -Itrace -Iui/shader -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/include/glib-2.0 -I/home/builder/lede_x86/staging_dir/target-x86_64_musl/usr/lib/glib-2.0/include -flto=auto -fdiagnostics-color=auto -Wall -Winvalid-pch -std=gnu11 -O2 -isystem /home/builder/lede_x86/build_dir/target-x86_64_musl/qemu-7.2.1/linux-headers -isystem linux-headers -iquote . -iquote /home/builder/lede_x86/build_dir/target-x86_64_musl/qemu-7.2.1 -iquote /home/builder/lede_x86/build_dir/target-x86_64_musl/qemu-7.2.1/include -iquote /home/builder/lede_x86/build_dir/target-x86_64_musl/qemu-7.2.1/tcg/i386 -pthread -D_GNU_SOURCE -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -Wstrict-prototypes -Wredundant-decls -Wundef -Wwrite-strings -Wmissing-prototypes -fno-strict-aliasing -fno-common -fwrapv -Wold-style-declaration -Wold-style-definition -Wtype-limits -Wformat-security -Wformat-y2k -Winit-self -Wignored-qualifiers -Wempty-body -Wnested-externs -Wendif-labels -Wexpansion-to-defined -Wimplicit-fallthrough=2 -Wno-missing-include-dirs -Wno-shift-negative-value -Wno-psabi -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/qemu-7.2.1=qemu-7.2.1 -Wformat -Werror=format-security -fstack-protector -Wl,-z,now -Wl,-z,relro -Os -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/qemu-7.2.1=qemu-7.2.1 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -fPIE -MD -MQ contrib/vhost-user-blk/vhost-user-blk.p/vhost-user-blk.c.o -MF contrib/vhost-user-blk/vhost-user-blk.p/vhost-user-blk.c.o.d -o contrib/vhost-user-blk/vhost-user-blk.p/vhost-user-blk.c.o -c ../contrib/vhost-user-blk/vhost-user-blk.c
cc1: note: someone does not honour COPTS correctly, passed 2 times
../contrib/vhost-user-blk/vhost-user-blk.c: In function 'vub_initialize_config':
../contrib/vhost-user-blk/vhost-user-blk.c:535:5: error: unknown type name 'off64_t'; did you mean 'off_t'?
  535 |     off64_t capacity;
      |     ^~~~~~~
      |     off_t
../contrib/vhost-user-blk/vhost-user-blk.c:537:16: warning: implicit declaration of function 'lseek64'; did you mean 'lseek'? [-Wimplicit-function-declaration]
  537 |     capacity = lseek64(fd, 0, SEEK_END);
      |                ^~~~~~~
      |                lseek
../contrib/vhost-user-blk/vhost-user-blk.c:537:16: warning: nested extern declaration of 'lseek64' [-Wnested-externs]
ninja: build stopped: subcommand failed.
make[5]: *** [Makefile:165: run-ninja] Error 1
make[5]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/qemu-7.2.1/build'
make[4]: *** [GNUmakefile:11: all] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/qemu-7.2.1'
make[3]: *** [Makefile:450: /home/builder/lede_x86/build_dir/target-x86_64_musl/qemu-7.2.1/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/utils/qemu'
time: package/feeds/packages/qemu/compile#1.06#0.31#1.74
    ERROR: package/feeds/packages/qemu failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/qemu/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231: world] Error 2

GreatMichaelLee, 2024-01-15:
> 又是这个问题，修的想吐 [...]

还有，大佬，你那个nfs-kernel那个没有提交哇，我还在本地改的

WYC-2020, 2024-01-15:
> 还有，大佬，你那个nfs-kernel那个没有提交哇，我还在本地改的

自己去提我不喜欢提，我只喜欢修

GreatMichaelLee, 2024-01-15:
> 自己去提我不喜欢提，我只喜欢修

大佬提一下萨，我们都是白手党.....