coolsnowwolf/lede #11785: x86 内核编译遇vmlinux报错 Unsupported DW_TAG_unspecified_type(0x3b)

MilesPoupart, 2024-01-15:
### 详细叙述

make target/compile 时报错
full log: https://github.com/MilesPoupart/OpenWRT_x86_x64/actions/runs/7529239913/job/20493589987

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

```
2024-01-15T13:44:35.4733138Z make[2]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/scripts/config'
2024-01-15T13:44:35.4734114Z make[2]: 'conf' is up to date.
2024-01-15T13:44:35.4735833Z make[2]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/scripts/config'
2024-01-15T13:44:37.0469823Z make[1]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt'
2024-01-15T13:44:39.3825795Z make[2]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux'
2024-01-15T13:44:39.3960843Z make[3]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux/x86'
2024-01-15T13:44:39.4602560Z rm -f /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/localversion
2024-01-15T13:44:39.4622921Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/scripts/kconfig.pl  + + /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux/generic/config-6.1 /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux/x86/config-6.1 /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/target/linux/x86/64/config-6.1 > /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.target
2024-01-15T13:44:39.5138776Z awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/.config >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.target
2024-01-15T13:44:39.5201671Z echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.target
2024-01-15T13:44:39.5215491Z echo "# CONFIG_KALLSYMS_ALL is not set" >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.target
2024-01-15T13:44:39.5229035Z echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.target
2024-01-15T13:44:39.5244067Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/scripts/package-metadata.pl kconfig /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/tmp/.packageinfo /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/.config 6.1 > /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.override
2024-01-15T13:44:39.6281967Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/scripts/kconfig.pl 'm+' '+' /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.target /dev/null /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.override > /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.set
2024-01-15T13:44:39.6849754Z mv /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.set /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.old
2024-01-15T13:44:39.6872171Z grep -v INITRAMFS /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.old > /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.set
2024-01-15T13:44:39.6902489Z echo 'CONFIG_INITRAMFS_SOURCE=""' >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.set
2024-01-15T13:44:39.6916236Z echo '# CONFIG_INITRAMFS_FORCE is not set' >> /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.set
2024-01-15T13:44:39.6929628Z rm -rf /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/modules
2024-01-15T13:44:39.6950845Z cmp -s /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.set /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.prev || { cp /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.set /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config; cp /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.set /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.prev; }
2024-01-15T13:44:39.6975048Z export MAKEFLAGS= ; [ -d /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/user_headers ] || make -C /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72 KCFLAGS="-fmacro-prefix-map=/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Jan 15 09:40:44 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.1.72 CC="x86_64-openwrt-linux-musl-gcc"  INSTALL_HDR_PATH=/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/user_headers headers_install
2024-01-15T13:44:39.6984229Z grep '=[ym]' /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.config.set | LC_ALL=C sort | /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/host/bin/mkhash md5 > /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.vermagic
2024-01-15T13:44:39.7015480Z touch /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/.configured
2024-01-15T13:44:39.7034285Z rm -f /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/vmlinux /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72/System.map
2024-01-15T13:44:39.7057825Z make -C /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.72 KCFLAGS="-fmacro-prefix-map=/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon Jan 15 09:40:44 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.1.72 CC="x86_64-openwrt-linux-musl-gcc" olddefconfig
[... cut ...]
2024-01-15T13:45:17.1338830Z time: target/linux/compile#33.13#4.84#37.75
2024-01-15T13:45:17.1340029Z     ERROR: target/linux failed to build.
2024-01-15T13:45:17.1344132Z make[1]: *** [target/Makefile:30: target/linux/compile] Error 1
2024-01-15T13:45:17.1350764Z make[1]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt'
2024-01-15T13:45:17.1360469Z make: *** [/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/include/toplevel.mk:231：target/compile] 错误 2
```

coolsnowwolf, 2024-01-15:
.config 不对，重新生成一次

MilesPoupart, 2024-01-15:
> .config 不对，重新生成一次

居然是本尊回复（激动一下

俺是在自托管的设备上云编译的 第一次编译之后报这个错 rm-rf了整个_work目录后重新编译的

两次都是这个问题，还问下怎么排查，谢谢

.config文件内容是[config.txt](https://github.com/coolsnowwolf/lede/files/13939879/config.txt)

MilesPoupart, 2024-01-16:
切换dwarves到1.25版本之后编译dwarves报错```'EM_AARCH64' undeclared```，似乎是```elf.h```中没有定义？
```
2024-01-16T13:34:07.9132065Z make[2]: Entering directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/tools/dwarves'
2024-01-16T13:34:07.9698019Z MAKEFLAGS="" /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/host/bin/ninja  -j1 -C /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25 
2024-01-16T13:34:07.9720516Z ninja: Entering directory `/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25'
2024-01-16T13:34:08.4089402Z [1/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/bpf.c.o
2024-01-16T13:34:08.4903115Z [2/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/bpf_prog_linfo.c.o
2024-01-16T13:34:09.4805843Z [3/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/btf.c.o
2024-01-16T13:34:10.0195975Z [4/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/btf_dump.c.o
2024-01-16T13:34:10.3534987Z [5/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/gen_loader.c.o
2024-01-16T13:34:10.4339942Z [6/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/hashmap.c.o
2024-01-16T13:34:13.3153359Z [7/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/libbpf.c.o
2024-01-16T13:34:13.3678731Z [8/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/libbpf_errno.c.o
2024-01-16T13:34:13.4685373Z [9/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/libbpf_probes.c.o
2024-01-16T13:34:14.1059386Z [10/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/linker.c.o
2024-01-16T13:34:14.3745581Z [11/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/netlink.c.o
2024-01-16T13:34:14.4454341Z [12/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/nlattr.c.o
2024-01-16T13:34:14.8026184Z [13/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/relo_core.c.o
2024-01-16T13:34:14.9563642Z [14/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/ringbuf.c.o
2024-01-16T13:34:15.0330033Z [15/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/strset.c.o
2024-01-16T13:34:15.2867539Z [16/55] Building C object CMakeFiles/bpf.dir/lib/bpf/src/usdt.c.o
2024-01-16T13:34:15.9377003Z [17/55] Building C object CMakeFiles/dwarves.dir/dwarves.c.o
2024-01-16T13:34:16.5141298Z [18/55] Building C object CMakeFiles/dwarves.dir/dwarves_fprintf.c.o
2024-01-16T13:34:16.5788986Z [19/55] Building C object CMakeFiles/dwarves.dir/gobuffer.c.o
2024-01-16T13:34:16.7709272Z [20/55] Building C object CMakeFiles/dwarves.dir/ctf_loader.c.o
2024-01-16T13:34:16.9169000Z [21/55] Building C object CMakeFiles/dwarves.dir/libctf.c.o
2024-01-16T13:34:17.3422736Z [22/55] Building C object CMakeFiles/dwarves.dir/btf_encoder.c.o
2024-01-16T13:34:17.5200211Z [23/55] Building C object CMakeFiles/dwarves.dir/btf_loader.c.o
2024-01-16T13:34:17.6064078Z [24/55] Building C object CMakeFiles/dwarves.dir/dwarf_loader.c.o
2024-01-16T13:34:17.6066790Z make[2]: *** [Makefile:36: /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25/.built] Error 1
2024-01-16T13:34:17.6067855Z FAILED: CMakeFiles/dwarves.dir/dwarf_loader.c.o 
2024-01-16T13:34:17.6073899Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/host/bin/gcc -DDWARVES_MAJOR_VERSION=1 -DDWARVES_MINOR_VERSION=25 -D_GNU_SOURCE -Ddwarves_EXPORTS -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25 -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25/lib/include -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25/lib/bpf/include/uapi -O2 -I/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/staging_dir/host/include -pthread -D_LARGEFILE64_SOURCE -D_FILE_OFFSET_BITS=64 -Wall -O2 -fPIC -MD -MT CMakeFiles/dwarves.dir/dwarf_loader.c.o -MF CMakeFiles/dwarves.dir/dwarf_loader.c.o.d -o CMakeFiles/dwarves.dir/dwarf_loader.c.o -c /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25/dwarf_loader.c
2024-01-16T13:34:17.6080664Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25/dwarf_loader.c: In function 'arch__nr_register_params':
2024-01-16T13:34:17.6082864Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25/dwarf_loader.c:1013:14: error: 'EM_AARCH64' undeclared (first use in this function); did you mean 'EM_IA_64'?
2024-01-16T13:34:17.6084228Z  1013 |         case EM_AARCH64:
2024-01-16T13:34:17.6084653Z       |              ^~~~~~~~~~
2024-01-16T13:34:17.6085079Z       |              EM_IA_64
2024-01-16T13:34:17.6086975Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25/dwarf_loader.c:1013:14: note: each undeclared identifier is reported only once for each function it appears in
2024-01-16T13:34:17.6088535Z make[1]: *** [tools/Makefile:221: tools/dwarves/compile] Error 1
2024-01-16T13:34:17.6089925Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25/dwarf_loader.c: In function 'arch__set_register_params':
2024-01-16T13:34:17.6092084Z /home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/build_dir/host/dwarves-1.25/dwarf_loader.c:1055:14: error: 'EM_AARCH64' undeclared (first use in this function); did you mean 'EM_IA_64'?
2024-01-16T13:34:17.6093424Z  1055 |         case EM_AARCH64:
[... cut ...]
2024-01-16T13:34:17.6095849Z make[2]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/tools/dwarves'
2024-01-16T13:34:17.6096714Z time: tools/dwarves/compile#9.42#0.27#9.70
2024-01-16T13:34:17.6097233Z     ERROR: tools/dwarves failed to build.
2024-01-16T13:34:17.6098103Z make[1]: Leaving directory '/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt'
2024-01-16T13:34:17.6099511Z make: *** [/home/miles/x86p2/_work/OpenWRT_x86_x64/OpenWRT_x86_x64/openwrt/include/toplevel.mk:231：tools/compile] 错误 2
2024-01-16T13:34:17.6133715Z ##[error]Process completed with exit code 2.
```

MilesPoupart, 2024-01-16:
参考 [friendlyarm/friendlywrt/c98e09f](https://github.com/friendlyarm/friendlywrt/commit/c98e09f01bc6788644d22ffd02cd7aaaef63f5ea)
俺再改下试试...

PR #11793 (2024-01-16): dwarves: bump to 1.25
fix: #11785
closed 2024-01-16 by coolsnowwolf