coolsnowwolf/lede #11787: 6.6内核编译不成功

HawaiiGit, 2024-01-16:
### 详细叙述

[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Ubuntu 22.04.3 LTS

### 详细日志

ERROR: module '/home/hpc/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.10/drivers/gpu/drm/drm_exec.ko' is missing.
make[3]: *** [modules/video.mk:306: /home/hpc/lede/bin/targets/x86/64/packages/kmod-drm_6.6.10-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/hpc/lede/package/kernel/linux'
time: package/kernel/linux/compile#11.98#6.53#15.93
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/hpc/lede'
make[1]: *** [package/Makefile:110: /home/hpc/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/hpc/lede'
make: *** [/home/hpc/lede/include/toplevel.mk:231: world] Error 2

nebulite, 2024-01-16:
same issue:
`make[2]: *** [toolchain/Makefile:97: toolchain/musl/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
x86_64-openwrt-linux-musl-gcc: unknown assembler invoked
scripts/Kconfig.include:50: Sorry, this assembler is not supported.
make[5]: *** [scripts/kconfig/Makefile:77: oldconfig] Error 1
make[4]: *** [Makefile:704: oldconfig] Error 2
make[4]: Leaving directory '/home/devops/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/linux-6.1.72'
make[3]: *** [Makefile:114: /home/devops/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/linux-6.1.72/.configured] Error 2
make[3]: Leaving directory '/home/devops/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#0.26#0.13#0.38
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/home/devops/lede'
make[1]: *** [toolchain/Makefile:93: /home/devops/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/devops/lede'
make: *** [/home/devops/lede/include/toplevel.mk:231: world] Error 2
`

WYC-2020, 2024-01-16:
没自修的能力别去搞6.6，l还没修完的，慌什么

coolsnowwolf, 2024-01-16:
并没有说已经支持 x86 可以编译 6.6 了， 那是给龙芯的

nebulite, 2024-01-17:
并没有特意去修改.config, 
git pull && ./scripts/feeds update -a && ./scripts/feeds install -a
之后make编译报错，之前就是只编译x86
有什么需要特别修改的配置么

WYC-2020, 2024-01-17:
> 并没有特意去修改.config, git pull && ./scripts/feeds update -a && ./scripts/feeds install -a 之后make编译报错，之前就是只编译x86 有什么需要特别修改的配置么

这楼的作者是针对的6.6 你这个6.1就别再这里混起了 要么新开一个issue 并且日志请帖单线程，你这样帖神仙都算不出来，L也说了x86是目前不支持6.6的 有能力的 自己把6.1config复制一份成6.6 然后几个modle需要扩展到6.6 就可以编程x86 6.6内核的，建议是等L他们搞，自己动手能力不行的 就别去折腾6.6

closed 2024-01-25 by coolsnowwolf