coolsnowwolf/lede #11788: branch  20230609   6.1.32 编译失败

caixin1218, 2024-01-16:
### 详细叙述

2024-01-16编译失败，8小时前还可以成功。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

make[4]: Leaving directory '/home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1'
touch /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/.built
mkdir -p /home/caixin/router/6.1.32/bin/targets/x86/64/packages /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-x86_64/node/CONTROL /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo
install -d -m0755 /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-x86_64/node/usr/bin
install -m0755 /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-install/usr/bin/node /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-x86_64/node/usr/bin/
find /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-x86_64/node -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="/home/caixin/router/6.1.32/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/caixin/router/6.1.32/scripts/strip-kmod.sh" PATCHELF="/home/caixin/router/6.1.32/staging_dir/host/bin/patchelf" /home/caixin/router/6.1.32/scripts/rstrip.sh /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-x86_64/node
rstrip.sh: /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-x86_64/node/usr/bin/node: executable
(cd /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-x86_64/node/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/caixin/router/6.1.32/bin/packages/x86_64/packages
/home/caixin/router/6.1.32/staging_dir/host/bin/fakeroot /home/caixin/router/6.1.32/staging_dir/host/bin/bash /home/caixin/router/6.1.32/scripts/ipkg-build -m "" /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-x86_64/node /home/caixin/router/6.1.32/bin/packages/x86_64/packages
Packaged contents of /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-x86_64/node into /home/caixin/router/6.1.32/bin/packages/x86_64/packages/node_v16.19.1-1_x86_64.ipk
rm -rf /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/.pkgdir/node.installed /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/.pkgdir/node
mkdir -p /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/.pkgdir/node
install -d -m0755 /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/.pkgdir/node/usr/bin
install -m0755 /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-install/usr/bin/node /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/.pkgdir/node/usr/bin/
touch /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/.pkgdir/node.installed
rm -rf /home/caixin/router/6.1.32/tmp/stage-node
mkdir -p /home/caixin/router/6.1.32/tmp/stage-node/host /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/packages
install -d -m0755 /home/caixin/router/6.1.32/tmp/stage-node/usr/include
cp -fpR /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/ipkg-install/usr/include/* /home/caixin/router/6.1.32/tmp/stage-node/usr/include/
find /home/caixin/router/6.1.32/tmp/stage-node -name '*.la' | xargs -r rm -f; 
if [ -f /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/packages/node.list ]; then /home/caixin/router/6.1.32/scripts/clean-package.sh "/home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/packages/node.list" "/home/caixin/router/6.1.32/staging_dir/target-x86_64_musl"; fi
if [ -d /home/caixin/router/6.1.32/tmp/stage-node ]; then (cd /home/caixin/router/6.1.32/tmp/stage-node; find ./ > /home/caixin/router/6.1.32/tmp/stage-node.files); 	SHELL= flock /home/caixin/router/6.1.32/tmp/.staging-dir.flock -c ' mv /home/caixin/router/6.1.32/tmp/stage-node.files /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/packages/node.list && cp -fpR /home/caixin/router/6.1.32/tmp/stage-node/* /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/caixin/router/6.1.32/tmp/stage-node
touch /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/stamp/.node_installed
mkdir -p /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/caixin/router/6.1.32/tmp/.root-copy.flock -c 'cp -fpR /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/node-v16.19.1/.pkgdir/node/. /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/'
touch /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp/.node_installed
echo "node" >> /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/node.default.install
make[3]: Leaving directory '/home/caixin/router/6.1.32/feeds/packages/lang/node'
time: package/feeds/packages/node/compile#72963.89#4243.09#2360.70
make[2]: Leaving directory '/home/caixin/router/6.1.32'
make[1]: *** [package/Makefile:110: /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/caixin/router/6.1.32'
make: *** [/home/caixin/router/6.1.32/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2024-01-16:
日志都不对没法算，单线程日志

caixin1218, 2024-01-16:
> 日志都不对没法算，单线程日志
对不起，这是单线程输出的。
time: package/libs/ncurses/compile#1.67#0.06#0.28
make[3]: Entering directory '/home/caixin/router/6.1.32/package/utils/util-linux'
mkdir -p /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp
echo '1' | cmp -s - /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/libuuid.version || { echo '1' > /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/libuuid.version;  } 
SHELL= flock /home/caixin/router/6.1.32/tmp/.root-copy.flock -c 'cp -fpR /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/util-linux-2.38.1/.pkgdir/libuuid/. /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/'
touch /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp/.libuuid_installed
echo "libuuid" >> /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/util-linux.default.install
mkdir -p /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp
echo '1' | cmp -s - /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/libblkid.version || { echo '1' > /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/libblkid.version;  } 
SHELL= flock /home/caixin/router/6.1.32/tmp/.root-copy.flock -c 'cp -fpR /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/util-linux-2.38.1/.pkgdir/libblkid/. /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/'
touch /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp/.libblkid_installed
echo "libblkid" >> /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/util-linux.default.install
mkdir -p /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp
echo '1' | cmp -s - /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/libfdisk.version || { echo '1' > /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/libfdisk.version;  } 
SHELL= flock /home/caixin/router/6.1.32/tmp/.root-copy.flock -c 'cp -fpR /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/util-linux-2.38.1/.pkgdir/libfdisk/. /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/'
touch /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp/.libfdisk_installed
echo "libfdisk" >> /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/util-linux.default.install
mkdir -p /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp
echo '1' | cmp -s - /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/libmount.version || { echo '1' > /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/libmount.version;  } 
SHELL= flock /home/caixin/router/6.1.32/tmp/.root-copy.flock -c 'cp -fpR /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/util-linux-2.38.1/.pkgdir/libmount/. /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/'
touch /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp/.libmount_installed
echo "libmount" >> /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/util-linux.default.install
mkdir -p /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp
echo '1' | cmp -s - /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/libsmartcols.version || { echo '1' > /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/libsmartcols.version;  } 
SHELL= flock /home/caixin/router/6.1.32/tmp/.root-copy.flock -c 'cp -fpR /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/util-linux-2.38.1/.pkgdir/libsmartcols/. /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/'
touch /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp/.libsmartcols_installed
echo "libsmartcols" >> /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/util-linux.default.install
mkdir -p /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/caixin/router/6.1.32/tmp/.root-copy.flock -c 'cp -fpR /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/util-linux-2.38.1/.pkgdir/blkid/. /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/'
touch /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp/.blkid_installed
echo "blkid" >> /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/util-linux.default.install
mkdir -p /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/caixin/router/6.1.32/tmp/.root-copy.flock -c 'cp -fpR /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/util-linux-2.38.1/.pkgdir/cfdisk/. /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/'
touch /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp/.cfdisk_installed
echo "cfdisk" >> /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/util-linux.default.install
mkdir -p /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/caixin/router/6.1.32/tmp/.root-copy.flock -c 'cp -fpR /home/caixin/router/6.1.32/build_dir/target-x86_64_musl/util-linux-2.38.1/.pkgdir/fdisk/. /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/'
touch /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp/.fdisk_installed
echo "fdisk" >> /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/pkginfo/util-linux.default.install
mkdir -p /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/root-x86/stamp
[... cut ...]
time: package/feeds/Alist/alist/compile#89.15#42.95#3.30
    ERROR: package/feeds/Alist/alist failed to build.
make[2]: *** [package/Makefile:116: package/feeds/Alist/alist/compile] Error 1
make[2]: Leaving directory '/home/caixin/router/6.1.32'
make[1]: *** [package/Makefile:110: /home/caixin/router/6.1.32/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/caixin/router/6.1.32'
make: *** [/home/caixin/router/6.1.32/include/toplevel.mk:231：world] 错误 2

WYC-2020, 2024-01-16:
[image]

WYC-2020, 2024-01-16:
遇到问题 先搜索嘛 https://github.com/coolsnowwolf/lede/issues/11093 关键字 就是这个fuse.h头文件

caixin1218, 2024-01-16:
> 遇到问题 先搜索嘛 #11093 关键字 就是这个fuse.h头文件

感谢大神

WYC-2020, 2024-01-16:
> > 遇到问题 先搜索嘛 #11093 关键字 就是这个fuse.h头文件
> 
> 感谢大神

先看下管用不嘛，我是把alist这个改成的依赖，好像通过安装也是可以的，你试试再说

caixin1218, 2024-01-16:
> > > 遇到问题 先搜索嘛 #11093 关键字 就是这个fuse.h头文件
> > 
> > 
> > 感谢大神
> 
> 先看下管用不嘛，我是把alist这个改成的依赖，好像通过安装也是可以的，你试试再说

root@Deepin-PC:/home/caixin# cd router/6.1.32
root@Deepin-PC:/home/caixin/router/6.1.32# sudo apt update -y
rf haveged help2man intltool libc6-dev-i386 libelf-dev libfuse-dev libglib2.0-dev libgmp3-dev \
libltdl-dev libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libpython3-dev libreadline-dev \
libssl-dev libtool lrzsz mkisofs msmtp ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 \
python3-pyelftools python3-setuptools qemu-utils rsync scons squashfs-tools subversion swig texinfo \
0% [正在连接 community-packages.deepin.com] [正在连接 proposed-packages.deepin.com] [正在连接 pro-driver-packages.uniontech.com]                                                                             
获取:1 https://app-store-files.uniontech.com/240116165658849/appstore deepin InRelease [6,131 B]                                            
获取:2 https://app-store-files.uniontech.com/240116165658849/appstore deepin/appstore amd64 Packages [5,845 kB]                                                                
获取:3 https://app-store-files.uniontech.com/240116165658849/appstore deepin/appstore i386 Packages [739 kB]                                                                            
命中:4 https://proposed-packages.deepin.com/dde-apricot unstable InRelease                                                                                                                    
命中:5 https://community-packages.deepin.com/deepin apricot InRelease                                                                       
命中:6 https://community-packages.deepin.com/driver driver InRelease
命中:7 https://community-packages.deepin.com/printer eagle InRelease
命中:8 https://pro-driver-packages.uniontech.com eagle InRelease
已下载 6,590 kB，耗时 3秒 (2,260 kB/s)
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
所有软件包均为最新。
root@Deepin-PC:/home/caixin/router/6.1.32# sudo apt full-upgrade -y
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
正在计算更新... 完成
升级了 0 个软件包，新安装了 0 个软件包，要卸载 0 个软件包，有 0 个软件包未被升级。
root@Deepin-PC:/home/caixin/router/6.1.32# sudo apt install -y ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential \
> bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext gcc-multilib g++-multilib \
> git gperf haveged help2man intltool libc6-dev-i386 libelf-dev libfuse-dev libglib2.0-dev libgmp3-dev \
> libltdl-dev libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libpython3-dev libreadline-dev \
> libssl-dev libtool lrzsz mkisofs msmtp ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 \
> python3-pyelftools python3-setuptools qemu-utils rsync scons squashfs-tools subversion swig texinfo \
> uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
没有可用的软件包 mkisofs，但是它被其它的软件包引用了。
这可能意味着这个缺失的软件包可能已被废弃，
或者只能在其他发布源中找到
然而下列软件包会取代它：
  genisoimage:i386 genisoimage

E: 软件包 mkisofs 没有可安装候选

应该不行，这个依赖都已经全部安装过的。
我再试试 更换 golang 1.20.x 版本

WYC-2020, 2024-01-16:
单独安装 一整个复制有可能会漏掉的 ，肯定能解决这个玩意的 这个和go应该关系不大

caixin1218, 2024-01-16:
> 单独安装 一整个复制有可能会漏掉的 ，肯定能解决这个玩意的 这个和go应该关系不大

Signing package index...
make[2]: Leaving directory '/home/caixin/router/6.1.32'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/home/caixin/router/6.1.32'
WORK_DIR=/home/caixin/router/6.1.32/build_dir/target-x86_64_musl/json_info_files /home/caixin/router/6.1.32/scripts/json_overview_image_info.py /home/caixin/router/6.1.32/bin/targets/x86/64/profiles.json 
make[2]: Leaving directory '/home/caixin/router/6.1.32'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/caixin/router/6.1.32'
make[2]: Leaving directory '/home/caixin/router/6.1.32'
make[1]: Leaving directory '/home/caixin/router/6.1.32'
caixin@Deepin-PC:~/router/6.1.32$

ok
编译成功，但我不明白，为什么不能整个复制安装依赖，为什么会漏掉？

closed 2024-01-16 by caixin1218