coolsnowwolf/lede #11790: armv8报错，今天的源码

binge8, 2024-01-16:
### 详细叙述

actions编译新环境新config

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

armv8

### 详细日志

```bash
time: package/feeds/packages/wget/ssl/compile#38.57#6.82#46.92
make[3]: Entering directory '/home/runner/work/op/op/openwrt/feeds/packages/utils/xfsprogs'
rm -f /home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/xfsprogs-5.9.0/.built
touch /home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/xfsprogs-5.9.0/.built_check
CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/xfsprogs-5.9.0=xfsprogs-5.9.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DHAVE_MAP_SYNC  -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/xfsprogs-5.9.0=xfsprogs-5.9.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DHAVE_MAP_SYNC  -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include " LDFLAGS="-L/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/home/runner/work/op/op/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -znow -zrelro  " make  -C /home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/xfsprogs-5.9.0/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/xfsprogs-5.9.0=xfsprogs-5.9.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DHAVE_MAP_SYNC" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;
make[4]: Entering directory '/home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/xfsprogs-5.9.0'
   [HEADERS] include
   [HEADERS] libxfs
Building include
Building libfrog
Building libxfs
    [CC]     rdwr.lo
rdwr.c: In function 'libxfs_bcompare':
rdwr.c:338:12: warning: format '%lx' expects argument of type 'long unsigned int', but argument 3 has type 'pthread_t' {aka 'struct __pthread *'} [-Wformat=]
  338 |         "%lx: Badness in key lookup (length)\n"
      |          ~~^
      |            |
      |            long unsigned int
  339 |         "bp=(bno 0x%llx, len %u bytes) key=(bno 0x%llx, len %u bytes)\n",
  340 |                                 pthread_self(),
      |                                 ~~~~~~~~~~~~~~
      |                                 |
      |                                 pthread_t {aka struct __pthread *}
rdwr.c: At top level:
rdwr.c:725:40: error: unknown type name 'off64_t'; did you mean 'off_t'?
  725 | __read_buf(int fd, void *buf, int len, off64_t offset, int flags)
      |                                        ^~~~~~~
      |                                        off_t
rdwr.c: In function 'libxfs_readbufr':
rdwr.c:753:17: warning: implicit declaration of function '__read_buf' [-Wimplicit-function-declaration]
  753 |         error = __read_buf(fd, bp->b_addr, bytes, LIBXFS_BBTOOFF64(blkno), flags);
      |                 ^~~~~~~~~~
rdwr.c: In function 'libxfs_readbufr_map':
rdwr.c:793:17: error: unknown type name 'off64_t'; did you mean 'off_t'?
  793 |                 off64_t offset = LIBXFS_BBTOOFF64(bp->b_maps[i].bm_bn);
      |                 ^~~~~~~
      |                 off_t
rdwr.c: At top level:
rdwr.c:962:41: error: unknown type name 'off64_t'; did you mean 'off_t'?
  962 | __write_buf(int fd, void *buf, int len, off64_t offset, int flags)
      |                                         ^~~~~~~
      |                                         off_t
rdwr.c: In function 'libxfs_bwrite':
rdwr.c:1015:31: warning: implicit declaration of function '__write_buf'; did you mean '__initbuf'? [-Wimplicit-function-declaration]
 1015 |                 bp->b_error = __write_buf(fd, bp->b_addr, bp->b_bcount,
      |                               ^~~~~~~~~~~
      |                               __initbuf
rdwr.c:1022:25: error: unknown type name 'off64_t'; did you mean 'off_t'?
 1022 |                         off64_t offset = LIBXFS_BBTOOFF64(bp->b_maps[i].bm_bn);
      |                         ^~~~~~~
      |                         off_t
gmake[6]: *** [../include/buildrules:75: rdwr.lo] Error 1
gmake[5]: *** [include/buildrules:36: libxfs] Error 2
make[4]: *** [Makefile:91: default] Error 2
make[4]: Leaving directory '/home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/xfsprogs-5.9.0'
make[3]: *** [Makefile:98: /home/runner/work/op/op/openwrt/build_dir/target-aarch64_generic_musl/xfsprogs-5.9.0/.built] Error 2
make[3]: Leaving directory '/home/runner/work/op/op/openwrt/feeds/packages/utils/xfsprogs'
time: package/feeds/packages/xfsprogs/compile#1.03#0.39#1.39
    ERROR: package/feeds/packages/xfsprogs failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/xfsprogs/compile] Error 1
make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/op/op/openwrt'
```

WYC-2020, 2024-01-16:
[image]
如此如此，这般这般，可以了记得把issue关闭了，别等l来给你关

binge8, 2024-01-16:
好的，感谢

closed 2024-01-16 by binge8