coolsnowwolf/lede #11797: 多线程编译老是报错，单线程编译就能complete

GreatMichaelLee, 2024-01-17:
### 详细叙述

不知道什么原因，以前从没遇到过，极少数偶尔有，重新编一次也就好了，但这次，把把都不过，单线程必过，谁能指导一下什么原因？太痛苦了。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

多线程编译日志的错误之处：从开始编译第一处error到停止前的最后一处摘录

```
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/net/mosdns'
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Installing libuv via submodule
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/libs/protobuf'
time: package/feeds/packages/protobuf/compile#0.51#4.41#5.55
make[3]: Entering directory '/home/builder/lede_x86/feeds/helloworld/xray-core'
error: pathspec 'external/libuv' did not match any file(s) known to git
fatal: not a git repository: /home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/external/libuv/../../.git/modules/external/libuv
-- Installing uvw via submodule
mkdir -p /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/nps-0.26.10/.pkgdir/npc/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
error: pathspec 'external/uvw' did not match any file(s) known to git
*
......
FAILED: src/CMakeFiles/lan-play.dir/proxy_direct.cpp.o
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-g++ -DLANPLAY_LINUX -DLANPLAY_LITTLE_ENDIAN -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/external/uvw/src -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/base/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/lwip/custom -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/lwip/src/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/uv_lwip/. -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/external/libuv/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3=switch-lan-play-0.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -DLANPLAY_VERSION=\"20230609-336-g5ed3a14e0\" -Wall -Wno-unused-function -std=gnu++17 -MD -MT src/CMakeFiles/lan-play.dir/proxy_direct.cpp.o -MF src/CMakeFiles/lan-play.dir/proxy_direct.cpp.o.d -o src/CMakeFiles/lan-play.dir/proxy_direct.cpp.o -c /home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/src/proxy_direct.cpp
x86_64-openwrt-linux-musl-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
......
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/net/vnstat2'
[78/84] /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-g++ -DLANPLAY_LINUX -DLANPLAY_LITTLE_ENDIAN -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/external/uvw/src -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/base/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/lwip/custom -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/lwip/src/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/uv_lwip/. -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/external/libuv/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3=switch-lan-play-0.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -DLANPLAY_VERSION=\"20230609-336-g5ed3a14e0\" -Wall -Wno-unused-function -std=gnu++17 -MD -MT src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o -MF src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o.d -o src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o -c /home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/src/proxy_socks5.cpp
FAILED: src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-g++ -DLANPLAY_LINUX -DLANPLAY_LITTLE_ENDIAN -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/external/uvw/src -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/base/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/lwip/custom -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/lwip/src/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/uv_lwip/. -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/external/libuv/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3=switch-lan-play-0.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -DLANPLAY_VERSION=\"20230609-336-g5ed3a14e0\" -Wall -Wno-unused-function -std=gnu++17 -MD -MT src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o -MF src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o.d -o src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o -c /home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/src/proxy_socks5.cpp
x86_64-openwrt-linux-musl-g++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
.....
make[3]: Entering directory '/home/builder/lede_x86/feeds/packages/utils/qemu'
[80/84] /home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-g++ -DLANPLAY_LINUX -DLANPLAY_LITTLE_ENDIAN -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/external/uvw/src -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/base/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/lwip/custom -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/lwip/src/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/uv_lwip/. -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/external/libuv/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3=switch-lan-play-0.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -DLANPLAY_VERSION=\"20230609-336-g5ed3a14e0\" -Wall -Wno-unused-function -std=gnu++17 -MD -MT src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o -MF src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o.d -o src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o -c /home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/src/proxy_socks5.cpp
FAILED: src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o
/home/builder/lede_x86/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-g++ -DLANPLAY_LINUX -DLANPLAY_LITTLE_ENDIAN -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/external/uvw/src -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/base/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/lwip/custom -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/lwip/src/include -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/uv_lwip/. -I/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/external/libuv/include -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3=switch-lan-play-0.2.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DNDEBUG -DLANPLAY_VERSION=\"20230609-336-g5ed3a14e0\" -Wall -Wno-unused-function -std=gnu++17 -MD -MT src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o -MF src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o.d -o src/CMakeFiles/lan-play.dir/proxy_socks5.cpp.o -c /home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/src/proxy_socks5.cpp
[... cut ...] 6_64_musl/root-x86/'
make[3]: *** [Makefile:61: /home/builder/lede_x86/build_dir/target-x86_64_musl/switch-lan-play-0.2.3/.built] Error 1
make[3]: Leaving directory '/home/builder/lede_x86/feeds/opentopd/switch-lan-play'
time: package/feeds/opentopd/switch-lan-play/compile#81.38#71.78#65.88
    ERROR: package/feeds/opentopd/switch-lan-play failed to build.
make[2]: *** [package/Makefile:116: package/feeds/opentopd/switch-lan-play/compile] Error 1
make[2]: *** Waiting for unfinished jobs....

```

WYC-2020, 2024-01-17:
自己不知道从那里加载一个switch-lan-play 什么玩意的包，没看到 这个包提醒报错么
[image]
要么自己网络不稳定 要么这个包有问题，我试过多线程编译这玩意包没任何问题，所以只能是你的网络不行，这种玩意你也来提issue 别人才不想背锅

aiamadeus, 2024-01-17:
`fatal error: Killed signal terminated program cc1plus` 很明显是内存不够

GreatMichaelLee, 2024-01-17:
> `fatal error: Killed signal terminated program cc1plus` 很明显是内存不够

还是A大佬妙手神医，增加了虚拟机从8G到16G内存（我的$nproc =16,可能实际也不需要8G但总有些tricky在里头，内存大包治百病）后好了, 吊打二楼的W大佬，嘿嘿

WYC-2020, 2024-01-17:
> > `fatal error: Killed signal terminated program cc1plus` 很明显是内存不够
> 
> 还是A大佬妙手神医，增加了虚拟机从8G到16G内存（我的$nproc =16,可能实际也不需要8G但总有些tricky在里头，内存大包治百病）后好了, 吊打二楼的W大佬，嘿嘿

大惊小怪，那后面你的问题一律不解答了，自己都混迹这个编译这么久了，还不知道什么配置么，我看l应该要给你写好机器配置你们才不会遇到这种奇葩的问题

GreatMichaelLee, 2024-01-17:
别啊…
大佬我错了…

------------------ 原始邮件 ------------------
发件人: W_Y_CPP ***@***.***&gt;
发送时间: 2024年1月17日 19:04
收件人: coolsnowwolf/lede ***@***.***&gt;
抄送: michael lee ***@***.***&gt;, Author ***@***.***&gt;
主题: Re: [coolsnowwolf/lede] 多线程编译老是报错，单线程编译就能complete (Issue #11797)

BoBppy, 2024-01-17:
> > `fatal error: Killed signal terminated program cc1plus` 很明显是内存不够
> 
> 吊打二楼的W大佬，嘿嘿

贴吧打过来了？？

closed 2024-01-17 by GreatMichaelLee