coolsnowwolf/lede #11800: aliyundrive-webdav报错

binge8, 2024-01-18:
### 详细叙述

昨天开始编译报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86，armv8

### 详细日志

```bash
time: package/feeds/packages/rust/host-compile#0.89#2.81#6.08
make[3]: Entering directory '/home/runner/work/op/op/openwrt/feeds/packages/multimedia/aliyundrive-webdav'
rm -rf /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/aliyundrive-webdav-2.3.2/.pkgdir/aliyundrive-webdav.installed /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/aliyundrive-webdav-2.3.2/.pkgdir/aliyundrive-webdav
mkdir -p /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/aliyundrive-webdav-2.3.2/.pkgdir/aliyundrive-webdav
install -d -m0755 /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/aliyundrive-webdav-2.3.2/.pkgdir/aliyundrive-webdav/usr/bin
install -m0755 /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/aliyundrive-webdav-2.3.2/target/x86_64-unknown-linux-musl/stripped/aliyundrive-webdav /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/aliyundrive-webdav-2.3.2/.pkgdir/aliyundrive-webdav/usr/bin/
install: cannot stat '/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/aliyundrive-webdav-2.3.2/target/x86_64-unknown-linux-musl/stripped/aliyundrive-webdav': No such file or directory
make[3]: *** [Makefile:57: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/aliyundrive-webdav-2.3.2/.pkgdir/aliyundrive-webdav.installed] Error 1
make[3]: Leaving directory '/home/runner/work/op/op/openwrt/feeds/packages/multimedia/aliyundrive-webdav'
time: package/feeds/packages/aliyundrive-webdav/compile#0.16#0.06#0.18
    ERROR: package/feeds/packages/aliyundrive-webdav failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/aliyundrive-webdav/compile] Error 1
make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/op/op/openwrt'
make: *** [/home/runner/work/op/op/openwrt/include/toplevel.mk:231: world] Error 2
```

hztbhtbh, 2024-01-18:
使用Actions云编译也是报错，不知道怎么修改，请各位大佬有时间看一下
[image]
[image]

WYC-2020, 2024-01-18:
这错误都是小问题，最大的问题是 rust这玩意编译 可能会遇到 类似这种错误 
failed to parse bitcode for LTO module: Invalid alignment value (Producer: 'LLVM17.0.6-rust-1.75.0-stable' Reader: 'LLVM 17.0.6-rust-1.75.0-stable')
然后这种可能就是gcc适配有问题，因为 rust好像是下载的预编译好的llvm，在用预编译的工具链来编译的时候 就会出现这个问题 gcc13 是肯定了，这个时候只能开启不用预编译的 自己编译多增加一个小时，其他gcc没试过，看你们都走到这里才出错 那就是你们rust这玩意没出错运气好.
这个aliyundrive-webdav错误 这样修复一下：
[image]
以前这个阿里云是直接下载现成的不需要编译的不知道什么原因改成编译的，可能现成的不行吧

WYC-2020, 2024-01-18:
话说alist不香吗 哈哈

WYC-2020, 2024-01-18:
failed to parse bitcode for LTO module: Invalid alignment value (Producer: 'LLVM17.0.6-rust-1.75.0-stable' Reader: 'LLVM 17.0.6-rust-1.75.0-stable'  这个问题也搞定了靠，就是本地g++ 版本太低了升级一下即可，在这里留一个记录吧，方便其他人也遇到类似问题

schen39, 2024-01-19:
> 说alist不香吗哈哈

最新代码alist启动不了，之前的可以

WYC-2020, 2024-01-19:
> > 说alist不香吗哈哈
> 
> 最新代码alist启动不了，之前的可以

我的可以启动

schen39, 2024-01-19:
> > > 说alist不香吗哈哈
> > 
> > 
> > 最新代码alist启动不了，之前的可以
> 
> 我的可以启动

3.3.0吗？最新代码？

WYC-2020, 2024-01-19:
> > > > 说alist不香吗哈哈
> > > 
> > > 
> > > 最新代码alist启动不了，之前的可以
> > 
> > 
> > 我的可以启动
> 
> 3.3.0吗？最新代码？

[image]
你说的这个版本吧，可以撒 
[image]

schen39, 2024-01-19:
> ![image](https://private-user-images.githubusercontent.com/34025388/297992948-345398e4-ecb9-4439-b0aa-4a470e3857bd.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

是的，为啥我的启动不了，晚上我再试试

hztbhtbh, 2024-01-19:
> 这错误都是小问题，最大的问题是 rust这玩意编译 可能会遇到 类似这种错误 failed to parse bitcode for LTO module: Invalid alignment value (Producer: 'LLVM17.0.6-rust-1.75.0-stable' Reader: 'LLVM 17.0.6-rust-1.75.0-stable') [...]

请问云编译怎么修改呢？

binge8, 2024-01-19:
sed -i 's/stripped/release/g' feeds/packages/multimedia/aliyundrive-webdav/Makefile

> 请问云编译怎么修改呢？

hztbhtbh, 2024-01-19:
> sed -i 's/stripped/release/g' feeds/packages/multimedia/aliyundrive-webdav/Makefile
> 
> > > 这错误都是小问题，最大的问题是 rust这玩意编译 可能会遇到 类似这种错误 failed to parse bitcode for LTO module: Invalid alignment value (Producer: 'LLVM17.0.6-rust-1.75.0-stable' Reader: 'LLVM 17.0.6-rust-1.75.0-stable') 然后这种可能就是gcc适配有问题，因为 rust好像是下载的预编译好的llvm，在用预编译的工具链来编译的时候 就会出现这个问题 gcc13 是肯定了，这个时候只能开启不用预编译的 自己编译多增加一个小时，其他gcc没试过，看你们都走到这里才出错 那就是你们rust这玩意没出错运气好. 这个aliyundrive-webdav错误 这样修复一下： ![image](https://private-user-images.githubusercontent.com/34025388/297692622-c1d0e07a-c08b-4314-995d-6f655949ff23.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDU2NzI4MDEsIm5iZiI6MTcwNTY3MjUwMSwicGF0aCI6Ii8zNDAyNTM4OC8yOTc2OTI2MjItYzFkMGUwN2EtYzA4Yi00MzE0LTk5NWQtNmY2NTU5NDlmZjIzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTE5VDEzNTUwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTlkOGUxMGQ2MjJjZWRmMmJjZDk3NjE5MGY3ODZiNGJhMmI5MmYzNjIyMjA3MzhiZjVjYmJiZWI4MzFjYjE0ODYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.jfN9wcd201l5Q0-Tv81ZujztkKb_Vz0zvPIYZnuxpEI) 以前这个阿里云是直接下载现成的不需要编译的不知道什么原因改成编译的，可能现成的不行吧
> > 
> > 
> > 请问用P3TERX/Actions-OpenWrt云编译怎么修改呢？之前都设置好了，能编译成功了，这几天就是编译aliyunwebdav就报错。

hztbhtbh, 2024-01-19:
> sed -i 's/stripped/release/g' feeds/packages/multimedia/aliyundrive-webdav/Makefile

hztbhtbh, 2024-01-20:
谢谢各位，用手机看的没有注意到，哈哈，我已经关闭重复的issues了

NeedforGit, 2024-02-04:
> sed -i 's/stripped/release/g' feeds/packages/multimedia/aliyundrive-webdav/Makefile

感谢！

zhangj3, 2024-02-05:
> failed to parse bitcode for LTO module: Invalid alignment value (Producer: 'LLVM17.0.6-rust-1.75.0-stable' Reader: 'LLVM 17.0.6-rust-1.75.0-stable' 这个问题也搞定了靠，就是本地g++ 版本太低了升级一下即可，在这里留一个记录吧，方便其他人也遇到类似问题

你好，请问下您是怎么修复的？ 我按照这个升级了g++http://pointborn.com/article/2023/6/12/2175.html 后还是报错

[image]

ddcatgg, 2024-02-09:
同问

ddcatgg, 2024-02-09:
> 你好，请问下您是怎么修复的？ 我按照这个升级了g++http://pointborn.com/article/2023/6/12/2175.html 后还是报错 [...]

同问

coolsnowwolf, 2024-02-22:
> 你好，请问下您是怎么修复的？ 我按照这个升级了g++http://pointborn.com/article/2023/6/12/2175.html 后还是报错 [...]

需要 g++ 9 以上，主系统

hztbhtbh, 2024-02-23:
我这边是虚拟机安装ubuntu18出现的问题，然后我直接换成ubuntu20或者更高版本就不存在这个问题了，可能ubuntu20里的g++也升级了吧

Zurisepinoy, 2024-08-10:
> 这错误都是小问题，最大的问题是 rust这玩意编译 可能会遇到 类似这种错误 failed to parse bitcode for LTO module: Invalid alignment value (Producer: 'LLVM17.0.6-rust-1.75.0-stable' Reader: 'LLVM 17.0.6-rust-1.75.0-stable') [...]

[image]

你好，请问我这个也是gcc适配的问题吗，但我试过了gcc 9 11 13,都失败了

closed 2024-10-16 by binge8