coolsnowwolf/lede #11803: 云编译丢失插件

binge8, 2024-01-19:
### 详细叙述

lede库，feeds没改，本地新制作上传的config，比如选了20个插件，编译完固件里只有10个左右，仔细看了日志，丢失的那些make download的时候就没下载。过程中也没报错，顺利编译完成。反倒我单独下载的别的库的插件一个也不丢。比如[luci-app-softethervpn。make download的时候日志李就没有，编译日志里也没有。确定config里选的Y

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

。。。

binge8, 2024-01-19:
附上编译前制作上传的config和编译后的config.buildinfo
[config.buildinfo.txt](https://github.com/coolsnowwolf/lede/files/13987240/config.buildinfo.txt)
[config.txt](https://github.com/coolsnowwolf/lede/files/13987242/config.txt)

binge8, 2024-01-19:
大哥大姐们，帮忙解解惑吧，问题到底出在哪里啊

WYC-2020, 2024-01-19:
这有啥好解答的，自己云编译没有吧config替换呗

binge8, 2024-01-19:
> 这有啥好解答的，自己云编译没有吧config替换呗

不应该，比如dockerman，默认不选的，但是编译出来有

WYC-2020, 2024-01-19:
说直白点就是你自己本地编译这些，改了什么，你云编译就得改什么，不说了看电视去了

binge8, 2024-01-19:
> 说直白点就是你自己本地编译这些，改了什么，你云编译就得改什么，不说了看电视去了

你的意思我明白，但是我丢的包都是源码里feed里自带的选上还丢，自己额外加的不丢。比如luci-app-softethervpn。所以太诡异了

WYC-2020, 2024-01-20:
仓库编译脚本仓库放出来，过去看日志，这个可能需要看下是否仓库脚本有问题了

binge8, 2024-01-20:
> 仓库编译脚本仓库放出来，过去看日志，这个可能需要看下是否仓库脚本有问题了

不好意思麻烦大哥了。好像真是config替换失败了，下午我改改代码再试一试

closed 2024-01-20 by binge8