coolsnowwolf/lede #11809: 新编译的OpenWrt 如何设置可以让局域网内 PPTP 连接穿透

yflwz, 2024-01-20:
### 详细叙述

最近自己编译了一个斐讯K3的固件，主要是有自己的需求，VPN服务编译了l2tp ipsec pptp，这些服务端都测试过可以正常使用，只是到最后发现网域网下面的手机和电脑都不能使用PPTP客户端连公司的PPTP服务器了。也看了官方的文档说要安装kmod-nf-nathelper-extra和kmod-ipt-raw等，我查了一下编译的时候本身已经有安装上，查找问题里发现同样问题的解决办法也试过加上，https://github.com/coolsnowwolf/lede/issues/329，但是还是用不了。
[image]
[image]
[image]
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

斐讯K3

### 详细日志

系统日志和内核日志没有相关信息

coolsnowwolf, 2024-01-26:
固件默认已支持

yflwz, 2024-01-30:
好的，已经解决，和固件无关，是因为走了openclash代理。

closed 2024-01-30 by yflwz