coolsnowwolf/lede #11813: OpenSSL缺少devcrypto.so

imEgo, 2024-01-21:
### 详细叙述

使用240115的代码，编译后缺少/usr/lib/engines-3/devcrypto.so，导致OpenSSL无法执行

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

FATAL: Startup failure (dev note: apps_startup()) for openssl
489B9CA2A97F0000:error:12800067:DSO support routines:dlfcn_load:could not load the shared library:crypto/dso/dso_dlfcn.c:118:filename(/usr/lib/engines-3/devcrypto.so): Error loading shared library /usr/lib/engines-3/devcrypto.so: No such file or director

WYC-2020, 2024-01-22:
这种 回退编译的 把所有的第三方 都的回退，因为 openssl这玩意 都不知道升级多少次了，他们各自之间的相互依赖 不通用的 这个只能全新编译才能搞定，并且你在安装别人的ipk的时候 也可能会遇到各种奇葩依赖的问题，因为有的依赖都是后面才加入的，或者有的依赖已经被后续的替换了，后续编译的ipk自然不能安装到以前的上面

imEgo, 2024-01-22:
> 这种 回退编译的 把所有的第三方 都的回退，因为 openssl这玩意 都不知道升级多少次了，他们各自之间的相互依赖 不通用的 这个只能全新编译才能搞定，并且你在安装别人的ipk的时候 也可能会遇到各种奇葩依赖的问题，因为有的依赖都是后面才加入的，或者有的依赖已经被后续的替换了，后续编译的ipk自然不能安装到以前的上面

openssl本体都不能执行，没有涉及到第三方吧

补充一下问题描述：
默认选项里openssl的硬件加速是选中的，生成的openssl.cnf会include devcrypto.conf，但是devcrypto.so没有生成，原版openwrt有libopenssl-devcrypto是1.1版本的，怀疑是openssl更新到3后没有更新对应的libopenssl-conf

手动解决办法：
编译时去掉openssl硬件加速选项，或删除固件里的```/etc/ssl/modules.cnf.d/devcrypto.conf```即可

WYC-2020, 2024-01-23:
> openssl本体都不能执行，没有涉及到第三方吧
>
> 补充一下问题描述： 默认选项里openssl的硬件加速是选中的，生成的openssl.cnf会include devcrypto.conf，但是devcrypto.so没有生成，原版openwrt有libopenssl-devcrypto是1.1版本的，怀疑是openssl更新到3后没有更新对应的libopenssl-conf [...]

那是因为受这个选项的影响，你吧这个关闭了就能单独选择devcrypto编译了
[image]
这个选项好像是吧功能内置到crypto了，没细研究过这玩意，反正我都是用这个库的默认值

imEgo, 2024-01-23:
> 那是因为受这个选项的影响，你吧这个关闭了就能单独选择devcrypto编译了 [...]

也有可能是openssl binary跟libopenssl不匹配的原因吧，因为ddns的aliyun插件会用到openssl binary命令来生成参数签名，默认编译选项会出现我上面的报错，但是其他用到ssl的模块都是正常的，我这个情况只能关闭engines support来解决，具体原因没有深入去看，还不清楚这个openssl binary是哪个依赖引入的

WYC-2020, 2024-01-24:
> ddns

那不清楚了，我没阿里云的域名 所以我随便写了一点假的数据测试了 没出现你这个问题，这个只能你自己排查了

imEgo, 2024-02-04:
由于LXC部署的原因，内核缺少依赖模块导致，如果是LXC使用，编译时需要将OpenSSL的```Acceleration support through /dev/crypto```的勾选去除，这样openssl及curl均可以正常使用
[image]

closed 2024-02-04 by imEgo

flyhigao, 2024-05-27:
多谢,lxc遇到这个问题,导致host,openssl都不可用,现在解决了