coolsnowwolf/lede #11819: 编译6.6内核失败

GreatMichaelLee, 2024-01-23:
### 详细叙述

如题，编译报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

SHELL= flock /home/builder/lede_x86/tmp/.root-copy.flock -c 'cp -fpR /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/.pkgdir/kernel/. /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/'
touch /home/builder/lede_x86/staging_dir/target-x86_64_musl/root-x86/stamp/.kernel_installed
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kernel into /home/builder/lede_x86/bin/targets/x86/64/packages/kernel_6.6.13-1-e288ba84816c63cc3929d1d4abd0559e_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-dax/lib/modules/6.6.13/dax.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-dax into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-dax_6.6.13-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13/crypto/crypto_hash.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-hash into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-hash_6.6.13-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13/crypto/crypto_null.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-null into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-null_6.6.13-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13/crypto/aead.ko' is built-in.
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-aead/lib/modules/6.6.13/geniv.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-aead into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-aead_6.6.13-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13/crypto/cryptomgr.ko' is built-in.
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-manager into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-manager_6.6.13-1_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-cbc/lib/modules/6.6.13/cbc.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-cbc into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-cbc_6.6.13-1_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-hmac/lib/modules/6.6.13/hmac.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-hmac into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-hmac_6.6.13-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13/crypto/rng.ko' is built-in.
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-rng/lib/modules/6.6.13/jitterentropy_rng.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-rng/lib/modules/6.6.13/drbg.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-rng into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-rng_6.6.13-1_x86_64.ipk
NOTICE: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13/lib/crypto/libsha256.ko' is built-in.
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-sha256/lib/modules/6.6.13/sha256_generic.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-sha256/lib/modules/6.6.13/sha256-ssse3.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-sha256 into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-sha256_6.6.13-1_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-sha1/lib/modules/6.6.13/sha1-ssse3.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-sha1/lib/modules/6.6.13/sha1_generic.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-crypto-sha1 into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-crypto-sha1_6.6.13-1_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-asn1-decoder/lib/modules/6.6.13/asn1_decoder.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-asn1-decoder into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-asn1-decoder_6.6.13-1_x86_64.ipk
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-asn1-encoder into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-asn1-encoder_6.6.13-1_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-oid-registry/lib/modules/6.6.13/oid_registry.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-oid-registry into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-oid-registry_6.6.13-1_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-random-core/lib/modules/6.6.13/rng-core.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-random-core into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-random-core_6.6.13-1_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-tpm/lib/modules/6.6.13/tpm.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-tpm into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-tpm_6.6.13-1_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-keys-trusted/lib/modules/6.6.13/trusted.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-keys-trusted into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-keys-trusted_6.6.13-1_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-keys-encrypted/lib/modules/6.6.13/encrypted-keys.ko: relocatable
Packaged contents of /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-keys-encrypted into /home/builder/lede_x86/bin/targets/x86/64/packages/kmod-keys-encrypted_6.6.13-1_x86_64.ipk
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-dm/lib/modules/6.6.13/dm-mirror.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-dm/lib/modules/6.6.13/dm-log.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-dm/lib/modules/6.6.13/dm-mod.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-dm/lib/modules/6.6.13/dm-region-hash.ko: relocatable
rstrip.sh: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/packages/ipkg-x86_64/kmod-dm/lib/modules/6.6.13/dm-crypt.ko: relocatable
[... cut ...]
time: package/kernel/linux/compile#8.21#4.27#12.91
    ERROR: package/kernel/linux failed to build.
make[2]: *** [package/Makefile:116: package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$

WYC-2020, 2024-01-23:
l已经说过了6.6没适配x86 别再问这些问题，我也说过 要适配有几个地方需要改，发这些issue没人给你解答的

GreatMichaelLee, 2024-01-23:
哦，我看今天提交了新版6.6内核的一些相关修改，我以为6.6 ok了呢

发自我的iPhone

closed 2024-01-25 by coolsnowwolf