coolsnowwolf/lede #11820: x86默认编译出错

392377870, 2024-01-23:
### 详细叙述

x86默认编译出错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

python headers                                                                    : Check disabled due to --disable-python 
Checking for a 64-bit host to support lmdb                                        : ok 
Checking linker accepts -Wl,-no-undefined                                         : yes 
Checking linker accepts ['-undefined', 'dynamic_lookup']                          : no 
Checking linker accepts ['-Wl,--wrap=test']                                       : yes 
Checking for flex
Checking for program 'flex'                                                       : /home/openwrt-19810729/lede/staging_dir/host/bin/flex 
Using flex version                                                                : flex 2.6.4 
Checking for u_char                                                               : ok 
Checking for u_int32_t                                                            : ok 
Checking for header err.h                                                         : yes 
Checking for header sys/bswap.h                                                   : no 
Checking for header sys/stropts.h                                                 : yes 
Checking for header sys/timeb.h                                                   : yes 
Checking for header sys/times.h                                                   : yes 
Checking for header timezone.h                                                    : no 
Checking for header ttyname.h                                                     : no 
Checking for header netinet/in6.h                                                 : no 
Checking for header netinet6/in6.h                                                : no 
Checking for header curses.h                                                      : yes 
Checking for header term.h                                                        : yes 
Checking for header termcap.h                                                     : yes 
Checking for atexit                                                               : ok 
Checking for cgetent                                                              : not found 
Checking for getprogname                                                          : not found 
Checking for setprogname                                                          : not found 
Checking for gethostname                                                          : ok 
Checking for putenv                                                               : ok 
Checking for rcmd                                                                 : not found 
Checking for readv                                                                : ok 
Checking for sendmsg                                                              : ok 
Checking for setitimer                                                            : ok 
Checking for strlwr                                                               : not found 
Checking for strncasecmp                                                          : ok 
Checking for strptime                                                             : ok 
Checking for strsep                                                               : ok 
Checking for strsep_copy                                                          : not found 
Checking for strtok_r                                                             : ok 
Checking for strupr                                                               : not found 
Checking for swab                                                                 : ok 
Checking for umask                                                                : ok 
Checking for uname                                                                : ok 
Checking for unsetenv                                                             : ok 
Checking for closefrom                                                            : not found 
Checking for err                                                                  : ok 
Checking for warn                                                                 : ok 
Checking for errx                                                                 : ok 
Checking for warnx                                                                : ok 
Checking for flock                                                                : ok 
Checking for writev                                                               : ok 
Checking for hstrerror                                                            : ok 
Checking for socket                                                               : ok 
Checking for getipnodebyname                                                      : not found 
Checking for gethostent                                                           : ok 
Checking for gethostent_r                                                         : not found 
Checking for sethostent                                                           : ok 
[... 85 lines cut ...]
Checking for header usersec.h                                                     : no 
Checking for member method_attrlist in struct secmethod_table                     : not found 
Checking for member method_version in struct secmethod_table                      : not found 
Checking for library ncurses                                                      : yes 
Checking for tgetent in ncurses                                                   : ok 
for compiling <readline/readline.h> with strict prototypes                        : not found 
for workaround to <readline/readline.h> strict prototypes issue                   : not found 
Checking for rl_completion_t                                                      : not found 
Checking for CPPFunction                                                          : ok 
Checking for library readline                                                     : yes 
Checking for rl_completion_matches in readline                                    : ok 
Checking for history_list in readline                                             : ok 
Checking for SHA1_Update                                                          : not found 
Attempting to compile with runtime-switchable x86_64 Intel AES instructions. WARNING - this is temporary.
perl module Parse::Yapp::Driver 1.05                                              : not found 
perl module "Parse::Yapp::Driver" not found
make[3]: *** [Makefile:462: /home/openwrt-19810729/lede/build_dir/target-x86_64_musl/samba-4.14.14/.configured_e7513375d4aee0412e142789921ec856] Error 1
make[3]: Leaving directory '/home/openwrt-19810729/lede/feeds/packages/net/samba4'
time: package/feeds/packages/samba4/compile#20.83#18.10#41.25
    ERROR: package/feeds/packages/samba4 failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/samba4/compile] Error 1
make[2]: Leaving directory '/home/openwrt-19810729/lede'
make[1]: *** [package/Makefile:110: /home/openwrt-19810729/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/openwrt-19810729/lede'
make: *** [/home/openwrt-19810729/lede/include/toplevel.mk:231：world] 错误 2

xingshu1990, 2024-01-24:
[image]
编译小白。看提示是samba4问题，
搜索了一下谷歌，老的姿势如上，https://www.right.com.cn/forum/thread-8267539-1-1.html。
不知道新的修复姿势是怎么样的。
如果是我的话，可能先取消勾选这个samba4先。

GreatMichaelLee, 2024-01-24:
你估计是编译了6.6然后又退回去编译6.1出现的吧。

rm -rf build_dir

closed 2024-01-26 by coolsnowwolf