coolsnowwolf/lede #11821: R2S/R5S 一天左右就死机

zuozhehao, 2024-01-24:
### 详细叙述

R2S/R5S 一天左右就死机，内网PING不通、公网PING也不通，需要拔电源重启，HDMI 不显示，如何查看崩溃日志是什么。
R2S是早期版本，R5S最近新购买的。
2个都有开启IPV6

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R2S/R5S

### 详细日志

make V=s -j1

WYC-2020, 2024-01-24:
可怜的孩子 也只有 这些各种奇葩的硬路由 有这些各种问题

zuozhehao, 2024-01-24:
> 看着像电源的问题

电压不足么？试了下R86S也是一样的。

zuozhehao, 2024-01-25:
> **还有，如果是x86设备依旧死机的话，可以编译的时候选择 intel-microcode 和 amd64-microcode，再试下，因为在这个设备上装ubuntu的话，把这个包卸载了也一样死机 (Linux 6.1内核)**

谢谢，其实我是想确认下，这种死机有什么日志之类的可以排查，这样就比较好针对性解决问题，现在是R2S/R5S/R86S 都有同样问题。@mdaylight

fxxccp, 2024-01-28:
固定时间死机还是任意时间都有可能死机？用ddns测试下，看IP同步是什么时间停止的，这样就可判断死机时间是哪种情况。测试好了我或许可以给你个解决方案。

WYC-2020, 2024-01-29:
说真的 这个coremark 跑分的 就是 自慰的 哈哈

mzyyl, 2024-06-25:
R86S 是n5105的话，尤其用6.1固件的，建议找售后要bios更新，然后刷回5.4固件

zoyo94, 2025-02-16:
##### 每次出现问题的时候，设备wanlan灯就灭了。
##### 查看top，发现有进程占用特别高。
##### 最后放弃了 暂时不用  等后期代码更新 再编译或许就好了

coolsnowwolf, 2026-09-05:
已核对当前仓库及全部后续反馈：该 issue 同时涉及 R2S、R5S、R86S 三种不同硬件，没有串口、kdump、watchdog、温度/电源或崩溃前日志，无法定位统一源码问题。另需注意，选择 coremark 包会安装每日 04:00 执行的 CPU 压测任务（/etc/coremark.sh）；评论中已有反馈删除 cpumark/coremark 后可避免固定时间死机，R86S/N5105 还涉及 BIOS 和供电。建议后续按单一型号保留串口/远程日志，先禁用 coremark、检查电源和 BIOS，再提交可复现日志。本 issue 关闭为硬件/可选任务环境问题。

closed 2026-09-05 by coolsnowwolf