coolsnowwolf/lede #11824: 6.6 内核编译失败

GreatMichaelLee, 2024-01-25:
### 详细叙述

首先，如果说目前还是如前还没有official的支持6.6 test，那就请忽略这个issue保持open就好啦，因为我看大佬提交了add support kernel 6.6的commit, 而且makefile里吧6.6加为test_ver了，所以是不是可以认为作为测试版本支持了？

提两个问题：

issue1 :  #11819 没有解决，依然报同样的错误，不知道为何关了，我这里稍微改了一下，可以过，请大佬参考，看能否修复。

[image]

issue2:  ksmbd报错，说指针类型不匹配，这我就不知道咋弄了，请大佬看看，具体错误参考详细日志

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/glob.h:15,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/unicode.c:12:
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/smbacl.h: In function 'posix_acl_uid_translate':
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/smbacl.h:231:30: error: passing argument 1 of 'make_vfsuid' from incompatible pointer type [-Werror=incompatible-pointer-types]
  231 |         vfsuid = make_vfsuid(mnt_userns, &init_user_ns, pace->e_uid);
      |                              ^~~~~~~~~~
      |                              |
      |                              struct user_namespace *
In file included from ./include/linux/fs.h:44,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/unicode.c:9:
./include/linux/mnt_idmapping.h:118:40: note: expected 'struct mnt_idmap *' but argument is of type 'struct user_namespace *'
  118 | vfsuid_t make_vfsuid(struct mnt_idmap *idmap,
      |                      ~~~~~~~~~~~~~~~~~~^~~~~
In file included from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/vfs.h:17,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/vfs_cache.h:16,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/glob.h:15,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/unicode.c:12:
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/smbacl.h: In function 'posix_acl_gid_translate':
/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/smbacl.h:264:30: error: passing argument 1 of 'make_vfsgid' from incompatible pointer type [-Werror=incompatible-pointer-types]
  264 |         vfsgid = make_vfsgid(mnt_userns, &init_user_ns, pace->e_gid);
      |                              ^~~~~~~~~~
      |                              |
      |                              struct user_namespace *
In file included from ./include/linux/fs.h:44,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/unicode.c:9:
./include/linux/mnt_idmapping.h:121:40: note: expected 'struct mnt_idmap *' but argument is of type 'struct user_namespace *'
  121 | vfsgid_t make_vfsgid(struct mnt_idmap *idmap,
      |                      ~~~~~~~~~~~~~~~~~~^~~~~
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:243: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/unicode.o] Error 1
make[5]: *** [/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13/Makefile:1913: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8] Error 2
make[4]: *** [Makefile:234: __sub-make] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13'
make[3]: *** [Makefile:76: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/ksmbd-3.4.8/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/package/kernel/ksmbd'
time: package/kernel/ksmbd/compile#0.59#0.82#1.20
    ERROR: package/kernel/ksmbd failed to build.
make[2]: *** [package/Makefile:116: package/kernel/ksmbd/compile] Error 1
make[2]: Leaving directory '/home/builder/lede_x86'
make[1]: *** [package/Makefile:110: /home/builder/lede_x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/builder/lede_x86'
make: *** [/home/builder/lede_x86/include/toplevel.mk:231：world] 错误 2
builder@Build-Server:/home/builder/lede_x86$
builder@Build-Server:/home/builder/lede_x86$

WYC-2020, 2024-01-25:
咋说呢 ，你这个是只管补不管兼容性，然后囫囵吞枣 哈哈
[image]
[image]
如此如此这般这般

GreatMichaelLee, 2024-01-25:
> 咋说呢 ，你这个是只管补不管兼容性，然后囫囵吞枣 哈哈 [...]

哈哈，大佬批评的对，我就是一村干部式的土鳖，只管6.6能出包就行别的阳春白雪咱也不会啊，还是大佬的良方好用:P

WYC-2020, 2024-01-25:
[ksmbd.zip](https://github.com/coolsnowwolf/lede/files/14050204/ksmbd.zip)
删除原来的 直接替换，编译后 自己试试功能是否正常

jdnt, 2024-01-25:
我已经编译成功了。不过是自用精简版，基本没上什么插件可能冲突项较少

hcym, 2024-01-25:
感谢大雕，试试看66

GreatMichaelLee, 2024-01-25:
> 已知问题全部修复，包括无线也都已支持

新issue:

# CC [M]  /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.o
  x86_64-openwrt-linux-musl-gcc -Wp,-MMD,/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/.prot_buf_ip.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -fno-jump-tables -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -falign-functions=16 -fno-strict-overflow -fno-stack-check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wframe-larger-than=2048 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-restrict -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-overflow -Wno-stringop-truncation -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -g -fno-var-tracking -femit-struct-debug-baseonly -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves -DDEBUG -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_new -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/lua -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/lua/include  -DMODULE  -DKBUILD_BASENAME='"prot_buf_ip"' -DKBUILD_MODNAME='"xt_LUA"' -D__KBUILD_MODNAME=kmod_xt_LUA -c -o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --retpoline --stackval --static-call --uaccess   --module /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.o
In file included from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.c:20:
./include/net/checksum.h: In function 'csum_shift':
./include/net/checksum.h:90:40: error: implicit declaration of function 'ror32' [-Werror=implicit-function-declaration]
   90 |                 return (__force __wsum)ror32((__force u32)sum, 8);
      |                                        ^~~~~
In file included from ./include/linux/kernel.h:22,
                 from ./include/linux/skbuff.h:13,
                 from ./include/linux/tcp.h:17,
                 from ./include/net/tcp.h:20,
                 from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.c:21:
./include/linux/bitops.h: At top level:
./include/linux/bitops.h:134:21: error: conflicting types for 'ror32'; have '__u32(__u32,  unsigned int)' {aka 'unsigned int(unsigned int,  unsigned int)'}
  134 | static inline __u32 ror32(__u32 word, unsigned int shift)
      |                     ^~~~~
In file included from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.c:20:
./include/net/checksum.h:90:40: note: previous implicit declaration of 'ror32' with type 'int()'
   90 |                 return (__force __wsum)ror32((__force u32)sum, 8);
      |                                        ^~~~~
cc1: some warnings being treated as errors
make[10]: *** [scripts/Makefile.build:243: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.o] Error 1
make[9]: *** [scripts/Makefile.build:480: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA] Error 2
make[8]: *** [/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13/Makefile:1913: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions] Error 2
make[7]: *** [Makefile:234: __sub-make] Error 2
make[7]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13'
make[6]: *** [Makefile:461: modules] Error 2
[... cut ...]
make[5]: *** [Makefile:621: all-recursive] Error 1
make[5]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22'
make[4]: *** [Makefile:393: all] Error 2
make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22'
make[3]: *** [Makefile:177: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/.built] Error 2
make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/xtables-addons'

hcym, 2024-01-26:
和楼上一样，退回6.1ok，还是老老实实做人

King-Of-Knights, 2024-01-26:
@GreatMichaelLee 你编译出来的6.6稳定嘛？我编出来后，1分钟重启一次T_T

GreatMichaelLee, 2024-01-26:
> @GreatMichaelLee 你编译出来的6.6稳定嘛？我编出来后，1分钟重启一次T_T

我6.6都出不来，卡在上面那个xtable的插件出错了，而且我会退6.1出来的包好像也有问题了-_-! 一会就OOM内核崩了，我现在回退之前的6.1.72内核试试

今天九点多崩了一次：
```
Fri Jan 26 09:10:50 2024 daemon.info netdata[24096]: Initializing file /var/cache/netdata/netdata.statsd_metrics/main.db.
Fri Jan 26 09:10:50 2024 daemon.info netdata[24096]: Initializing file /var/cache/netdata/netdata.statsd_metrics/gauges.db.
Fri Jan 26 09:10:50 2024 daemon.info netdata[24096]: Initializing file /var/cache/netdata/netdata.statsd_metrics/counters.db.
Fri Jan 26 09:10:50 2024 daemon.info netdata[24096]: Initializing file /var/cache/netdata/netdata.statsd_metrics/timers.db.
Fri Jan 26 09:10:50 2024 daemon.info netdata[24096]: thread created with task id 24981
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.076218] netifd invoked oom-killer: gfp_mask=0x140cca(GFP_HIGHUSER_MOVABLE|__GFP_COMP), order=0, oom_score_adj=0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.077013] CPU: 1 PID: 2505 Comm: netifd Not tainted 6.1.73 #0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.077570] Hardware name: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.21805430.B64.2305221826 05/22/2023
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.078654] Call Trace:
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.079092]  <TASK>
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.079467]  dump_stack_lvl+0x49/0x5f
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.079914]  dump_stack+0x10/0x12
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.080335]  dump_header+0x4a/0x1a4
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.080794]  oom_kill_process.cold+0xb/0x10
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.081257]  out_of_memory+0x107/0x2d0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.081697]  __alloc_pages_slowpath.constprop.0+0xa9d/0xc60
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.082223]  __alloc_pages+0x25b/0x290
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.082670]  __folio_alloc+0xf/0x20
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.083107]  __filemap_get_folio+0x163/0x2c0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.083578]  filemap_fault+0x3cb/0x7a0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.088102]  __do_fault+0x31/0x90
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.088556]  __handle_mm_fault+0x9d0/0xdc0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.089038]  handle_mm_fault+0x1ee/0x2a0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.089522]  exc_page_fault+0x196/0x520
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.092677]  asm_exc_page_fault+0x27/0x30
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.093188] RIP: 0033:0x7f912ac56e1a
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.093658] Code: Unable to access opcode bytes at 0x7f912ac56df0.
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.094262] RSP: 002b:00007ffcccc7be40 EFLAGS: 00010207
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.095161] RAX: 0000000000000000 RBX: 00007f912ac5e2c0 RCX: 00007f912acb5f1a
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.095803] RDX: 00007ffcccc7be20 RSI: 00007f912acd6b84 RDI: 0000000000000000
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.096443] RBP: 00007ffcccc7be58 R08: 0000000000000008 R09: 0000000000000004
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.097082] R10: 00000000000003e8 R11: 0000000000000246 R12: 000000000042d900
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.097715] R13: 00007f912ac5e360 R14: 00007f912ac5e040 R15: 0000000000000000
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.098535]  </TASK>
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.099113] Mem-Info:
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.099542] active_anon:307888 inactive_anon:32787 isolated_anon:0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.099542]  active_file:287 inactive_file:2292 isolated_file:0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.099542]  unevictable:0 dirty:95 writeback:0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.099542]  slab_reclaimable:14629 slab_unreclaimable:118819
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.099542]  mapped:527 shmem:12072 pagetables:1285
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.099542]  sec_pagetables:0 bounce:0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.099542]  kernel_misc_reclaimable:0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.099542]  free:15176 free_pcp:348 free_cma:0
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.107202] Node 0 active_anon:1231552kB inactive_anon:131148kB active_file:1148kB inactive_file:9168kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:2108kB dirty:380kB writeback:0kB shmem:48288kB writeback_tmp:0kB kernel_stack:4064kB pagetables:5140kB sec_pagetables:0kB all_unreclaimable? no
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.112588] DMA free:8000kB boost:0kB min:120kB low:148kB high:176kB reserved_highatomic:0KB active_anon:4240kB inactive_anon:2012kB active_file:0kB inactive_file:208kB unevictable:0kB writepending:0kB present:15992kB managed:15360kB mlocked:0kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.115284] lowmem_reserve[]: 0 1973 1973 1973
Fri Jan 26 09:12:17 2024 kern.warn kernel: [341541.115866] DMA32 free:54500kB boost:36580kB min:52840kB low:56904kB high:60968kB reserved_highatomic:0KB active_anon:204036kB inactive_anon:1152412kB active_file [... cut ...]
Fri Jan 26 11:15:29 2024 kern.warn kernel: [ 5838.578574] haproxy invoked oom-killer: gfp_mask=0x140cca(GFP_HIGHUSER_MOVABLE|__GFP_COMP), order=0, oom_score_adj=0
Fri Jan 26 11:15:29 2024 kern.warn kernel: [ 5838.579357] CPU: 0 PID: 11599 Comm: haproxy Not tainted 6.1.73 #0
Fri Jan 26 11:15:29 2024 kern.warn kernel: [ 5838.579879] Hardware name: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.21805430.B64.2305221826 05/22/2023
Fri Jan 26 11:15:29 2024 kern.warn kernel: [ 5838.580901] Call Trace:
Fri Jan 26 11:15:29 2024 kern.warn kernel: [ 5838.581269]  <TASK>
```

WYC-2020, 2024-01-26:
> > 已知问题全部修复，包括无线也都已支持
> 
> 新issue, 麻烦大大看看
> 
> ```
> # CC [M]  /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.o
>   x86_64-openwrt-linux-musl-gcc -Wp,-MMD,/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/.prot_buf_ip.o.d -nostdinc -I./arch/x86/include -I./arch/x86/include/generated  -I./include -I./arch/x86/include/uapi -I./arch/x86/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include ./include/linux/compiler_types.h -D__KERNEL__ -fmacro-prefix-map=./= -std=gnu11 -fshort-wchar -funsigned-char -fno-common -fno-PIE -fno-strict-aliasing -mno-sse -mno-mmx -mno-sse2 -mno-3dnow -mno-avx -fcf-protection=none -m64 -falign-jumps=1 -falign-loops=1 -mno-80387 -mno-fp-ret-in-387 -mpreferred-stack-boundary=3 -mskip-rax-setup -mtune=generic -mno-red-zone -mcmodel=kernel -Wno-sign-compare -fno-asynchronous-unwind-tables -mindirect-branch=thunk-extern -mindirect-branch-register -mindirect-branch-cs-prefix -fno-jump-tables -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fstack-protector -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -falign-functions=16 -fno-strict-overflow -fno-stack-check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wframe-larger-than=2048 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-types -Werror=designated-init -Wenum-conversion -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-restrict -Wno-packed-not-aligned -Wno-format-overflow -Wno-format-truncation -Wno-stringop-overflow -Wno-stringop-truncation -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -Wno-sign-compare -g -fno-var-tracking -femit-struct-debug-baseonly -fmacro-prefix-map=/home/builder/lede_x86/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves -DDEBUG -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_new -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/lua -I/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/lua/include  -DMODULE  -DKBUILD_BASENAME='"prot_buf_ip"' -DKBUILD_MODNAME='"xt_LUA"' -D__KBUILD_MODNAME=kmod_xt_LUA -c -o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.o /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.c   ; ./tools/objtool/objtool --hacks=jump_label --hacks=noinstr --retpoline --stackval --static-call --uaccess   --module /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.o
> In file included from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.c:20:
> ./include/net/checksum.h: In function 'csum_shift':
> ./include/net/checksum.h:90:40: error: implicit declaration of function 'ror32' [-Werror=implicit-function-declaration]
>    90 |                 return (__force __wsum)ror32((__force u32)sum, 8);
>       |                                        ^~~~~
> In file included from ./include/linux/kernel.h:22,
>                  from ./include/linux/skbuff.h:13,
>                  from ./include/linux/tcp.h:17,
>                  from ./include/net/tcp.h:20,
>                  from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.c:21:
> ./include/linux/bitops.h: At top level:
> ./include/linux/bitops.h:134:21: error: conflicting types for 'ror32'; have '__u32(__u32,  unsigned int)' {aka 'unsigned int(unsigned int,  unsigned int)'}
>   134 | static inline __u32 ror32(__u32 word, unsigned int shift)
>       |                     ^~~~~
> In file included from /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.c:20:
> ./include/net/checksum.h:90:40: note: previous implicit declaration of 'ror32' with type 'int()'
>    90 |                 return (__force __wsum)ror32((__force u32)sum, 8);
>       |                                        ^~~~~
> cc1: some warnings being treated as errors
> make[10]: *** [scripts/Makefile.build:243: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA/prot_buf_ip.o] Error 1
> make[9]: *** [scripts/Makefile.build:480: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions/LUA] Error 2
> make[8]: *** [/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13/Makefile:1913: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/extensions] Error 2
> make[7]: *** [Makefile:234: __sub-make] Error 2
> make[7]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.13'
[... cut ...]
> make[4]: *** [Makefile:393: all] Error 2
> make[4]: Leaving directory '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22'
> make[3]: *** [Makefile:177: /home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/xtables-addons-3.22/.built] Error 2
> make[3]: Leaving directory '/home/builder/lede_x86/feeds/packages/net/xtables-addons'
> ```
[211-fix-build-with-kernel-6.6.patch](https://github.com/coolsnowwolf/lede/files/14060067/211-fix-build-with-kernel-6.6.patch)

GreatMichaelLee, 2024-01-26:
> [211-fix-build-with-kernel-6.6.patch](https://github.com/coolsnowwolf/lede/files/14060067/211-fix-build-with-kernel-6.6.patch)

大佬又出手了，每次大佬出手，我就知道我离出包又近了一步...^^

kxdn3, 2024-01-26:
为啥我随便编都成功，稳得很，用的云编译

coolsnowwolf, 2024-01-26:
楼上的 oom 是 netdata 的问题

coolsnowwolf, 2024-01-26:
xtables-addons 的问题已修复， netdata 别编译了，这东西其实没啥用，吃光你内存而已

GreatMichaelLee, 2024-01-26:
> xtables-addons 的问题已修复， netdata 别编译了，这东西其实没啥用，吃光你内存而已

好的，感谢l大建议。

GreatMichaelLee, 2024-01-26:
出img包了！目前我的环境上所有6.6的问题已修复，感谢几位大佬的帮助！

Pacio, 2024-01-26:
> 出img包了！目前我的环境上所有6.6的问题已修复，感谢几位大佬的帮助！

试试还有重启Bug吗

GreatMichaelLee, 2024-01-26:
> > 出img包了！目前我的环境上所有6.6的问题已修复，感谢几位大佬的帮助！
> 
> 试试还有重启Bug吗

我先跑跑，按l大建议去掉了netdata,如果24H没事，我认为就比较稳定了。

WYC-2020, 2024-01-26:
> > > 出img包了！目前我的环境上所有6.6的问题已修复，感谢几位大佬的帮助！
> > 
> > 
> > 试试还有重启Bug吗
> 
> 我先跑跑，按l大建议去掉了netdata,如果24H没事，我认为就比较稳定了。

我netdata稳如狗呢哈哈

WYC-2020, 2024-01-26:
[image]
[image]
干干净净的日志毫无错误

GreatMichaelLee, 2024-01-26:
> _20240126155206.png (view on web)
>  _20240126155215.png (view on web)

我咋个觉得你把我和l大的脸一起打了，哈哈

发自我的iPhone

WYC-2020, 2024-01-26:
> 我咋个觉得你把我和l大的脸一起打了，哈哈 发自我的iPhone
> […](#)

你这话说的我就不爱听了，你是想给我引战么小样，别人大雕说的是oom那个人这个引起的并没说 netdata在6.6上面可能有啥问题哈哈，不知道你们都编译了什么，我这边目前没出现这个玩意
[image]

coolsnowwolf, 2024-01-26:
我这边测试 netdata 也没发现 oom，可能是被其他组件影响，例如之前有人编译 oaf

GreatMichaelLee, 2024-01-26:
大佬，take it easy，玩笑的话别这么严肃嘛～

发自我的iPhone

  

  
 image.png (view on web)

binge8, 2024-01-26:
大婶们来活了，继续缝缝补补了
```bash
/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.3/net/batman-adv/netlink.c:1497:21: error: initialization of 'int (*)(const struct genl_split_ops *, struct sk_buff *, struct genl_info *)' from incompatible pointer type 'int (*)(const struct genl_ops *, struct sk_buff *, struct genl_info *)' [-Werror=incompatible-pointer-types]
 1497 |         .pre_doit = batadv_pre_doit,
      |                     ^~~~~~~~~~~~~~~
/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.3/net/batman-adv/netlink.c:1497:21: note: (near initialization for 'batadv_netlink_family.pre_doit')
/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.3/net/batman-adv/netlink.c:1498:22: error: initialization of 'void (*)(const struct genl_split_ops *, struct sk_buff *, struct genl_info *)' from incompatible pointer type 'void (*)(const struct genl_ops *, struct sk_buff *, struct genl_info *)' [-Werror=incompatible-pointer-types]
 1498 |         .post_doit = batadv_post_doit,
      |                      ^~~~~~~~~~~~~~~~
/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.3/net/batman-adv/netlink.c:1498:22: note: (near initialization for 'batadv_netlink_family.post_doit')
cc1: some warnings being treated as errors
make[6]: *** [scripts/Makefile.build:243: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.3/net/batman-adv/netlink.o] Error 1
make[5]: *** [/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.14/Makefile:1913: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.3/net/batman-adv] Error 2
make[4]: *** [Makefile:234: __sub-make] Error 2
make[3]: *** [Makefile:104: /home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/batman-adv-2022.3/.built] Error 2
make[4]: Leaving directory '/home/runner/work/op/op/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.14'
make[3]: Leaving directory '/home/runner/work/op/op/openwrt/feeds/routing/batman-adv'
time: package/feeds/routing/batman-adv/compile#8.91#1.50#10.41
    ERROR: package/feeds/routing/batman-adv failed to build.
make[2]: *** [package/Makefile:116: package/feeds/routing/batman-adv/compile] Error 1
make[2]: Leaving directory '/home/runner/work/op/op/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/op/op/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/runner/work/op/op/openwrt'
make: *** [/home/runner/work/op/op/openwrt/include/toplevel.mk:231: world] Error 2
```

coolsnowwolf, 2024-01-26:
batman-adv 已修复支持 6.6 内核了，继续吧

closed 2024-01-26 by coolsnowwolf

gbm777, 2024-02-01:
filogic 是否已经有 6.1或者6.6的patch了？