coolsnowwolf/lede #11829: 6.6内核编译mt7922失败

dph5199278, 2024-01-26:
### 详细叙述

做个标记
[mt7921u active monitor mode breaks driver](https://github.com/openwrt/mt76/issues/839#issuecomment-1819149375)

原先mt76包编译不通过，升级到了最新包，还是编译不通过。
不知大佬们有没有解决方法，或者需要等官方mt76驱动适配6.6之后才能编译？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64

### 详细日志

```bash
touch /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816/.built_check
make -C /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.14 KCFLAGS="-fmacro-prefix-map=/opt/lede/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/opt/lede/staging_dir/host/include -I/opt/lede/staging_dir/hostpkg/include -I/opt/lede/staging_dir/target-x86_64_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="Dely" KBUILD_BUILD_HOST="China" KBUILD_BUILD_TIMESTAMP="Fri Jan 26 05:13:56 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/opt/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/batman-adv.symvers /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/button-hotplug.symvers /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/cryptodev-linux.symvers /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/fast-classifier.symvers /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/fullconenat.symvers /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/mac80211.symvers /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8125.symvers /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8152.symvers /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/r8168.symvers /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/shortcut-fe.symvers /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/siit.symvers" KERNELRELEASE=6.6.14   M="/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816" NOSTDINC_FLAGS="-nostdinc  -isystem /opt/lede/staging_dir/toolchain-x86_64_gcc-12.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/12.3.0/include -I/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816 -I/opt/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport/uapi -I/opt/lede/staging_dir/target-x86_64_musl/usr/include/mac80211-backport -I/opt/lede/staging_dir/target-x86_64_musl/usr/include/mac80211/uapi -I/opt/lede/staging_dir/target-x86_64_musl/usr/include/mac80211 -include backport/autoconf.h -include backport/backport.h -DCONFIG_MAC80211_MESH" modules
make[4]: Entering directory '/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.14'
  CC [M]  /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816/mmio.o
In file included from /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816/mmio.c:6:
/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816/mt76.h: In function 'mt76_put_page_pool_buf':
/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816/mt76.h:1646:9: error: implicit declaration of function 'page_pool_put_full_page' [-Werror=implicit-function-declaration]
 1646 |         page_pool_put_full_page(page->pp, page, allow_direct);
      |         ^~~~~~~~~~~~~~~~~~~~~~~
/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816/mt76.h: In function 'mt76_get_page_pool_buf':
/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816/mt76.h:1654:16: error: implicit declaration of function 'page_pool_dev_alloc_frag' [-Werror=implicit-function-declaration]
 1654 |         page = page_pool_dev_alloc_frag(q->page_pool, offset, size);
      |                ^~~~~~~~~~~~~~~~~~~~~~~~
/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816/mt76.h:1654:14: error: assignment to 'struct page *' from 'int' makes pointer from integer without a cast [-Werror=int-conversion]
 1654 |         page = page_pool_dev_alloc_frag(q->page_pool, offset, size);
      |              ^
cc1: all warnings being treated as errors
make[6]: *** [scripts/Makefile.build:243: /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816/mmio.o] Error 1
make[5]: *** [/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.14/Makefile:1913: /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816] Error 2
make[4]: *** [Makefile:234: __sub-make] Error 2
make[4]: Leaving directory '/opt/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.6.14'
make[3]: *** [Makefile:629: /opt/lede/build_dir/target-x86_64_musl/linux-x86_64/mt76-2024-01-18-f7718816/.built] Error 2
make[3]: Leaving directory '/opt/lede/package/kernel/mt76'
time: package/kernel/mt76/compile#0.85#0.11#8.96
    ERROR: package/kernel/mt76 failed to build.
make[2]: *** [package/Makefile:116: package/kernel/mt76/compile] Error 1
make[2]: Leaving directory '/opt/lede'
make[1]: *** [package/Makefile:110: /opt/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/opt/lede'
make: *** [/opt/lede/include/toplevel.mk:231: world] Error 2
```

coolsnowwolf, 2024-01-26:
已修复

WYC-2020, 2024-01-26:
哈哈，刚修好说准备放patch被捷足先登了

closed 2024-01-26 by coolsnowwolf