coolsnowwolf/lede #11831: 6.6 内核在满载测速的时候出现OOM

GreatMichaelLee, 2024-01-26:
### 详细叙述

上次说是netdata的问题，这次没编译netdata,好家伙，把xray给我干了。具体见日志。只要一测速，就大概率挂，出kernel dump,有时候掉网，看OOM干的是哪个。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.398379] swapper/3: page allocation failure: order:0, mode:0x820(GFP_ATOMIC), nodemask=(null),cpuset=/,mems_allowed=0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.399770] CPU: 3 PID: 0 Comm: swapper/3 Tainted: G           O       6.6.14 #0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.400756] Hardware name: VMware, Inc. VMware7,1/440BX Desktop Reference Platform, BIOS VMW71.00V.21805430.B64.2305221826 05/22/2023
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.403007] Call Trace:
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.403634]  <IRQ>
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.404673]  duƹ??lvl+0x48/0x60
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.405822]  duƹ?k+0x10/0x20
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.406295]  w?n_??+0x100/0x180
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.406943]  ?????wp?h.??p?+0xb4c/0xc00
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.407442]  ???+0x25b/0x290
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.407879]  n}????kb+0x140/0x1c0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.415242]  igc_update_stats+0x1d1b/0x30e0 [igc]
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.415783]  n}i?l+0x27/0x160
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.418068]  ?r???+0x12d/0x250
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.418496]  ?of?rq+0xb6/0x1f6
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.418902]  ǚسrcu+0x41/0x60
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.419295]  ysve?}i??m??r?t+0x74/0x80
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.419713]  </IRQ>
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.420189]  <TASK>
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.423255]  ?m?sve?}i??m??r?t+0x1b/0x20
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.423734] RIP: 0010:v_n?iv?af?h?t+0x13/0x20
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.426005] Code: 00 00 00 00 00 66 90 0f 22 d7 c3 66 66 2e 0f 1f 84 00 00 00 00 00 90 8b 05 6a 55 8e 00 85 c0 7e 07 0f 00 2d df 10 27 00 fb f4 <c3> 66 66 2e 0f 1f 84 00 00 00 00 00 90 83 ff 03 74 26 7f 0e 83 ff
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.430112] RSP: 0000:ffffc900000b7e20 EFLAGS: 00000242
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.430647] RAX: 0000000000000001 RBX: ffff888001b63400 RCX: 4000000000000000
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.431232] RDX: ffff88807dd80000 RSI: ffff888001a80000 RDI: 0000000000000001
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.431836] RBP: ffffc900000b7e28 R08: ffff888001b63400 R09: fffffffffffffffc
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.432441] R10: 0000000000000004 R11: 0000000000000000 R12: ffff888001a80064
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.433059] R13: ffff888001a80064 R14: ffffffff820d8040 R15: ffffffff820d8028
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.433667]  ? ?i?af?h?t+0x19/0x30
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.434122]  ???d???y+0x3d/0x60
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.434559]  ????+0x8a/0xd0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.434975]  Ϫ????+0x77/0x2d0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.435394]  Ϫ??+0x2f/0x40
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.435824]  ?l?Ϫ?+0x1e/0x40
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.436253]  dת?+0xd2/0x170
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.436671]  Ϲ)???y+0x28/0x30
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.437100]  ??<??d?y+0x101/0x110
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.437556]  ??d?y?)?=_n?v?ify+0x178/0x17b
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.438027]  </TASK>
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.438456] Mem-Info:
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.438879] active_anon:11027 inactive_anon:258760 isolated_anon:0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.438879]  active_file:782 inactive_file:36 isolated_file:0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.438879]  unevictable:0 dirty:459 writeback:0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.438879]  slab_reclaimable:14751 slab_unreclaimable:108618
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.438879]  mapped:511 shmem:4376 pagetables:1143
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.438879]  sec_pagetables:0 bounce:0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.438879]  kernel_misc_reclaimable:0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.438879]  free:4774 free_pcp:627 free_cma:0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.442597] Node 0 active_anon:44108kB inactive_anon:1035040kB active_file:3128kB inactive_file:144kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:2044kB dirty:1836kB writeback:0kB shmem:17504kB writeback_tmp:0kB kernel_stack:3728kB pagetables:4572kB sec_pagetables:0kB all_unreclaimable? no
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.444511] DMA free:7940kB boost:0kB min:120kB low:148kB high:176kB reserved_highatomic:0KB active_anon:108kB inactive_anon:140kB active_file:40kB inactive_file:0kB unevictable:0kB writepending:0kB present:15992kB managed:15360kB mlocked:0kB bounce:0kB free_pcp:20kB local_pcp:0kB free_cma:0kB
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.449355] ping invoked oom-killer: gfp_mask=0x140cca(GFP_HIGHUSER_MOVABLE|__GFP_COMP), order=0, oom_score_adj=0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.451568] lowmem_reserve[]: 0 1975 1975 1975
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.451573] DMA32 free:11156kB boost:0kB min:16260kB low:20324kB high:24388kB reserved_highatomic:12288KB active_anon:1040716kB inactive_anon:38184kB active_file:1196kB inactive_file:2036kB unevictable:0kB writepending:1836kB present:2080156kB managed:2025596kB mlocked:0kB bounce:0kB free_pcp:2484kB local_pcp:228kB free_cma:0kB
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.451577] lowmem_reserve[]: 0 0 0 0
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.451579] DMA: 1*4kB (M) 2*8kB (M) 3*16kB (UM) 2*32kB (UM) 2*64kB (U) 2*128kB (UM) 1*256kB (U) 2*512kB (UM) 2*1024kB (UM) 2*2048kB (UM) 0*4096kB = 7940kB
Sat Jan 27 00:14:32 2024 kern.warn kernel: [  795.451590] DMA32: 125*4kB (UMH) 23*8kB (UMH) 27*16kB (H) 21*32kB (H) 7*64kB (H) 2*128kB (H) 1*256kB (H) 1*512kB (H) 1*1024kB (H) 1*2048kB (H) 1*4096kB (H) = 10428kB
[... 108 lines cut ...]
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.579806] [   2820]     0  2820      385      128    45056        0             0 dynamic_dns_upd
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.580448] [   2822]     0  2822      385      160    36864        0             0 dynamic_dns_upd
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.581136] [   2824]     0  2824      385       96    45056        0             0 dynamic_dns_upd
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.581732] [   2826]     0  2826      385      128    40960        0             0 dynamic_dns_upd
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.582458] [  27302]     0 27302     8561     1394    98304        0             0 zerotier-one
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.583063] [  16635]     0 16635      239       96    45056        0             0 dropbear
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.583640] [  16636]     0 16636      284       32    45056        0             0 ash
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.584290] [  23842]     0 23842      281        0    40960        0             0 sleep
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.585033] [  29821]     0 29821      283       64    40960        0             0 sleep
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.586013] [  30896]     0 30896      281       32    40960        0             0 sleep
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.587014] [  30906]     0 30906      281        0    40960        0             0 sleep
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.587584] [  31065]     0 31065      281       64    45056        0             0 sleep
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.588365] [  31536]     0 31536      281       32    45056        0             0 sleep
[... cut ...]
Sat Jan 27 00:14:32 2024 kern.info kernel: [  795.598635] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=xray,pid=18148,uid=0
Sat Jan 27 00:14:32 2024 kern.err kernel: [  795.599585] Out of memory: Killed process 18148 (xray) total-vm:2217424kB, anon-rss:959936kB, file-rss:128kB, shmem-rss:0kB, UID:0 pgtables:2080kB oom_score_adj:0
Sat Jan 27 00:14:32 2024 kern.err kernel: [  796.031625] SQUASHFS error: Failed to read block 0x9e1172e: -12

GreatMichaelLee, 2024-01-26:
只能先把虚拟机内存2G调到4G再观察一下。。。。。。。

coolsnowwolf, 2024-01-27:
用 SFE 还是 Flow 加速？ 怎么虚拟机里面还有  IGC ？

jliveusa, 2024-01-27:
6.6内核，使用flow加速同样出现这个问题

GreatMichaelLee, 2024-01-27:
> igc

虚拟机用的网卡直通，软件流量分载那个没勾选。

jliveusa, 2024-01-27:
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.751690] ------------[ cut here ]------------
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.752124] rcuref - imbalanced put()
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.752130] WARNING: CPU: 3 PID: 7471 at rcu?f?u>?wp?h+0x5f/0x70
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.752877] Modules linked in: pppoe ppp_async i915 amdgpu video pppox ppp_mppe ppp_generic ipt_REJECT ax88179_178a aqc111 zstd xt_time xt_tcpudp xt_tcpmss xt_string xt_statistic xt_state xt_socket xt_recent xt_quota2(O) xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_length xt_iprange xt_hl xt_helper xt_geoip(O) xt_ecn xt_dscp xt_conntrack xt_connmark xt_connlimit xt_connlabel xt_connbytes xt_comment xt_cgroup xt_bpf xt_addrtype xt_TPROXY xt_TCPMSS xt_REDIRECT xt_PROTO(O) xt_NETMAP xt_MASQUERADE xt_LUA(O) xt_LOG xt_HL xt_FLOWOFFLOAD xt_DSCP xt_CT xt_CLASSIFY xt_CHECKSUM xt_ACCOUNT(O) wmi via_velocity usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_fsm ts_bm tcp_bbr slhc rtl8150 r8168(O) r8152(O) r8125(O) pcnet32 nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_numgen nft_log nft_limit nft_hash nft_ct nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables nf_socket_ipv6
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.752933]  nf_socket_ipv4 nf_reject_ipv4 nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast ts_kmp nf_conntrack_amanda nf_conncount mlx5_core mlx4_en mlx4_core macvlan lzo_rle lzo iptable_raw iptable_nat iptable_mangle iptable_filter ipt_ECN ip6table_raw ip_tables igc iavf i40e gpu_sched forcedeth e1000e drm_display_helper drm_buddy crc_ccitt compat_xtables(O) br_netfilter bnx2x bnx2 asn1_decoder amdxcp alx 8139too 8139cp ntfs3 evdev i2c_dev cryptodev(O) xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ipmac ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink ip6table_nat
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.758326]  nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip6t_NPT ip6t_rt ip6t_mh ip6t_ipv6header ip6t_hbh ip6t_frag ip6t_eui64 ip6t_ah ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 nfsv4 nfsv3 nfsd nfs msdos bonding ixgbe igbvf e1000 amd_xgbe mdio_devres sit mdio netlink_diag tunnel4 ip_tunnel udp_diag tcp_diag raw_diag inet_diag rpcsec_gss_krb5 auth_rpcgss oid_registry tun lockd sunrpc grace dns_resolver raid10 raid1 raid0 md_mod nls_utf8 vxlan udp_tunnel ip6_udp_tunnel ena crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha512_ssse3 sha512_generic sha1_ssse3 sha1_generic seqiv geniv drbg md5 hmac des_generic libdes cts chacha20poly1305 cbc authencesn authenc arc4 drm_suballoc_helper drm_exec fuse nls_iso8859_1 nls_cp437 uas sdhci_pltfm igb vfat fat exfat btrfs zstd_decompress zstd_compress zstd_common xxhash xor raid6_pq lzo_decompress lzo_compress libcrc32c button_hotplug(O) tg3 mii libphy
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.770365] CPU: 3 PID: 7471 Comm: dnsmasq Tainted: G           O       6.6.14 #0
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.771027] Hardware name: Microsoft Corporation Virtual Machine/Virtual Machine, BIOS Hyper-V UEFI Release v4.0 11/01/2019
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.771795] RIP: 0010:rcu?f?u>?wp?h+0x5f/0x70
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.772387] Code: 77 0c 85 c0 79 e2 c7 03 00 00 00 a0 eb da 80 3d 4c ab cd 00 00 75 15 48 c7 c7 02 55 de 81 c6 05 3c ab cd 00 01 e8 51 13 c9 ff <0f> 0b c7 03 00 00 00 e0 31 c0 eb b4 cc cc cc cc cc 48 89 f9 83 e1
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.774169] RSP: 0018:ffffc9000f767af0 EFLAGS: 00010282
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.774853] RAX: 0000000000000019 RBX: ffff888012b9e040 RCX: 0000000000000027
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.775544] RDX: ffff88805f8dd588 RSI: 0000000000000001 RDI: ffff88805f8dd580
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.776229] RBP: ffffc9000f767af8 R08: 0000000000000000 R09: 00000000ffffefff
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.776915] R10: ffffc9000f767978 R11: ffffffff820b2f08 R12: ffff88805f8e5800
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.777643] R13: ffff888012b9e000 R14: 0000000000000001 R15: ffff8880149fe780
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.778333] FS:  00007f9432715b48(0000) GS:ffff88805f8c0000(0000) knlGS:0000000000000000
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.779050] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.779707] CR2: 0000000001e58820 CR3: 00000000112d2000 CR4: 00000000003506e0
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.780399] Call Trace:
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.780955]  <TASK>
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.781547]  ? ?_?s+0x61/0x70
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.782102]  ? w?n+0x6e/0xd0
[... cut ...]
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.796796] RBP: 00005581cfdc4af0 R08: 00007f9432716ef4 R09: 0000000000000010
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.797288] R10: 0000000000000000 R11: 0000000000000246 R12: 000000000000002c
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.797744] R13: 00007f9432716ed0 R14: 000000000000000c R15: 000000000000002a
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.798188]  </TASK>
Sat Jan 27 19:42:12 2024 kern.warn kernel: [   72.798494] ---[ end trace 0000000000000000 ]---

closed 2024-01-31 by coolsnowwolf