coolsnowwolf/lede #11836: libopenssl-legacy 依赖问题

caixin1218, 2024-01-28:
### 详细叙述

helloworld  提示依赖找不到

并且，广告屏蔽大师 Plus+ 只要启用网络就断了。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

WARNING: Makefile 'package/feeds/helloworld/shadowsocksr-libev/Makefile' has a dependency on 'libopenssl-legacy', which does not exist
WARNING: Makefile 'package/feeds/helloworld/shadowsocksr-libev/Makefile' has a dependency on 'libopenssl-legacy', which does not exist
WARNING: Makefile 'package/feeds/helloworld/shadowsocksr-libev/Makefile' has a dependency on 'libopenssl-legacy', which does not exist
WARNING: Makefile 'package/feeds/helloworld/shadowsocksr-libev/Makefile' has a dependency on 'libopenssl-legacy', which does not exist
WARNING: Makefile 'package/feeds/helloworld/shadowsocksr-libev/Makefile' has a dependency on 'libopenssl-legacy', which does not exist

ilxp, 2024-01-28:
同样编译报错。

WYC-2020, 2024-01-29:
你们自己openssl 回退过吧 我记得不错的话 这个是openssl 3 后面新做的一个模块 老的协议 必须依赖这个玩意

caixin1218, 2024-01-29:
> 你们自己openssl 回退过吧 我记得不错的话 这个是openssl 3 后面新做的一个模块 老的协议 必须依赖这个玩意

没有回退，git pull 
./scripts/feeds update -a   
./scripts/feeds install -a
之后就是这样的。

WYC-2020, 2024-01-29:
那应该这里有才对撒 
[image]
不可能会出现没有这个包哦

caixin1218, 2024-01-29:
> > 你们自己openssl 回退过吧 我记得不错的话 这个是openssl 3 后面新做的一个模块 老的协议 必须依赖这个玩意
> 
> 没有回退，git pull ./scripts/feeds update -a ./scripts/feeds install -a 之后就是这样的。

[image]

还真的没有

WYC-2020, 2024-01-29:
你怕是有毒吧哈哈，我看l的仓库 openssl就有这玩意 
[image]

caixin1218, 2024-01-29:
> 你怕是有毒吧哈哈，我看l的仓库 openssl就有这玩意 [...]

你真幽默，我也看到了，但就不知道是什么原因，makd distclean 都干过没用。

caixin1218, 2024-01-29:
> 你怕是有毒吧哈哈，我看l的仓库 openssl就有这玩意 [...]

如果 没有Helloworld 就没有问题。

WYC-2020, 2024-01-29:
怕不是什么配置 导致 openssl没有全部打开选项吧 你是默认config配置 都不能看到这个玩意？

caixin1218, 2024-01-29:
> 怕不是什么配置 导致 openssl没有全部打开选项吧 你是默认config配置 都不能看到这个玩意？

rm -rf    feeds   .config  还是看不到

caixin1218, 2024-01-29:
[image]
还是这样，应该是源码哪个地方问题。

WYC-2020, 2024-01-29:
可怜的孩子

closed 2024-01-30 by caixin1218

725216, 2024-03-01:
一样的问题
Configuring ppp-mod-pppoe.
Configuring dnsmasq-full.
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-i18n-ssr-plus-zh-cn:
 * 	libopenssl-legacy
 * opkg_install_cmd: Cannot install package luci-i18n-ssr-plus-zh-cn.
Makefile:154: recipe for target 'package_install' failed
make[2]: *** [package_install] Error 255
Makefile:112: recipe for target '_call_image' failed
make[1]: *** [_call_image] Error 2
Makefile:207: recipe for target 'image' failed
make: *** [image] Error 2