coolsnowwolf/lede #11837: xl2tpd在MT7621上断流

lengmo23, 2024-01-28:
### 详细叙述

校园网使用L2TP拨号，编译带xl2tpd模块，修改MTU1400，成功获得IPV4并且能够ping通外网。但是使用：UU加速器，todesk等（认为是UDP协议会导致ping外网失去相应）。此时UU加速器开始出现丢包，关闭UU加速器，恢复正常，中继校网WIFI，不存在断流问题。使用官方OP无上述情况

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

MT7621 Xiaomi R3G

### 详细日志

编译正常

WYC-2020, 2024-01-29:
其他没必要的插件关闭看下 特别是sfe那个加速

lengmo23, 2024-01-29:
> 其他没必要的插件关闭看下 特别是sfe那个加速

MT7621编译了turboacc，不开流量分载的时候，默认显示Mediatek HWNAT，是硬件NAT，开启的时候显示flow offload。有怀疑过是HWNAT问题，因为OP官方不含，但是LEAN大编译应该是默认带HWNAT，无法确定

coolsnowwolf, 2026-09-05:
该 issue 没有提供 xl2tpd/pppd 日志、MTU/PMTU、路由策略或断流时刻，只描述 UDP 应用流量触发丢包。当前 xl2tpd 1.3.18 的 netifd 协议脚本会将配置的 `mtu` 同时设置为 PPP MTU/MRU，默认 options 也使用 1400；L2TP/PPPoE/校园网路径仍可能需要更小 MTU、禁用 UDP 分片或调整 MSS。没有证据表明当前 xl2tpd 代码有 MT7621 特定缺陷，无法安全提交修复；请补充完整日志和抓包后再分析。issue 关闭。

closed 2026-09-05 by coolsnowwolf