coolsnowwolf/lede #11838: iperf3报错

ywt114, 2024-01-28:
### 详细叙述

报错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

2024-01-28T20:49:01.0797613Z Collected errors:
2024-01-28T20:49:01.0801169Z  * check_data_file_clashes: Package iperf3-ssl wants to install file /home/runner/work/test/test/openwrt/build_dir/target-x86_64_musl/root-x86/usr/bin/iperf3
2024-01-28T20:49:01.0804184Z 	But that file is already provided by package  * iperf3
2024-01-28T20:49:01.0805634Z Configuring kmod-usb-audio.
2024-01-28T20:49:01.0807064Z  * opkg_install_cmd: Cannot install package iperf3-ssl.
2024-01-28T20:49:01.0808160Z make[2]: *** [package/Makefile:70: package/install] Error 255
2024-01-28T20:49:01.0855505Z make[2]: Leaving directory '/home/runner/work/test/test/openwrt'
2024-01-28T20:49:01.0882900Z make[1]: *** [package/Makefile:111: /home/runner/work/test/test/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
2024-01-28T20:49:01.0887527Z make[1]: Leaving directory '/home/runner/work/test/test/openwrt'
2024-01-28T20:49:01.0935112Z make: *** [/home/runner/work/test/test/openwrt/include/toplevel.mk:231: world] Error 2
2024-01-28T20:49:01.1315422Z ##[error]Process completed with exit code 2.

WYC-2020, 2024-01-29:
[image]
 冲突了吧  安装，选择其中一个安装即可了

ywt114, 2024-01-29:
> ![image](https://private-user-images.githubusercontent.com/34025388/300321495-35eba35f-23e3-4c95-b728-639233be1e1f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

感觉跟lean调整了device_packages有关

WYC-2020, 2024-01-29:
> 感觉跟lean调整了device_packages有关

l我看了 最新的提交 他默认就选中了
[image]
 自然 会冲突了 你取消你的iperf3就行了，我是自己有仓库 所以 我都是把没必要的全部砍掉了的，你如果不是自己仓库 用的l的 那就只能取消你自己那个 或者通过脚本自定义去掉这个

zgc, 2024-01-29:
同样的问题

zgc, 2024-01-29:
> > > > ![image](https://private-user-images.githubusercontent.com/34025388/300321495-35eba35f-23e3-4c95-b728-639233be1e1f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY0OTc4MTgsIm5iZiI6MTcwNjQ5NzUxOCwicGF0aCI6Ii8zNDAyNTM4OC8zMDAzMjE0OTUtMzVlYmEzNWYtMjNlMy00Yzk1LWI3MjgtNjM5MjMzYmUxZTFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI5VDAzMDUxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZlZDAxY2ZkMzlmOWYxMGQ5M2QzZDRmOTY4Yjc0ODdlMjUyOWUyMDAyZDJjOWE4YWQ1NjY5ZWY1MDUyNjg0ZDAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Iezgyr9QqrtwuyX_lsVB_nVLLfhIS6dkF1egfD0vqjQ)
> 
> > > > 冲突了吧 安装，选择其中一个安装即可了
> 
> > > 
> 
> > > 
> 
> > > 感觉跟lean调整了device_packages有关
> 
> > 
> 
> > l我看了 最新的提交 他默认就选中了 ![image](https://private-user-images.githubusercontent.com/34025388/300324376-56e7eaa3-8f21-42f9-a467-bdda6f2a9f48.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY0OTkyNDEsIm5iZiI6MTcwNjQ5ODk0MSwicGF0aCI6Ii8zNDAyNTM4OC8zMDAzMjQzNzYtNTZlN2VhYTMtOGYyMS00MmY5LWE0NjctYmRkYTZmMmE5ZjQ4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTI5VDAzMjkwMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTdjZDVhYmQwZmRmODFjNjc4MjM2ZmM2YTdlN2NmM2JiNDAwY2ViMDJlY2EzMWZkMzIxYzE5MjNiZjQ1M2ExMTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.KrJR9nSoIMfryvhX1UIvkvPUgC3xVsE6wugg0fL7e44) 自然 会冲突了 你取消你的iperf3就行了，我是自己有仓库 所以 我都是把没必要的全部砍掉了的，你如果不是自己仓库 用的l的 那就只能取消你自己那个 或者通过脚本自定义去掉这个
> 
> 
> 
> 是的，正合我意
> 
> sed -i 's/iperf3-ssl[[:space:]]*//g' openwrt/target/linux/x86/Makefile
> 
> 

同样的问题，同样的处理方式

closed 2024-01-29 by aiamadeus