coolsnowwolf/lede #11839: 能启用firewall4吗？passwall2的防火墙工具Nftables貌似依赖这个，当前默认是firewall，复选firewall4导致编译出错

szdosar, 2024-01-29:
### 详细叙述

如题。可否考虑能启用firewall4？
1.目前官方openwrt编译时firewall、firewall4可二选一，单独复选，而且编译出来的passwall2的防火墙工具可以选择Nftables选项；
2.而咱们lede只能选择firewall，导致passwall2的防火墙工具不能选择Nftables，真能选择Iptables；
3.利好，自然Nftables更优秀一些，它被设计为 iptables 的继任者，提供了一个更现代和简化的方式来设置网络规则，使用单一的框架和语法来处理所有类型的流量，消除了需要使用多个不同工具的需要。它还引入了更有效的数据结构，如哈希表，以改善性能，特别是在处理大量规则时表现更优异。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

暂无

WYC-2020, 2024-01-29:
这玩意我记得不错 好像在讨论区讨论过为啥 没切来的 ，你去
[image]
这里找找看看 我记得有人讨论过

ecrasy, 2024-01-29:
可以自行修改
```
sed -i 's/+firewall/+uci-firewall/g' feeds/luci/applications/luci-app-firewall/Makefile
```
然後make menuconfig手動去掉firewall，選擇firewall4

szdosar, 2024-01-29:
收到。我试了下，可以。谢谢两位大佬，后续有什么问题再讨教，叨扰了。@ecrasy @WYC-2020

aiamadeus, 2024-01-29:
第三点是哪个神棍瞎扯的，从数据架构的角度来说明明 iptables 更优秀。

szdosar, 2024-01-29:
> 第三点是哪个神棍瞎扯的，从数据架构的角度来说明明 iptables 更优秀。

见笑了，就当 iptables 更优秀好了。

WYC-2020, 2024-01-29:
> > 第三点是哪个神棍瞎扯的，从数据架构的角度来说明明 iptables 更优秀。
> 
> 见笑了，就当 iptables 更优秀好了。

这个二选一可以的 我看看我能加上不 我也喜欢支持定制的 就像内核 我自己就加了定制 
[image]

szdosar, 2024-01-29:
> 这个二选一可以的 我看看我能加上不 我也喜欢支持定制的 就像内核 我自己就加了定制 [...]

你的建议很好呀。官方openwrt就是可以二选一。

Pacio, 2024-01-29:
nftables不好用，干什么都要手动设置规则，开个upnp我还是手动配置，不适合小白，你要是对nf防火墙规则不熟悉建议别瞎追

szdosar, 2024-01-30:
> nftables不好用，干什么都要手动设置规则，开个upnp我还是手动配置，不适合小白，你要是对nf防火墙规则不熟悉建议别瞎追

谢谢提供建议。我参考一下。

szdosar, 2024-01-30:
我反馈一下目前启用firewall4遇到的情况和问题：
--效果1.passwall2的防火墙工具Nftables可以启用的，但是效能未知；
--后果1.Turbo ACC 网络加速设置的“流量分载”和“全锥型 NAT”不能正常启用；
--后果2.以及导致zerotier插件虽然正常启用，但是无法正常访问同局域网内的其他网段了。
**再次膜拜一下楼上各位大佬。我滚回去了。**
逆向还原
`sed -i 's/+uci-firewall/+firewall/g' feeds/luci/applications/luci-app-firewall/Makefile
`
make menuconfig手动去掉firewall4，老老实实选择firewall。

WYC-2020, 2024-01-30:
> 我反馈一下目前启用firewall4遇到的情况和问题： --效果1.passwall2的防火墙工具Nftables可以启用的，但是效能未知； --后果1.Turbo ACC 网络加速设置的“流量分载”和“全锥型 NAT”不能正常启用； --后果2.以及导致zerotier插件虽然正常启用，但是无法正常访问同局域网内的其他网段了。 **再次膜拜一下楼上各位大佬。我滚回去了。** 逆向还原 [...]

不切是有不切的原因的，还是老实用吧，再说这玩意基本上对于使用来说没有任何区别，没必要去纠结下面到底用的什么，很多库都是依赖的Iptables 没更新，你强制改了 就有可能 出现你上面那些各种杂七杂八的不适配问题，官方op的 不一定都是最适合的，你看luci就是一个典型的代表

szdosar, 2024-01-30:
> 不切是有不切的原因的，还是老实用吧，再说这玩意基本上对于使用来说没有任何区别，没必要去纠结下面到底用的什么，很多库都是依赖的Iptables 没更新，你强制改了 就有可能 出现你上面那些各种杂七杂八的不适配问题，官方op的 不一定都是最适合的，你看luci就是一个典型的代表

说得在理。就像很多ipv4的项目还未支持ipv6一个理，战线拉长（可能要维护两个项目），工作量大，收效却是未知数。

szdosar, 2024-01-30:
> > > > 我反馈一下目前启用firewall4遇到的情况和问题： --效果1.passwall2的防火墙工具Nftables可以启用的，但是效能未知； --后果1.Turbo ACC 网络加速设置的“流量分载”和“全锥型 NAT”不能正常启用； --后果2.以及导致zerotier插件虽然正常启用，但是无法正常访问同局域网内的其他网段了。 **再次膜拜一下楼上各位大佬。我滚回去了。** 逆向还原 `sed -i 's/+uci-firewall/+firewall/g' feeds/luci/applications/luci-app-firewall/Makefile ` make menuconfig手动去掉firewall4，老老实实选择firewall。
> > > 
> > > 
> > > 不切是有不切的原因的，还是老实用吧，再说这玩意基本上对于使用来说没有任何区别，没必要去纠结下面到底用的什么，很多库都是依赖的Iptables 没更新，你强制改了 就有可能 出现你上面那些各种杂七杂八的不适配问题，官方op的 不一定都是最适合的，你看luci就是一个典型的代表
> > 
> > 
> > 说得在理。就像很多ipv4的项目还未支持ipv6一个理，战线拉长（可能要维护两个项目），工作量大，收效却是未知数。
> 
> 如果非要纠结这个性能问题 完全可以直接修改Iptables 增加其性能，我记得dnsmasq就有一个魔改版的，改用了map提交了不知多少性能，但是那也只是在极端很多dns请求的情况下 ，所以基本也就那样，别纠结那丢丢的性能

感谢你的建议，等我弄明白才行。

closed 2024-05-28 by aiamadeus

hank8884, 2024-08-20:
大佬，能出个具体的更换步骤吗，我想把istore云编译的firewall换成4，拜谢