coolsnowwolf/lede #11840: 小米R3G由于batman-adv导致的编译失败

DestoryG, 2024-01-29:
### 详细叙述

由于batman-adv导致的编译出错，本地和Actions均有此情况

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

小米R3G

### 详细日志

make[3]: Entering directory '/workdir/openwrt/feeds/routing/batman-adv'
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/.configured_*
rm -f /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.batman-adv_installed
(cd /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/./; if [ -x ./configure ]; then find /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/ -name config.guess | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/ -name config.guess | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.guess; find /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/ -name config.sub | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/ -name config.sub | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.sub; AR="mipsel-openwrt-linux-musl-gcc-ar" AS="mipsel-openwrt-linux-musl-gcc -c -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3=batman-adv-2023.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=mipsel-openwrt-linux-musl-ld NM="mipsel-openwrt-linux-musl-gcc-nm" CC="mipsel-openwrt-linux-musl-gcc" GCC="mipsel-openwrt-linux-musl-gcc" CXX="mipsel-openwrt-linux-musl-g++" RANLIB="mipsel-openwrt-linux-musl-gcc-ranlib" STRIP=mipsel-openwrt-linux-musl-strip OBJCOPY=mipsel-openwrt-linux-musl-objcopy OBJDUMP=mipsel-openwrt-linux-musl-objdump SIZE=mipsel-openwrt-linux-musl-size CFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3=batman-adv-2023.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -mips16 -minterlink-mips16 -fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3=batman-adv-2023.3 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib -znow -zrelro "   ./configure --target=mipsel-openwrt-linux --host=mipsel-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/.configured_f655dd153b598ca176cf8c99cd9d6c79
rm -f /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/.built
touch /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/.built_check
make  -C "/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.209" KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-mipsel_24kc_musl=target-mipsel_24kc_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include -I/workdir/openwrt/staging_dir/hostpkg/include -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="mipsel-openwrt-linux-musl-" ARCH="mips" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Sun Jan 28 08:09:53 2024" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/cryptodev-linux.symvers /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/gpio-button-hotplug.symvers /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/fullconenat.symvers /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/symvers/mac80211.symvers" KERNELRELEASE=5.10.209 M="/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv" CONFIG_BATMAN_ADV=m CONFIG_BATMAN_ADV_DEBUG=n CONFIG_BATMAN_ADV_BLA=y CONFIG_BATMAN_ADV_DAT=y CONFIG_BATMAN_ADV_MCAST=y CONFIG_BATMAN_ADV_NC=n CONFIG_BATMAN_ADV_BATMAN_V=y CONFIG_BATMAN_ADV_TRACING=n  EXTRA_CFLAGS=" -DCONFIG_BATMAN_ADV=1  -DCONFIG_BATMAN_ADV_BLA=1  -DCONFIG_BATMAN_ADV_DAT=1  -DCONFIG_BATMAN_ADV_MCAST=1  -DCONFIG_BATMAN_ADV_BATMAN_V=1 " NOSTDINC_FLAGS="-nostdinc  -isystem /workdir/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/lib/gcc/mipsel-openwrt-linux-musl/8.4.0/include -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211-backport/uapi -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211 -I/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/usr/include/mac80211/uapi -I/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/include/ -include backport/autoconf.h -include backport/backport.h -include /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/compat-hacks.h -DBATADV_SOURCE_VERSION=\\\"2023.3-openwrt-3\\\"" modules
make[4]: Entering directory '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.10.209'
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/bat_algo.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/bat_iv_ogm.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/bat_v.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/bat_v_elp.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/bat_v_ogm.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/bitarray.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/bridge_loop_avoidance.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/distributed-arp-table.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/fragmentation.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/gateway_client.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/gateway_common.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/hard-interface.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/hash.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/main.o
  CC [M]  /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/batman-adv-2023.3/net/batman-adv/multicast.o
[... cut ...]
make[3]: Leaving directory '/workdir/openwrt/feeds/routing/batman-adv'
time: package/feeds/routing/batman-adv/compile#7.84#0.80#8.41
    ERROR: package/feeds/routing/batman-adv failed to build.
make[2]: *** [package/Makefile:116: package/feeds/routing/batman-adv/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2

WYC-2020, 2024-01-29:
我记得不错 是官方那边 是砍掉了 这个 库 支持的 低版本内核的，等会我看下 我这边这个  可以的话 等会丢给你 你替换 用这个就行了，或者你拉取官方 某个分支的也可以，最新的分支 这个是砍掉了很多 内核支持的

WYC-2020, 2024-01-29:
[batman-adv.tar.gz](https://github.com/coolsnowwolf/lede/files/14079381/batman-adv.tar.gz)
用这个版本试试，这个是我自己用的版本 没有升级过 因为 不咋用所以一直放到 没升级过

DestoryG, 2024-01-29:
我用的是5.10啊，连这个也不支持了？

WYC-2020, 2024-01-29:
> 我用的是5.10啊，连这个也不支持了？

因为openwrt官方没有5.10 自然都砍了 自己看这个库的提交你就知道

WYC-2020, 2024-01-29:
[0006-fix-build-with-kernel-6.6.patch](https://github.com/coolsnowwolf/lede/files/14080071/0006-fix-build-with-kernel-6.6.patch)
我那个 加上这个补丁 通杀5.4 5.10 5.15 6.1 6.6

WYC-2020, 2024-01-29:
有时间我看下 能否升级一下

WYC-2020, 2024-01-29:
[2023.3-batman-adv.tar.gz](https://github.com/coolsnowwolf/lede/files/14081054/2023.3-batman-adv.tar.gz)
拿去吧 基于官方最新改的 我这边x86测试编译 过所有内核没问题，你拿去试试 功能看下是否正常 @DestoryG

closed 2024-01-29 by DestoryG

DL53, 2024-02-17:
这个bug啥时候能解决么？K2编译带easymesh同样出现这个问题，尝试解决失败。

DL53, 2024-02-17:
> [2023.3-batman-adv.tar.gz](https://github.com/coolsnowwolf/lede/files/14081054/2023.3-batman-adv.tar.gz) 拿去吧 基于官方最新改的 我这边x86测试编译 过所有内核没问题，你拿去试试 功能看下是否正常 @DestoryG

这个要怎么用呀？换了自动下载的文件，还是失败

No06, 2024-04-28:
可以用我的 [compat-hacks.h](https://[raw.githubusercontent.com/No06/routing/main/batman-adv/src/compat-hacks.h](https://raw.githubusercontent.com/No06/routing/main/batman-adv/src/compat-hacks.h)) 替换掉 feeds/routing/batman-adv/src 下的 compat-hacks.h，可编译通过

zgc0220, 2024-04-30:
> 可以用我的 compat-hacks.h 替换掉 feeds/routing/batman-adv/src 下的 compat-hacks.h，可编译通过

https://raw.githubusercontent.com/No06/routing/main/batman-adv/src/compat-hacks.h

yflwz, 2024-05-06:
我的是K3，测试可用解决办法为下载以上连接中的batman-adv.tar.gz和0006-fix-build-with-kernel-6.6.patch补丁并放到对应目录下，注意不是2023.3-batman-adv.tar.gz

tarryyun, 2025-03-13:
> 我是K3，测试可用解决办法为下载以上连接中的batman-adv.tar.gz和0006-fix-build-with-kernel-6.6.patch补丁并放到对应目录下，注意不是2023.3-batman-adv.tar.gz

亲，您好，这个补丁放在哪个目录？0006-fix-build-with-kernel-6.6.patch