coolsnowwolf/lede #11841: R66s 6.1.73内核下WAN口崩溃

AndyLin2020, 2024-01-29:
### 详细叙述

开机正常，一会就崩溃了，是哪里出问题了呢？

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

R66s

### 详细日志

内核日志：
[   15.158180] xt_time: kernel timezone is -0000
[   15.181078] PPP generic driver version 2.4.2
[   15.182177] PPP MPPE Compression module registered
[   15.183233] NET: Registered PF_PPPOX protocol family
[   15.192799] kmodloader: done loading kernel modules from /etc/modules.d/*
[   16.710683] EXT4-fs (mmcblk0p1): mounted filesystem without journal. Quota mode: none.
[   17.296364] EXT4-fs (mmcblk0p3): recovery complete
[   17.299932] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Quota mode: none.
[   19.828392] eth0: 0xffff800009c30000, 52:a4:1a:ad:1c:bd, IRQ 21
[   19.877026] br-lan: port 1(eth0) entered blocking state
[   19.877537] br-lan: port 1(eth0) entered disabled state
[   19.878212] device eth0 entered promiscuous mode
[   19.888780] eth1: 0xffff800009c50000, 52:a4:1a:ad:1c:bc, IRQ 33
[   22.609572] r8125: eth0: link up
[   22.609963] br-lan: port 1(eth0) entered blocking state
[   22.610454] br-lan: port 1(eth0) entered forwarding state
[   22.613389] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   24.581575] r8125: eth1: link up
[   24.581902] IPv6: ADDRCONF(NETDEV_CHANGE): eth1: link becomes ready
[  328.381573] eth1: cmd = 0xff, should be 0x07 
[  328.381573] .
[  328.381620] eth1: pci link is down 
[  328.381620] .
[  330.397378] eth1: cmd = 0xff, should be 0x07 
[  330.397378] .
[  330.397877] eth1: pci link is down 
[  330.397877] .
[  332.413315] eth1: cmd = 0xff, should be 0x07 
[  332.413315] .
[  332.414273] eth1: resv_0x1c_l = 0xffff, should be 0x0000 
[  332.414273] .
[  332.414815] eth1: resv_0x1c_h = 0xffff, should be 0x0000 
[  332.414815] .
[  332.415443] eth1: resv_0x20_l = 0xffff, should be 0x0004 
[  332.415443] .
[  332.416071] eth1: resv_0x20_h = 0xffff, should be 0xf021 
[  332.416071] .
[  332.416700] eth1: resv_0x24_l = 0xffff, should be 0x0000 
[  332.416700] .
[  332.417545] eth1: resv_0x24_h = 0xffff, should be 0x0000 
[  332.417545] .
[  332.419238] eth1: esd_flag = 0x03f1
[  332.419238] .
[  334.525242] eth1: cmd = 0xff, should be 0x07 
[  334.525242] .
[  334.525292] eth1: pci link is down 
[  334.525292] .
[  336.541061] eth1: io_base_l = 0xffff, should be 0x0001 
[  336.541061] .
[  336.541535] eth1: mem_base_l = 0xffff, should be 0x0004 
[  336.541535] .
[  336.542135] eth1: mem_base_h = 0xffff, should be 0xf020 
[  336.542135] .
[  336.542739] eth1: resv_0x1c_l = 0xffff, should be 0x0000 
[  336.542739] .
[... 133 lines cut ...]
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.406134] Workqueue: events rtl8125_open [r8125]
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.406571] pstate: 60400009 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.407180] pc : 0xffff800000dd493c
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.407495] lr : 0xffff800000dd4910
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.407809] sp : ffff80000947bc90
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.408099] x29: ffff80000947bc90 x28: 0000000000000140 x27: 00000000ffffffff
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.408728] x26: ffff00000135a440 x25: ffff000006ac03dc x24: 0000000000000003
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.409355] x23: ffff000006adf900 x22: 0000000000418958 x21: ffff000006ac0900
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.409983] x20: 0000000000000001 x19: 0000000000000064 x18: 0000000000000000
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.410610] x17: 000000000690405e x16: 000000003ca693e9 x15: 0000000000000000
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.411237] x14: 0000000000000002 x13: 0000000000058685 x12: ffff800008f87b78
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.411863] x11: 0000000072230000 x10: 0000000000007c00 x9 : 0000000000000100
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.412490] x8 : 0000000000000004 x7 : ffff800008928430 x6 : ffff000006ac0000
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.413116] x5 : 0000000000000000 x4 : ffff000006adf900 x3 : ffff000006ade900
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.413742] x2 : 0000000000000000 x1 : ffff80000947bc50 x0 : 0000000000005dc3
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.414370] Call trace:
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.414586]  0xffff800000dd493c
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.414870]  rtl8125_mac_ocp_read+0x9d4/0x2620 [r8125]
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.415331]  rtl8125_hw_config+0x127c/0x6da0 [r8125]
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.415775]  rtl8125_open+0xc00/0xea0 [r8125]
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.416167]  process_one_work+0x1d0/0x320
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.416524]  worker_thread+0x14c/0x444
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.416855]  kthread+0x10c/0x110
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.417143]  ret_from_fork+0x10/0x20
Mon Jan 29 14:53:44 2024 kern.warn kernel: [  346.417462] ---[ end trace 0000000000000000 ]---

AndyLin2020, 2024-01-29:
> 看来是上游 8125 驱动的锅？

切换到 5.15.146也是一样，晕了

WYC-2020, 2024-01-29:
这驱动 很久都没升级了嘛  感觉像是 最近升级了什么导致了 不匹配吧 试试 回滚几个 看看

AndyLin2020, 2024-01-29:
全锥型 NAT默认开启，关了后好像正常了

aiamadeus, 2024-01-29:
这个是 pcie 信号有问题，跟软件没关系

AndyLin2020, 2024-01-29:
> 这个是 pcie 信号有问题，跟软件没关系

会是硬件问题吗？今天刚入手，准备做一个旁路由

WYC-2020, 2024-01-30:
> > 这个是 pcie 信号有问题，跟软件没关系
> 
> 会是硬件问题吗？今天刚入手，准备做一个旁路由

你先编译一个 最最基本的 来看 不就知道了，如果还不行 用官方的 试试，肯定就能知道是不是硬件问题了

Diciya, 2024-02-17:
最新驱动rss仿佛失效了

Lasx, 2024-07-04:
同8125b最近经常崩溃。。

coolsnowwolf, 2026-09-05:
已核对。该问题是 2024-01 的 R66S + r8125-dkms 9.012.03 日志，核心是 PCIe link down 后读取到全 0xffff 寄存器，属于网卡/PCIe 链路或硬件状态异常；评论也确认关闭 Full Cone NAT 后可能恢复，并指出 PCIe 信号问题。当前 LEDE 的 r8125 包已是更新版本，且没有可由该回溯定位的通用修复补丁。请用最小配置、另一条 PCIe 链路/官方固件及当前驱动复测后再报告，本条关闭为 not planned。

closed 2026-09-05 by coolsnowwolf