coolsnowwolf/lede #11845: perl-html-parser编译失败

tedcriss, 2024-01-30:
### 详细叙述

直接运行```make -j1```和运行```make package/feeds/packages/perl-html-parser/{clean,compile} -j1 V=s```都提示编译失败，怎么搞啊
好像有类似的issue，但是不知道怎么解决，故再次提一下

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

target=x86

### 详细日志

```
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'libX11', which does not exist
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'libXext', which does not exist
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'tslib', which does not exist
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'directfb', which does not exist
make[2]: Entering directory '/home/xuyuntao/lede86NN/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/xuyuntao/lede86NN/scripts/config'
make[1]: Entering directory '/home/xuyuntao/lede86NN'
make[2]: Entering directory '/home/xuyuntao/lede86NN/feeds/packages/lang/perl-html-parser'
rm -rf /home/xuyuntao/lede86NN/build_dir/target-x86_64_musl/perl/HTML-Parser-3.75
rm -f /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/stamp/.perl-html-parser_installed
rm -f /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/packages/perl-html-parser.list
make[2]: Leaving directory '/home/xuyuntao/lede86NN/feeds/packages/lang/perl-html-parser'
time: package/feeds/packages/perl-html-parser/clean#0.06#0.73#0.78
make[1]: Leaving directory '/home/xuyuntao/lede86NN'
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'libX11', which does not exist
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'libXext', which does not exist
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'tslib', which does not exist
WARNING: Makefile 'package/feeds/video/qt4/Makefile' has a dependency on 'directfb', which does not exist
make[2]: Entering directory '/home/xuyuntao/lede86NN/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/xuyuntao/lede86NN/scripts/config'
make[1]: Entering directory '/home/xuyuntao/lede86NN'
make[2]: Entering directory '/home/xuyuntao/lede86NN/package/libs/toolchain'
echo "libc" >> /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libatomic" >> /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libstdcpp" >> /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
echo "librt" >> /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/xuyuntao/lede86NN/package/libs/toolchain'
time: package/libs/toolchain/compile#0.07#0.78#0.84
make[2]: Entering directory '/home/xuyuntao/lede86NN/package/libs/zlib'
echo "zlib" >> /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/pkginfo/zlib.default.install
make[2]: Leaving directory '/home/xuyuntao/lede86NN/package/libs/zlib'
time: package/libs/zlib/compile#0.12#0.98#1.02
make[2]: Entering directory '/home/xuyuntao/lede86NN/feeds/packages/libs/libxml2'
make[2]: Leaving directory '/home/xuyuntao/lede86NN/feeds/packages/libs/libxml2'
time: package/feeds/packages/libxml2/compile#0.04#1.48#1.41
make[2]: Entering directory '/home/xuyuntao/lede86NN/feeds/packages/libs/db47'
make[2]: Leaving directory '/home/xuyuntao/lede86NN/feeds/packages/libs/db47'
time: package/feeds/packages/db47/compile#0.09#1.39#1.40
make[2]: Entering directory '/home/xuyuntao/lede86NN/feeds/packages/libs/gdbm'
make[2]: Leaving directory '/home/xuyuntao/lede86NN/feeds/packages/libs/gdbm'
time: package/feeds/packages/gdbm/compile#0.06#0.89#0.88
make[2]: Entering directory '/home/xuyuntao/lede86NN/feeds/packages/lang/perl'
make[2]: Leaving directory '/home/xuyuntao/lede86NN/feeds/packages/lang/perl'
time: package/feeds/packages/perl/host-compile#0.18#1.93#1.92
make[2]: Entering directory '/home/xuyuntao/lede86NN/feeds/packages/lang/perl'
echo "perl" >> /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/pkginfo/perl.default.install
echo "perlbase-config" >> /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/pkginfo/perl.default.install
echo "perlbase-essential" >> /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/pkginfo/perl.default.install
echo "perlbase-xsloader" >> /home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/pkginfo/perl.default.install
make[2]: Leaving directory '/home/xuyuntao/lede86NN/feeds/packages/lang/perl'
[... 76 lines cut ...]
/home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/usr/lib/perl5/5.28/CORE/perl.h:2494:22: note: in definition of macro 'Off_t'
 2494 | #       define Off_t off64_t
      |                      ^~~~~~~
/home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/usr/lib/perl5/5.28/CORE/perl.h:2494:22: error: unknown type name 'off64_t'; did you mean 'off_t'?
 2494 | #       define Off_t off64_t
      |                      ^~~~~~~
/home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/usr/lib/perl5/5.28/CORE/perl.h:2494:22: note: in definition of macro 'Off_t'
 2494 | #       define Off_t off64_t
      |                      ^~~~~~~
/home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/usr/lib/perl5/5.28/CORE/perl.h:2494:22: error: unknown type name 'off64_t'; did you mean 'off_t'?
 2494 | #       define Off_t off64_t
      |                      ^~~~~~~
/home/xuyuntao/lede86NN/staging_dir/target-x86_64_musl/usr/lib/perl5/5.28/CORE/perl.h:2494:22: note: in definition of macro 'Off_t'
 2494 | #       define Off_t off64_t
      |                      ^~~~~~~
make[3]: *** [Makefile:356: Parser.o] Error 1
make[3]: Leaving directory '/home/xuyuntao/lede86NN/build_dir/target-x86_64_musl/perl/HTML-Parser-3.75'
make[2]: *** [Makefile:66: /home/xuyuntao/lede86NN/build_dir/target-x86_64_musl/perl/HTML-Parser-3.75/.built] Error 2
make[2]: Leaving directory '/home/xuyuntao/lede86NN/feeds/packages/lang/perl-html-parser'
time: package/feeds/packages/perl-html-parser/compile#0.28#1.57#1.86
    ERROR: package/feeds/packages/perl-html-parser failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/perl-html-parser/compile] Error 1
make[1]: Leaving directory '/home/xuyuntao/lede86NN'
make: *** [/home/xuyuntao/lede86NN/include/toplevel.mk:231: package/feeds/packages/perl-html-parser/compile] Error 2
```

WYC-2020, 2024-01-30:
[image]
如此修复 D_LARGEFILE64_SOURCE

tedcriss, 2024-01-30:
解决了这个，后面还有一个perl的包编译也会报错，放弃了，回老版本了

closed 2024-01-30 by tedcriss

woni928, 2024-01-31:
> 解决了这个，后面还有一个perl的包编译也会报错，放弃了，回老版本了

云编译怎么解决啊？指教下 谢谢