coolsnowwolf/lede #11846: 6.1内核下quectel-MHI编译失败

xosadmin, 2024-01-30:
### 详细叙述

在6.1内核下quectel-MHI的pcie_mhi编译失败，使用5.5内核编译正常。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

photonicat

### 详细日志

make -C "/home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.75" ARCH="arm64" CROSS_COMPILE="aarch64-openwrt-linux-musl-" CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/pcie_mhi-3.2=pcie_mhi-3.2 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" M="/home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/pcie_mhi-3.2"  modules
make[4]: Entering directory '/home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.75'
  CC [M]  /home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/pcie_mhi-3.2/core/mhi_init.o
  CC [M]  /home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/pcie_mhi-3.2/core/mhi_main.o
  CC [M]  /home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/pcie_mhi-3.2/core/mhi_pm.o
  CC [M]  /home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/pcie_mhi-3.2/core/mhi_boot.o
/home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/pcie_mhi-3.2/core/mhi_boot.c: In function '__mhi_download_rddm_in_panic':
/home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/pcie_mhi-3.2/core/mhi_boot.c:203:23: error: implicit declaration of function 'prandom_u32'; did you mean 'prandom_u32_max'? [-Werror=implicit-function-declaration]
  203 |         sequence_id = prandom_u32() & BHIE_RXVECSTATUS_SEQNUM_BMSK;
      |                       ^~~~~~~~~~~
      |                       prandom_u32_max
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: /home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/pcie_mhi-3.2/core/mhi_boot.o] Error 1
make[4]: *** [Makefile:2014: /home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/pcie_mhi-3.2] Error 2
make[4]: Leaving directory '/home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-6.1.75'
make[3]: *** [Makefile:50: /home/user/photonicat-lean/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/pcie_mhi-3.2/.built] Error 2
make[3]: Leaving directory '/home/user/photonicat-lean/package/wwan/driver/quectel_MHI'
time: package/wwan/driver/quectel_MHI/compile#4.54#0.53#5.14
    ERROR: package/wwan/driver/quectel_MHI failed to build.
make[2]: *** [package/Makefile:116: package/wwan/driver/quectel_MHI/compile] Error 1
make[2]: Leaving directory '/home/user/photonicat-lean'
make[1]: *** [package/Makefile:110: /home/user/photonicat-lean/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/user/photonicat-lean'
make: *** [/home/user/photonicat-lean/include/toplevel.mk:231: world] Error 2

WYC-2020, 2024-01-30:
尴尬了我x86 试了一下 没错 有不想编译 你那个平台 哈哈

WYC-2020, 2024-01-31:
刚亲测了和你相同的平台和内核 没有任何问题 
[image]

xosadmin, 2024-01-31:
好的，我更新下源码再试试看

xosadmin, 2024-01-31:
> 刚亲测了和你相同的平台和内核 没有任何问题 [...]

我更新了源码，编译还是存在相同的问题。请问这和编译环境有关系吗？请问您的编译环境是？

suizhenmin, 2024-02-11:
卡在了同一个地方，同样的报错

xosadmin, 2024-02-13:
> 在隔壁找到了原因
> 
> sudo apt install clang

您最后编译成功了吗？我安装了这个包后编译光影猫的固件，仍然存在相同的问题。

suizhenmin, 2024-02-15:
> > 在隔壁找到了原因
> > sudo apt install clang
> 
> 您最终编译成功了吗？我安装了这个包后编译了photonicat的固件，仍然存在相同的问题。

我安装后编译成功了

closed 2024-02-18 by xosadmin