coolsnowwolf/lede #11847: perl-xml-parser 这两天一直报错

sxml, 2024-01-30:
### 详细叙述

小白forks别人的云编译 也麻烦 每次更新不是这报错就那报错 这又是什么 怎么弄呢？卡在这编译不过
ERROR: package/feeds/packages/perl-xml-parser failed to build

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ARM64

### 详细日志

[image]

suizhenmin, 2024-01-30:
编译同样的错误

WYC-2020, 2024-01-31:
稍微长点心都不会问这种重复issue问题

sxml, 2024-01-31:
> issue

大佬是指 另外#11845 issue 里面你说的修复 D_LARGEFILE64_SOURCE ？
哭 小白云编译不会修复

suizhenmin, 2024-01-31:
> 稍微长点心都不会问这种重复问题

之前编译一直正常，就这2次更新后才报错了，小白确实不懂很多

coolsnowwolf, 2026-09-05:
该 issue 仅提供“package/feeds/packages/perl-xml-parser failed to build”截图，没有首个编译错误、构建命令、工具链版本或完整日志，无法定位问题。当前 packages 中 perl-xml-parser 2.46 的 Expat 路径配置和 host/target 构建规则完整，且工作树与上游对应实现一致。请在当前树提供完整 `make package/feeds/packages/perl-xml-parser/{clean,compile} V=s` 首个错误后再分析；本 issue 关闭。

closed 2026-09-05 by coolsnowwolf