coolsnowwolf/lede #11848: 安装依赖报错

dingshaohua-cn, 2024-01-30:
### 详细叙述

[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

debian12

### 详细日志

```
root@debian:~# apt install -y ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential \
> bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext gcc-multilib g++-multilib \
> git gperf haveged help2man intltool libc6-dev-i386 libelf-dev libfuse-dev libglib2.0-dev libgmp3-dev \
> libltdl-dev libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libpython3-dev libreadline-dev \
> libssl-dev libtool lrzsz mkisofs msmtp ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 \
> python3-pyelftools python3-setuptools qemu-utils rsync scons squashfs-tools subversion swig texinfo \
> uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'genisoimage' instead of 'mkisofs'
ack is already the newest version (3.4.0-1).
bzip2 is already the newest version (1.0.8-5+b1).
cpio is already the newest version (2.13+dfsg-7.1).
p7zip is already the newest version (16.02+dfsg-8).
p7zip set to manually installed.
p7zip-full is already the newest version (16.02+dfsg-8).
p7zip-full set to manually installed.
python3 is already the newest version (3.11.2-1+b1).
python3 set to manually installed.
unzip is already the newest version (6.0-28).
unzip set to manually installed.
wget is already the newest version (1.21.3-1+b2).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 curl : Depends: libcurl4 (= 7.74.0-1.3+deb11u11) but 7.88.1-10+deb12u5~bpo11+1 is to be installed
 g++ : Depends: cpp (= 4:10.2.1-1) but 4:12.2.0-3 is to be installed
 g++-multilib : Depends: cpp (= 4:10.2.1-1) but 4:12.2.0-3 is to be installed
                Depends: g++-10-multilib (>= 10.2.1-3~) but it is not going to be installed
 gcc : Depends: cpp (= 4:10.2.1-1) but 4:12.2.0-3 is to be installed
 gcc-multilib : Depends: cpp (= 4:10.2.1-1) but 4:12.2.0-3 is to be installed
                Depends: gcc-10-multilib (>= 10.2.1-3~) but it is not going to be installed
 libc6-dev : Depends: libc6 (= 2.31-13+deb11u7) but 2.36-9+deb12u3 is to be installed
             Depends: libc-dev-bin (= 2.31-13+deb11u7) but it is not going to be installed
             Depends: libcrypt-dev but it is not going to be installed
             Depends: libnsl-dev but it is not going to be installed
 libc6-i386 : Depends: libc6 (= 2.31-13+deb11u7) but 2.36-9+deb12u3 is to be installed
 libelf-dev : Depends: libelf1 (= 0.183-1) but 0.188-2.1 is to be installed
 libexpat1-dev : Depends: libexpat1 (= 2.2.10-2+deb11u5) but 2.5.0-1 is to be installed
 libfuse-dev : Depends: libfuse2 (= 2.9.9-5) but 2.9.9-6+b1 is to be installed
               Depends: libselinux-dev
 libglib2.0-dev : Depends: libglib2.0-0 (= 2.66.8-1) but 2.74.6-2 is to be installed
                  Depends: libglib2.0-bin (= 2.66.8-1) but 2.74.6-2 is to be installed
                  Depends: libglib2.0-dev-bin (= 2.66.8-1) but it is not going to be installed
                  Depends: libmount-dev (>= 2.35.2-7~) but it is not going to be installed
                  Depends: libselinux1-dev but it is not going to be installed
 libgmp-dev : Depends: libgmp10 (= 2:6.2.1+dfsg-1+deb11u1) but 2:6.2.1+dfsg1-1.1 is to be installed
 libltdl-dev : Depends: libltdl7 (= 2.4.6-15) but 2.4.7-5 is to be installed
 libmpc-dev : Depends: libmpc3 (= 1.2.0-1) but 1.3.1-1 is to be installed
 libmpfr-dev : Depends: libmpfr6 (= 4.1.0-3) but 4.2.0-1 is to be installed
 libncurses5-dev : Depends: libtinfo6 (= 6.2+20201114-2+deb11u2) but 6.4-4 is to be installed
                   Depends: libncurses-dev (= 6.2+20201114-2+deb11u2) but it is not going to be installed
 libncursesw5-dev : Depends: libtinfo6 (= 6.2+20201114-2+deb11u2) but 6.4-4 is to be installed
                    Depends: libncurses-dev (= 6.2+20201114-2+deb11u2) but it is not going to be installed
 libreadline-dev : Depends: libreadline8 (= 8.1-1) but 8.2-1.3 is to be installed
                   Depends: libncurses-dev but it is not going to be installed
 python3-setuptools : Depends: python3-pkg-resources (= 52.0.0-4) but 66.1.1-1 is to be installed
                      Depends: python3-distutils but it is not going to be installed
 texinfo : Depends: perlapi-5.32.0
           Depends: libxml-libxml-perl but it is not going to be installed
 vim : Depends: vim-common (= 2:8.2.2434-3+deb11u1) but 2:9.0.1378-2 is to be installed
 zlib1g-dev : Depends: zlib1g (= 1:1.2.11.dfsg-2+deb11u2) but 1:1.2.13.dfsg-1 is to be installed
E: Unable to correct problems, you have held broken packages.
```

dingshaohua-cn, 2024-01-30:
=====通过安装的=====
ack 
antlr3 
asciidoc autoconf automake 
autopoint binutils bison 
bzip2 ccache cmake
cpio device-tree-compiler 
fastjar flex gawk gettext 
git gperf haveged help2man intltool 
uglifyjs upx-ucl unzip
wget curl  xxd xmlto

 

=====安装报错的=====
libltdl-dev libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libpython3-dev libreadline-dev \
libssl-dev libtool lrzsz mkisofs msmtp ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 \
python3-pyelftools python3-setuptools qemu-utils rsync scons squashfs-tools subversion swig texinfo \
build-essential  

gcc-multilib g++-multilib
libc6-dev-i386 libelf-dev libfuse-dev libglib2.0-dev libgmp3-dev
vim  zlib1g-dev

apt-get remove: 卸载软件
apt-get purge: 卸载软件和配置文件
apt-get autoremove: 移除没有使用的依赖
apt-get clean: 删除/var/cache/apt/archives/下的所有deb包
apt-get autoclean: 删除/var/cache/apt/archives/下的过期deb包

curl依赖冲突问题？
apt install -y curl
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 curl : Depends: libcurl4 (= 7.64.0-4+deb10u2) but 7.88.1-10+deb12u5~bpo11+1 is to be installed
E: Unable to correct problems, you have held broken packages.

执行 apt purge libcurl4 后，重新安装

vim安装失败
root@debian:~# apt install vim -y
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 ncurses-base : Breaks: vim-common (< 2:9.0.1000-2) but 2:8.1.0875-5+deb10u2 is to be installed
E: Error, pkgProblemResolver::Resolve generated breaks, this may be caused by held packages.

apt purge vim-common
apt update
apt upgrade
apt install vim

gcc-multilib安装失败
root@debian:~# apt install -y gcc-multilib
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 gcc-multilib : Depends: cpp (= 4:8.3.0-1) but 4:12.2.0-3 is to be installed
                Depends: gcc (= 4:8.3.0-1) but it is not going to be installed
                Depends: gcc-8-multilib (>= 8.3.0-1~) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.

服了啊 全是依赖报错，解决一个除了10个。

rabbit-dayi, 2024-02-12:
用docker吧
```bash
apt install docker-compose -y
docker run --name tmper_docker -itd -v /home/ubuntu/openwrt_workdir:/app ubuntu:22.04 bash
docker attach tmper_docker
apt install update
apt install sudo 
chmod +777 /app
apt install -y ack antlr3 asciidoc auto ....and
adduser XXX
su XXX
cd /app

./scripts/feed ...
make ...
```

wbz93815, 2024-04-28:
我也遇到了这个错误。话说有人知道这包是干啥的吗？

whatawhiz, 2024-05-05:
debian12的问题 
1 mkisofs会换成 genisoimage
2 python2.7 不在apt库里
3 upx-ucl 也是
这三个包没法安装

coolsnowwolf, 2026-09-05:
日志显示系统软件源混用了 Debian 11（bullseye）与 Debian 12（bookworm）/backports：例如 libc6 2.31 与 2.36、gcc-10 与 gcc-12、libcurl 版本不一致，属于 APT 源配置/发行版混装，不是 LEDE 编译依赖定义问题。应清理错误源并统一 Debian 12 源后再安装依赖。该旧环境问题已关闭。

closed 2026-09-05 by coolsnowwolf