coolsnowwolf/lede #11849: 广告屏蔽大师 Plus+ 只要启用  网络就断了。

caixin1218, 2024-01-30:
### 详细叙述

广告屏蔽大师 Plus+ 只要启用  网络就断了。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

广告屏蔽大师 Plus+ 只要启用   网络就断了。

WYC-2020, 2024-01-31:
dnsmasq都启动失败了还想能上网 咋可能

crow4586, 2024-01-31:
同样问题，经过反复尝试看起来不像冲突，像是插件方面问题。目前只能关闭它了。

caixin1218, 2024-01-31:
> dnsmasq都启动失败了还想能上网 咋可能

是的，关了就好了。
Wed Jan 31 19:28:47 2024 daemon.info procd: Instance dnsmasq::cfg01411c s in a crash loop 6 crashes, 0 seconds since last crash
Wed Jan 31 19:29:45 2024 cron.err crond[15848]: crond (busybox 1.36.0) started, log level 5

HelloFy, 2024-02-01:
> 这个插件放弃吧 这个新版的规则写的有问题的，需要把那两行去掉dnsmasq才能起来，而且运行起来也不咋地，不如AdGuardHome舒服。

不过说实话，dnsmasq也有问题，一行规则有问题跳过 log warn就好了，直接退出启动也是绝了。

WYC-2020, 2024-02-01:
> > 这个插件放弃吧 这个新版的规则写的有问题的，需要把那两行去掉dnsmasq才能起来，而且运行起来也不咋地，不如AdGuardHome舒服。
> 
> 不过说实话，dnsmasq也有问题，一行规则有问题跳过 log warn就好了，直接退出启动也是绝了。

这个改下源码即可

caixin1218, 2024-02-01:
> 这个插件放弃吧 这个新版的规则写的有问题的，需要把那两行去掉dnsmasq才能起来，而且运行起来也不咋地，不如AdGuardHome舒服。

AdGuardHome同样问题

HelloFy, 2024-02-02:
> > 这个插件放弃吧 这个新版的规则写的有问题的，需要把那两行去掉dnsmasq才能起来，而且运行起来也不咋地，不如AdGuardHome舒服。
> 
> AdGuardHome同样问题

adguardhome跟dnsmasq也没啥关系，agh自己可以代理dns 直接监听53加个anti-ad规则非常好用

caixin1218, 2024-02-02:
> > > 这个插件放弃吧 这个新版的规则写的有问题的，需要把那两行去掉dnsmasq才能起来，而且运行起来也不咋地，不如AdGuardHome舒服。
> > 
> > 
> > AdGuardHome同样问题
> 
> adguardhome跟dnsmasq也没啥关系，agh自己可以代理dns 直接监听53加个anti-ad规则非常好用

adguardhome可以启动，但网络还是有问题。dnsmasq肯定要修，很多问题都是它导致的，算是个bug

WYC-2020, 2024-02-02:
> > > > 这个插件放弃吧 这个新版的规则写的有问题的，需要把那两行去掉dnsmasq才能起来，而且运行起来也不咋地，不如AdGuardHome舒服。
> > > 
> > > 
> > > AdGuardHome同样问题
> > 
> > 
> > adguardhome跟dnsmasq也没啥关系，agh自己可以代理dns 直接监听53加个anti-ad规则非常好用
> 
> adguardhome可以启动，但网络还是有问题。dnsmasq肯定要修，很多问题都是它导致的，算是个bug

看了 是规则有问题，导致 替换出现的问题而已  明显 这里 ||clkindsstt.com^m^ 替换后变成了
[image]
肯定 dnsmasq不认了，稍微修复一下即可，至于dnsmasq 可以提高他的兼容性 让他发现有错误的 直接跳过，打印告警即可，等会改下

WYC-2020, 2024-02-02:
[image]
[912-skipping-wrong-addresses-causes-service-to-exit.patch](https://github.com/coolsnowwolf/lede/files/14138505/912-skipping-wrong-addresses-causes-service-to-exit.patch)
拿去吧

WYC-2020, 2024-02-02:
> ![image](https://private-user-images.githubusercontent.com/34025388/301807225-5e86194b-54b1-44f5-8a1b-c1d5633b1a97.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

考虑到可能还有其他的格式我就没有去改广告大师的那个正则脚本，因为根本就不能预测到上有会出现什么格式的玩意，直接dnsmasq兼容这种非法的 然后给提示，你们自己去上游反馈一下应该就行了

caixin1218, 2024-02-02:
> 考虑到可能还有其他的格式我就没有去改广告大师的那个正则脚本，因为根本就不能预测到上有会出现什么格式的玩意，直接dnsmasq兼容这种非法的 然后给提示，你们自己去上游反馈一下应该就行了

这只是权宜之计，dnsmasq肯定要修复并提高兼容性，不然还会其它的app也会出现各种问题。还有好多我还没有测试出来而已。

crow4586, 2024-02-02:
编译时候取消该插件，加入 AdGuard Home ，调试了两天发现这个才是王道，我用它配合PASSWALL，目前使用体验绝佳。

xingshu1990, 2024-02-03:
anti-ad 以及国内很多广告屏蔽规则，会屏蔽tracker服务器，导致很多小姐姐的视频下载不了。

> > > 这个插件放弃吧 这个新版的规则写的有问题的，需要把那两行去掉dnsmasq才能起来，而且运行起来也不咋地，不如AdGuardHome舒服。
> > 
> > 
> > AdGuardHome同样问题
> 
> adguardhome跟dnsmasq也没啥关系，agh自己可以代理dns 直接监听53加个anti-ad规则非常好用

anti-ad 以及国内很多广告屏蔽规则，会屏蔽tracker服务器，导致很多小姐姐的视频下载不了。使用这些规则，得自行找tracker链接，写白名单。

caixin1218, 2024-02-03:
> 编译时候取消该插件，加入 AdGuard Home ，调试了两天发现这个才是王道，我用它配合PASSWALL，目前使用体验绝佳。

AdGuard Home 现在也是有问题的

closed 2024-02-03 by caixin1218

ali2liu, 2024-02-07:
> ![image](https://private-user-images.githubusercontent.com/34025388/301807225-5e86194b-54b1-44f5-8a1b-c1d5633b1a97.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

我fork了lean大的源码，在Action里远程编译，要把这个patch放到哪里才能用呢？

nb12nb34, 2024-02-14:
> > > > ![image](https://private-user-images.githubusercontent.com/34025388/301807225-5e86194b-54b1-44f5-8a1b-c1d5633b1a97.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY4Njg2MTksIm5iZiI6MTcwNjg2ODMxOSwicGF0aCI6Ii8zNDAyNTM4OC8zMDE4MDcyMjUtNWU4NjE5NGItNTRiMS00NGY1LThhMWItYzFkNTYzM2IxYTk3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjAyVDEwMDUxOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTZlMTNiMjZmM2FkN2U2MWFlM2RjMWFkOTNmN2JkM2UyYTNjZGVlNjNlMzI0MDdmZWZlM2I3ZTk2YzJkZDU2OWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4GWF2UieHyr5EQPFlzHJD6S9tksoQW7vkfZGAScgqSQ) [912-skipping-wrong-addresses-causes-service-to-exit.patch](https://github.com/coolsnowwolf/lede/files/14138505/912-skipping-wrong-addresses-causes-service-to-exit.patch) 拿去吧
> > > 
> > > 
> > > 考虑到可能还有其他的格式我就没有去改广告大师的那个正则脚本，因为根本就不能预测到上有会出现什么格式的玩意，直接dnsmasq兼容这种非法的 然后给提示，你们自己去上游反馈一下应该就行了
> > 
> > 
> > 这只是权宜之计，dnsmasq肯定要修复并提高兼容性，不然还会其它的app也会出现各种问题。还有好多我还没有测试出来而已。
> 
> 放心，这对于dnsmasq作者来说这就不是bug，你自己不按别人的规则来写，作者写成退出服务你也不能说他有问题，针对所以不符合规则的就让服务不起来，有的人起来了又会说为啥我加的规则不生效呢？作者才不想一天天给你们分析这种杂七杂八的，所以直接一刀砍，不按我要求来的直接就启动不了，这样你就知道有问题了

用的在线编译，这个patch 要怎么用呢。求大佬指教