coolsnowwolf/lede #11852: vsftpd 不能编译i386平台的

yelxl, 2024-01-31:
### 详细叙述

make[3]: Entering directory '/home/lu/lede/package/lean/vsftpd-alt'
rm -f /home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5/.built
touch /home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5/.built_check
/home/lu/lede/staging_dir/host/bin/sed -i -e 's/-lcrypt -lnsl/-lcrypt/' /home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5/Makefile
make -C /home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5 CC="i486-openwrt-linux-musl-gcc" CFLAGS="-Os -pipe -march=pentium-mmx -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5=vsftpd-3.0.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -flto -D_GNU_SOURCE -include fcntl.h" LDFLAGS="-L/home/lu/lede/staging_dir/toolchain-i386_pentium-mmx_gcc-11.3.0_musl/usr/lib -L/home/lu/lede/staging_dir/toolchain-i386_pentium-mmx_gcc-11.3.0_musl/lib -znow -zrelro -Wl,--gc-sections,--as-needed,--no-undefined,--no-allow-shlib-undefined" vsftpd
make[4]: Entering directory '/home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5'
i486-openwrt-linux-musl-gcc -c ptracesandbox.c -Os -pipe -march=pentium-mmx -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5=vsftpd-3.0.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -flto -D_GNU_SOURCE -include fcntl.h -idirafter dummyinc
ptracesandbox.c: In function 'validate_fcntl':
ptracesandbox.c:857:15: error: 'F_SETLK64' undeclared (first use in this function); did you mean 'F_SETLK'?
  857 |       arg2 != F_SETLK64 &&
      |               ^~~~~~~~~
      |               F_SETLK
ptracesandbox.c:857:15: note: each undeclared identifier is reported only once for each function it appears in
ptracesandbox.c:858:15: error: 'F_SETLKW64' undeclared (first use in this function); did you mean 'F_SETLKW'?
  858 |       arg2 != F_SETLKW64 &&
      |               ^~~~~~~~~~
      |               F_SETLKW
make[4]: *** [Makefile:28: ptracesandbox.o] Error 1
make[4]: Leaving directory '/home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5'
make[3]: *** [Makefile:102: /home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5/.built] Error 2
make[3]: Leaving directory '/home/lu/lede/package/lean/vsftpd-alt'
time: package/lean/vsftpd-alt/compile#0.30#0.13#0.59
    ERROR: package/lean/vsftpd-alt failed to build.
make[2]: *** [package/Makefile:116: package/lean/vsftpd-alt/compile] Error 1
make[2]: Leaving directory '/home/lu/lede'
make[1]: *** [package/Makefile:110: /home/lu/lede/staging_dir/target-i386_pentium-mmx_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lu/lede'
make: *** [/home/lu/lede/include/toplevel.mk:231：world] 错误 2

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

intel 774平台

### 详细日志

make[3]: Entering directory '/home/lu/lede/package/lean/vsftpd-alt'
rm -f /home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5/.built
touch /home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5/.built_check
/home/lu/lede/staging_dir/host/bin/sed -i -e 's/-lcrypt -lnsl/-lcrypt/' /home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5/Makefile
make -C /home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5 CC="i486-openwrt-linux-musl-gcc" CFLAGS="-Os -pipe -march=pentium-mmx -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5=vsftpd-3.0.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -flto -D_GNU_SOURCE -include fcntl.h" LDFLAGS="-L/home/lu/lede/staging_dir/toolchain-i386_pentium-mmx_gcc-11.3.0_musl/usr/lib -L/home/lu/lede/staging_dir/toolchain-i386_pentium-mmx_gcc-11.3.0_musl/lib -znow -zrelro -Wl,--gc-sections,--as-needed,--no-undefined,--no-allow-shlib-undefined" vsftpd
make[4]: Entering directory '/home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5'
i486-openwrt-linux-musl-gcc -c ptracesandbox.c -Os -pipe -march=pentium-mmx -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5=vsftpd-3.0.5 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections -flto -D_GNU_SOURCE -include fcntl.h -idirafter dummyinc
ptracesandbox.c: In function 'validate_fcntl':
ptracesandbox.c:857:15: error: 'F_SETLK64' undeclared (first use in this function); did you mean 'F_SETLK'?
  857 |       arg2 != F_SETLK64 &&
      |               ^~~~~~~~~
      |               F_SETLK
ptracesandbox.c:857:15: note: each undeclared identifier is reported only once for each function it appears in
ptracesandbox.c:858:15: error: 'F_SETLKW64' undeclared (first use in this function); did you mean 'F_SETLKW'?
  858 |       arg2 != F_SETLKW64 &&
      |               ^~~~~~~~~~
      |               F_SETLKW
make[4]: *** [Makefile:28: ptracesandbox.o] Error 1
make[4]: Leaving directory '/home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5'
make[3]: *** [Makefile:102: /home/lu/lede/build_dir/target-i386_pentium-mmx_musl/vsftpd-alt-3.0.5/vsftpd-3.0.5/.built] Error 2
make[3]: Leaving directory '/home/lu/lede/package/lean/vsftpd-alt'
time: package/lean/vsftpd-alt/compile#0.30#0.13#0.59
    ERROR: package/lean/vsftpd-alt failed to build.
make[2]: *** [package/Makefile:116: package/lean/vsftpd-alt/compile] Error 1
make[2]: Leaving directory '/home/lu/lede'
make[1]: *** [package/Makefile:110: /home/lu/lede/staging_dir/target-i386_pentium-mmx_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/lu/lede'
make: *** [/home/lu/lede/include/toplevel.mk:231：world] 错误 2

yelxl, 2024-01-31:
Note that it only happens on i386 as ptracesandbox.c is only compiled on this architecture.
I just went ahead and did the quick fix, adding -D_LARGEFILE64_SOURCE to CFLAGS on x86.
找到相关的方法了，加了以后就通过编译了，不知道能不加到源码里去

WYC-2020, 2024-01-31:
这年头居然还有i386

coolsnowwolf, 2026-09-05:
该 issue 报告的 `F_SETLK64/F_SETLKW64` 未定义问题已由当前树提交 `c7b64067f23` 修复：musl 构建增加 `-D_LARGEFILE64_SOURCE`，并将 vsftpd-alt release 提升为 2；随后又补充了 glibc 兼容修复。当前 `package/lean/vsftpd-alt` 已不再是报告中的旧状态，无需新增修改，issue 关闭。

closed 2026-09-05 by coolsnowwolf