coolsnowwolf/lede #11854: GL AXT1800 无线驱动问题

King-Of-Knights, 2024-01-31:
### 详细叙述

编译AXT1800固件，测试工具为iperf3，环境分别为有线和无线，有线方面采用2.5G网卡，无线方面，路由器设置为5G 36信道 80MHz带宽，PC测试网卡为Intel(R) Wi-Fi 6 AX200 160MHz 2.5G，网络未加密，对比速度如下：
[image]
另外，修改AXT1800的无线设置会导致无线网络停止工作，重启才能打开无线网络

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

GL AXT1800

### 详细日志

无

aiamadeus, 2024-01-31:
iperf3 的 server 和 client 都需要在路由器下的设备上，server 跑在路由器上是不准的。另外 ipq60xx 没有新的 ath11k-fw，所以无线很不稳定。

King-Of-Knights, 2024-01-31:
@aiamadeus 首先感谢大佬回复了，没有新的ath11k-fw意思是说目前的驱动是开源里面最好的咯，我编译完了以后，测家里的千兆带宽，速度不到500m，感觉有点离谱了，请问是否有好的办法？
[image]

aiamadeus, 2024-01-31:
> 没有新的ath11k-fw意思是说目前的驱动是开源里面最好的咯

那肯定不是，还得修

King-Of-Knights, 2024-02-01:
更新一下，换了ath11K 2.4.0.1驱动以后，网速上去了一些，但还未达到理想值

[image]

DoveKi, 2024-02-03:
> 更新一下，换了ath11K 2.4.0.1驱动以后，网速上去了一些，但达到了理想值
> 
> ![Screenshot_2024-02-01-12-23-34-171_net 网络工具](https://private-user-images.githubusercontent.com/25562792/301513396-6cb51d60-e137-41f8-821d-cabec1934992.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY5MzEwMTgsIm5iZiI6MTcwNjkzMDcxOCwicGF0aCI6Ii8yNTU2Mjc5Mi8zMDE1MTMzOTYtNmNiNTFkNjAtZTEzNy00MWY4LTgyMWQtY2FiZWMxOTM0OTkyLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjAzVDAzMjUxOFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWZmMjM4NTI4MTc1YTRmYTNjMWQ5ZThlOTlhZDkwYTE2ODA0ZWQzNGFjYzhlMTZjNTE0NTUwNjM0ZmJjMDZjODEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.TZJcJqVwUFHl6zJ7Z_1mp-W-DZ8niJA3NzuEtQoF6pk)

服务端192.168.1.1这是iperf3服务端挂在路由器上吧
楼上不是跟你说了 iperf3服务端要挂在lan或wan下的设备

King-Of-Knights, 2024-02-04:
@DoveKi 大佬好呀，不是我不放这个结果，这个我试了，只有200-300M，惨不忍睹

> iperf3服务端要挂在lan或者wan下设备，别挂在路由器上

根据大佬所说，我去找了一点资料，ath11k的nss补丁是否是[这个](https://github.com/Ansuel/openwrt/commit/d3787bf95127beb0edc98f557feb68c68c46eefd)呀？

coolsnowwolf, 2026-09-05:
该 issue 只有吞吐对比截图，没有无线驱动日志、固件版本、关联速率/调制信息或修改配置后的失败日志，无法定位所谓“驱动问题”。当前 LEDE 已包含 GL-AXT1800 的完整 ipq60xx DTS、校准数据、镜像和升级支持（提交 `c3f2bb7afc7`），mt76/ath11k 也已持续更新。5 GHz 80 MHz 的实际吞吐受客户端能力、协议开销和环境影响，不能仅凭截图判定缺陷；请用当前版本并提供完整日志后再复现。issue 关闭。

closed 2026-09-05 by coolsnowwolf