coolsnowwolf/lede #11855: 同样的配置文件，升级5.4.268版本后内核一直报phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0

yflwz, 2024-01-31:
### 详细叙述

编译斐讯K3，内核版本从5.4.267更新到5.4.268后查看内核日志都是
[73416.673269] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73421.690361] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73436.709193] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73441.722792] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73456.749273] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73461.762804] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73476.787474] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73481.800387] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73496.827878] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

斐讯K3

### 详细日志

[73075.989246] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73081.002786] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73096.029455] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73101.043059] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73116.072725] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73121.082785] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73136.108861] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73141.122800] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73156.154372] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73161.172808] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73176.192737] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73181.212788] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73196.232733] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73201.252779] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73216.272742] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73221.292817] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73236.308960] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73241.322798] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73256.350060] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73261.362917] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73276.392736] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73281.402786] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73296.429648] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73301.442777] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73316.469192] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73321.482788] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73336.507713] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73341.522835] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73356.549131] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73361.562812] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73376.589486] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73381.602807] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73396.630368] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73401.642807] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73416.673269] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73421.690361] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73436.709193] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73441.722792] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73456.749273] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73461.762804] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73476.787474] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73481.800387] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0
[73496.827878] ieee80211 phy1: brcmf_p2p_send_action_frame: Unknown Frame: category 0x5, action 0x0

yflwz, 2024-01-31:
不知道和batman-adv有没有关系，因为batman-adv更新到2023.3后编译出错，所以我就换回2022.3版本编译

WYC-2020, 2024-02-01:
你这个感觉和这个很像 https://github.com/raspberrypi/linux/issues/4093

yflwz, 2024-02-01:
感觉比我这个复杂多了

yflwz, 2024-02-02:
我这二天重新编译了，在原来的配置去掉了easymesh，暂时没有这个问题

WYC-2020, 2024-02-18:
> 我这二天重新编译了，在原来的配置去掉了easymesh，暂时没有这个问题

那就好，这种牵扯给内核打补丁的插件一定要谨慎使用，像那个过滤什么插件 有的人用了也会出现各种各样奇葩的现象

coolsnowwolf, 2026-09-05:
已回移上游 brcmfmac 修复并推送：LEDE 提交 `45a7e69ccd84b0c3fe470350cb382e47cc6c4b73`。该补丁在 mac80211 的 6.18.39 及旧版 patch 层均适用，将启用 tracing 时被错误提升为 ERR 的 `brcmf_info()` 恢复为 INFO，避免 `Unknown Frame: category 0x5` 刷屏；补丁已对当前构建目录做 dry-run 验证。

closed 2026-09-05 by coolsnowwolf