coolsnowwolf/lede #11856: lede 23.05编译red mi ax6000出错

gbm777, 2024-02-01:
### 详细叙述

lede 23.05编译red mi ax6000内核6.6出错

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

redmi ax6000

### 详细日志

  MODPOST /home/gbm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/leds-ws2812b/Module.symvers
ERROR: modpost: "__spi_register_driver" [/home/gbm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/leds-ws2812b/leds-ws2812b.ko] undefined!
ERROR: modpost: "spi_sync" [/home/gbm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/leds-ws2812b/leds-ws2812b.ko] undefined!
scripts/Makefile.modpost:145: recipe for target '/home/gbm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/leds-ws2812b/Module.symvers' failed
make[6]: *** [/home/gbm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/leds-ws2812b/Module.symvers] Error 1
/home/gbm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.6.14/Makefile:1865: recipe for target 'modpost' failed
make[5]: *** [modpost] Error 2
Makefile:234: recipe for target '__sub-make' failed
make[4]: *** [__sub-make] Error 2
make[4]: Leaving directory '/home/gbm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/linux-6.6.14'Makefile:34: recipe for target '/home/gbm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/leds-ws2812b/.built' failed
make[3]: *** [/home/gbm/lede/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_filogic/leds-ws2812b/.built] Error 2
make[3]: Leaving directory '/home/gbm/lede/package/kernel/leds-ws2812b'
time: package/kernel/leds-ws2812b/compile#0.48#0.03#0.50
    ERROR: package/kernel/leds-ws2812b failed to build.
package/Makefile:114: recipe for target 'package/kernel/leds-ws2812b/compile' failed
make[2]: *** [package/kernel/leds-ws2812b/compile] Error 1
make[2]: Leaving directory '/home/gbm/lede'
package/Makefile:110: recipe for target '/home/gbm/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: *** [/home/gbm/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/gbm/lede'
/home/gbm/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

gbm777, 2024-02-02:
[config.zip](https://github.com/coolsnowwolf/lede/files/14132688/config.zip)

gbm777, 2024-02-02:
看上去是led驱动leds-ws2812b编译的问题，请问大佬指导一下，怎么操作下？

gbm777, 2024-02-02:
BTW, 5.15编译是成功的，改编译6.1和6.6时才出现这个报错

gbm777, 2024-02-02:
> ？不支持 6.6 内核

6.1也是一样的报错，不知道是不是少文件
5.15编译成功了

gbm777, 2024-02-02:
似乎重新 make clean 可以搞定6.1了
接下来再试试 6.6

WYC-2020, 2024-02-02:
什么玩意 不是那么适配的 一堆的patch 没得 你以为 linux 内核原生支持这玩意么，很多都是 需要打补丁的 不像x86 这种，原生支持，所以即使不打任何补丁也只可以直接改版本号 玩，问题都不会很大，但是 你这个 肯定不能这么玩的，等大雕他们搞吧，我是不想搞硬路由

gbm777, 2024-02-02:
恩山已经有人放出来了6.1内核的硬路由固件了（https://www.right.com.cn/forum/thread-8294637-1-1.html），所以玩玩看呗

WYC-2020, 2024-02-02:
> 恩山已经有人放出来了6.1内核的硬路由固件了（ [ https://www.right.com.cn/forum/thread-8294637-1-1.html ](url) ），所以玩玩看呗

这im仓库嘛，看他提交好像是另外两个人提交的，你有能力扣出他的那些提交，基于l仓库搞也可以

DoveKi, 2024-02-03:
谁跟你说换内核版本把版本号改了就可以编译了

closed 2024-02-06 by coolsnowwolf