coolsnowwolf/lede #11859: 启用Adbyby plus+后不能上网，DNS出问题

ali2liu, 2024-02-02:
### 详细叙述

系统日志：
Fri Feb  2 10:38:22 2024 cron.err crond[16477]: crond (busybox 1.36.0) started, log level 8
Fri Feb  2 10:38:29 2024 daemon.crit dnsmasq[1]: Bad address in --address at line 11865 of /tmp/dnsmasq.d/dnsmasq-adbyby.d/04-dnsmasq.adblock
Fri Feb  2 10:38:29 2024 daemon.crit dnsmasq[1]: FAILED to start up
Fri Feb  2 10:38:34 2024 daemon.crit dnsmasq[1]: Bad address in --address at line 11865 of /tmp/dnsmasq.d/dnsmasq-adbyby.d/04-dnsmasq.adblock
Fri Feb  2 10:38:34 2024 daemon.crit dnsmasq[1]: FAILED to start up

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

编译正常

schen39, 2024-02-02:
你确定没有类似的issue?https://github.com/coolsnowwolf/lede/issues/11849

ali2liu, 2024-02-02:
sory

closed 2024-02-02 by ali2liu