coolsnowwolf/lede #11861: 编译HC5611 mt7603e闭源驱动问题

gforce1206, 2024-02-03:
### 详细叙述

RT，极路由go（即HC5611），默认开源驱动，编译时尝试改为闭源驱动，make menuconfig选择

> Luci —>Applications—>luci-app-mtwifi

> Network —>Wireless —>mt_wifi

> Kernel modules —>Wireless Drivers —>kmod-mt7603e

取消> Network —>wirelessAPD—>wpad-openssl

> Kernel modules —>Wireless Drivers—>kmod-mt7603

> Kernel modules —>Wireless Drivers—>kmod-cfg80211

> Kernel modules —>Wireless Drivers—>kmod-mac80211
[image]

编译出的固件luci wifi界面变成这样，有两个wifi，都是mt7615
[image]

检查了一下插件库，确实选的是mt7063e闭源驱动
[image]

### 重复 issue

- [x] 没有类似的 issue

### 具体型号

HC5611

### 详细日志

无

blzadx, 2024-03-18:
这不是型号识别错误，是移植闭源驱动的时候图方便，加之luci-app-mtwifi管理面板缺了个datconf依赖，不能解析l1profile（MTK闭源驱动是靠两级profile进行配置的，你可以看看/etc/wireless里的文件，很好理解），干脆把所有的芯片都当作7615管理了。一般ra0就是2.4G芯片，rai0就是5G芯片。闭源驱动是正常工作的，管理面板也能够是正常生效，就只是看上去别扭。

coolsnowwolf, 2026-09-05:
该 issue 没有编译日志或生成的 `.config`，仅凭 LuCI 显示两个 mt7615 无法判断是否同时选中了目标 profile 的内置驱动。HC5611 的 mt7603e 闭源驱动位于 `package/lean/mt/drivers/mt7603e`，与主线 mt76 开源驱动是互斥的；应确保取消 kmod-mt7603/kmod-mac80211/cfg80211，并检查最终 config 与固件实际安装的模块。闭源驱动不在 OpenWrt 主线维护范围，当前没有可引用的上游修复。该旧问题因缺少可复现证据现关闭。

closed 2026-09-05 by coolsnowwolf