coolsnowwolf/lede #11863: 最新的x86编译失败

liuyanantj, 2024-02-03:
### 详细叙述

编译最新版的固件，但是一直提示失败  求助

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/tc/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/tc/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" /usr/bin/env bash ../../lang/golang//golang-build.sh cache_cleanup
touch /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.built
rm -rf /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.pkgdir/mosdns.installed /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.pkgdir/mosdns
mkdir -p /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.pkgdir/mosdns
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/tc/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/tc/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/IrineSistiana/mosdns" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/IrineSistiana/mosdns/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1" GO_BUILD_DIR="/home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.go_work/build" GO_BUILD_BIN_DIR="/home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/tc/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src" /usr/bin/env bash ../../lang/golang//golang-build.sh install_bin "/home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.pkgdir/mosdns"
touch /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.pkgdir/mosdns.installed
mkdir -p /home/tc/lede/staging_dir/target-x86_64_musl/root-x86/stamp
SHELL= flock /home/tc/lede/tmp/.root-copy.flock -c 'cp -fpR /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.pkgdir/mosdns/. /home/tc/lede/staging_dir/target-x86_64_musl/root-x86/'
touch /home/tc/lede/staging_dir/target-x86_64_musl/root-x86/stamp/.mosdns_installed
mkdir -p /home/tc/lede/bin/targets/x86/64/packages /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/ipkg-x86_64/mosdns/CONTROL /home/tc/lede/staging_dir/target-x86_64_musl/pkginfo
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/tc/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/tc/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/IrineSistiana/mosdns" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/IrineSistiana/mosdns/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1" GO_BUILD_DIR="/home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.go_work/build" GO_BUILD_BIN_DIR="/home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/tc/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src" /usr/bin/env bash ../../lang/golang//golang-build.sh install_bin "/home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/ipkg-x86_64/mosdns"
find /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/ipkg-x86_64/mosdns -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="x86_64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="x86_64-openwrt-linux-musl-nm" STRIP="x86_64-openwrt-linux-musl-strip --strip-all" STRIP_KMOD="/home/tc/lede/scripts/strip-kmod.sh" PATCHELF="/home/tc/lede/staging_dir/host/bin/patchelf" /home/tc/lede/scripts/rstrip.sh /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/ipkg-x86_64/mosdns
rstrip.sh: /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/ipkg-x86_64/mosdns/usr/bin/mosdns: executable
(cd /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/ipkg-x86_64/mosdns/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
install -d -m0755 /home/tc/lede/bin/packages/x86_64/packages
/home/tc/lede/staging_dir/host/bin/fakeroot /home/tc/lede/staging_dir/host/bin/bash /home/tc/lede/scripts/ipkg-build -m "" /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/ipkg-x86_64/mosdns /home/tc/lede/bin/packages/x86_64/packages
Packaged contents of /home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/ipkg-x86_64/mosdns into /home/tc/lede/bin/packages/x86_64/packages/mosdns_5.3.1-1_x86_64.ipk
echo "mosdns" >> /home/tc/lede/staging_dir/target-x86_64_musl/pkginfo/mosdns.default.install
rm -rf /home/tc/lede/tmp/stage-mosdns
mkdir -p /home/tc/lede/tmp/stage-mosdns/host /home/tc/lede/staging_dir/target-x86_64_musl/packages
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/tc/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/tc/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/IrineSistiana/mosdns" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/IrineSistiana/mosdns/..." GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1" GO_BUILD_DIR="/home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.go_work/build" GO_BUILD_BIN_DIR="/home/tc/lede/build_dir/target-x86_64_musl/mosdns-5.3.1/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/tc/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src" /usr/bin/env bash ../../lang/golang//golang-build.sh install_src "/home/tc/lede/tmp/stage-mosdns"
find /home/tc/lede/tmp/stage-mosdns -name '*.la' | xargs -r rm -f; 
if [ -f /home/tc/lede/staging_dir/target-x86_64_musl/packages/mosdns.list ]; then /home/tc/lede/scripts/clean-package.sh "/home/tc/lede/staging_dir/target-x86_64_musl/packages/mosdns.list" "/home/tc/lede/staging_dir/target-x86_64_musl"; fi
if [ -d /home/tc/lede/tmp/stage-mosdns ]; then (cd /home/tc/lede/tmp/stage-mosdns; find ./ > /home/tc/lede/tmp/stage-mosdns.files); 	SHELL= flock /home/tc/lede/tmp/.staging-dir.flock -c ' mv /home/tc/lede/tmp/stage-mosdns.files /home/tc/lede/staging_dir/target-x86_64_musl/packages/mosdns.list && cp -fpR /home/tc/lede/tmp/stage-mosdns/* /home/tc/lede/staging_dir/target-x86_64_musl/; '; fi
rm -rf /home/tc/lede/tmp/stage-mosdns
touch /home/tc/lede/staging_dir/target-x86_64_musl/stamp/.mosdns_installed
make[3]: Leaving directory '/home/tc/lede/feeds/packages/net/mosdns'
time: package/feeds/packages/mosdns/compile#30.20#17.73#37.74
make[3]: Entering directory '/home/tc/lede/feeds/helloworld/gn'
MAKEFLAGS="" /home/tc/lede/staging_dir/host/bin/ninja  -j1 -C /home/tc/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/out
ninja: Entering directory `/home/tc/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/out'
[1/287] CXX src/base/command_line.o
FAILED: src/base/command_line.o 
clang++ -MMD -MF src/base/command_line.o.d -I../src -I. -DNDEBUG -O3 -fdata-sections -ffunction-sections -Werror -D_FILE_OFFSET_BITS=64 -D__STDC_CONSTANT_MACROS -D__STDC_FORMAT_MACROS -pthread -pipe -fno-exceptions -fno-rtti -fdiagnostics-color -Wall -Wextra -Wno-unused-parameter -Wextra-semi -Wundef -std=c++20 -Wrange-loop-analysis -Wextra-semi-stmt -c ../src/base/command_line.cc -o src/base/command_line.o
/bin/sh: 1: clang++: not found
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:58: /home/tc/lede/build_dir/hostpkg/gn-2024-01-24-a2e2717e/.built] Error 1
make[3]: Leaving directory '/home/tc/lede/feeds/helloworld/gn'
time: package/feeds/helloworld/gn/host-compile#0.07#0.01#0.08
    ERROR: package/feeds/helloworld/gn [host] failed to build.
make[2]: *** [package/Makefile:114: package/feeds/helloworld/gn/host/compile] Error 1
make[2]: Leaving directory '/home/tc/lede'
make[1]: *** [package/Makefile:110: /home/tc/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tc/lede'
make: *** [/home/tc/lede/include/toplevel.mk:231：world] 错误 2

lirg-111, 2024-02-03:
没装clang?

liuyanantj, 2024-02-03:
> 没装clang?

怎么安装呢  小白 不太懂

binge8, 2024-02-03:
gn报错，gn也不是这个库里的啊。另外你这是最新源码？人家早就修复了[](https://github.com/kenzok8/small/issues/122)

caixin1218, 2024-02-03:
> > 没装clang?
> 
> 怎么安装呢 小白 不太懂
sudo apt update -y
sudo apt full-upgrade -y
sudo apt install -y clang

liuyanantj, 2024-02-04:
> gn报错，gn也不是这个库里的啊。另外你这是最新源码？人家早就修复了[看看这个吧](https://github.com/kenzok8/small/issues/122)

我全部删除  重新下载编译的  还是有问题  我晚上回家试试安装下clang试试看

chengjyang, 2024-02-04:
sudo apt-get install llvm clang

liuyanantj, 2024-02-04:
> > > 没装clang?
> > 
> > 
> > 怎么安装呢 小白 不太懂
> > sudo apt update -y
> > sudo apt full-upgrade -y
> > sudo apt install -y clang

已经解决 谢谢啦

liuyanantj, 2024-02-04:
> sudo apt-get install llvm clang

谢谢啦

closed 2024-02-04 by liuyanantj