coolsnowwolf/lede #11872: 编译出来的openwrt-x86-64-generic-squashfs-combined-efi.img.gz解压报错

free-diy, 2024-02-06:
### 详细叙述

在windows下用了几种解压工具，解压都报错，虽然能强行解压出来，但解压出来的文件有问题

[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

[image]

wi11iamzhao, 2024-02-06:
编译出来的并不是tar.gz，应该使用gzip/gunzip命令（似乎powershell并不直接支持）
[image]
Windows11下用7zip直接解压会报错，但是可以直接用任务管理器解压，没有出现问题（rufus直接写SSD和转换成VMDK都验证过）

free-diy, 2024-02-07:
> 编译出来的并不是tar.gz，应该使用gzip/gunzip命令（似乎powershell并不直接支持） Windows11下用7zip直接解压会报错，但是可以直接用任务管理器解压，没有出现问题（rufus直接写SSD和转换成VMDK都验证过） ![图像](https://private-user-images.githubusercontent.com/49196668/302688771-fbf06ea9-8729-4df6-b1ba-59b36b2442a8.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDcyNzg0MDAsIm5iZiI6MTcwNzI3ODEwMCwicGF0aCI6Ii80OTE5NjY2OC8zMDI2ODg3NzEtZmJmMDZlYTktODcyOS00ZGY2LWIxYmEtNTliMzZiMjQ0MmE4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjA3VDAzNTUwMFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ2YTdjYmI3MzIwZDA5ZjMxYTczZTkzNmNmODRhMzBjYmExOGM3ODg4YzMxZDRmMGMyNDMzMzA4ODQzYmRkOTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.4wR3R_gjTogj5zFRmbvLcTukJoXPZxlP4J94xDPx1WM)

我用任务管理器解压转换成VMDK磁盘报错

WYC-2020, 2024-02-18:
我这边正常哈哈 ，我怀疑是l这边有那个附加数据的锅 我直接 就去掉了那个附加数据的 一切正常

free-diy, 2024-02-21:
编译出来勾选压缩包的解压就有错

swjtuhyq, 2024-02-28:
我编译了两次，都不能解压，这个问题有没有解决办法？

free-diy, 2024-03-06:
这个问题这么难解决吗？

swjtuhyq, 2024-03-19:
我弄错了，是因为Winterm的BUG导致我下载的文件被修改了，所以无法解压，后来使用SCP下载回来就没有问题了

jonwhoo, 2024-04-28:
我碰到同一问题，发现是Winterm下载的文件出错，用WinSCP下载，就正常

a892728, 2025-01-31:
我也遇到差不多的
好像是7-zip的問題 不過要怎麼回報我不太會
是那個的問題也不確定
我是在Windows 7 x64 +7-zip 24.09
[https://imgur.com/a/WRObrZI](url)
.gz 你用gunzip解開就可以了
但是我Windows下gzip(Gzip for Windows)
也是一樣
我只是抓官方的編譯檔來用而己

coolsnowwolf, 2026-09-05:
已验证当前镜像生成流程：本地 `openwrt-x86-64-generic-squashfs-rootfs.img.gz` 通过 `gzip -t` 校验；`combined-efi.img.gz` 解压内容为有效 DOS/MBR 磁盘镜像。后者末尾带有 LEDE image metadata，因此 gzip 会提示 `trailing garbage ignored`，这是镜像元数据设计，不是压缩损坏。该文件应先用 gzip/gunzip 解压，再写盘；Windows 下载工具或部分 7-Zip/传输工具可能误报或破坏带元数据文件。Issue 已确认不是源码缺陷并关闭。

closed 2026-09-05 by coolsnowwolf