coolsnowwolf/lede #11873: 卡在libxslt 编译失败？

sxml, 2024-02-06:
### 详细叙述

云编译的是arm N1盒子 config没变 .  奇怪怎么编译会有x86_64呢？
一直卡在这libxslt 编译失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

arm64

### 详细日志

[image]

coolsnowwolf, 2024-02-09:
主系统缺包

sxml, 2024-02-10:
> 主系统缺包
？大佬这是缺那个依赖包吗？编译的是ARM 怎么跑出x86包来
云编译 在yml执行脚本添加 sudo apt-get install -y libxml2-dev libxslt-dev 也没用

bit1001, 2024-02-16:
请问解决了么？我在编译6.6内核的时候遇到了同样的报错。

bit1001, 2024-02-16:
参考这个处理下 https://github.com/coolsnowwolf/lede/issues/10113

coolsnowwolf, 2026-09-05:
该 issue 没有提供 libxslt 的实际编译错误（截图显示的是云编译过程），而交叉编译 ARM64 时同时编译 host 工具属于正常流程，因此日志中出现 x86_64 并不代表目标架构错误。当前 libxslt 包可正常使用，OpenWrt 主线后续已迁移至新版本/CMake，但不影响该旧构建咨询。请提供完整失败命令和末尾错误后再定位；本 issue 现关闭。

closed 2026-09-05 by coolsnowwolf