coolsnowwolf/lede #11877: 编译Linksys EA8500时，luci-app-advanced-reboot不会被默认选中，并且luci-app-turboacc也没有办法选中。

Hirro029, 2024-02-07:
[image]
[image]
[image]

Hirro029, 2024-02-26:
看了一下更新日志，应该是内核从5.4升级到5.10的时候注释掉了
是这几个不支持新的核心了是么？

Hirro029, 2024-02-27:
看了一下更新记录，应该是内核从5.4更新到5.10的时候把这几个软件去掉的
想问一下是新的核心不支持这几个软件么？
[image]

coolsnowwolf, 2026-09-05:
当前 LuCI 分支已包含 `luci-app-turboacc`，且其配置菜单对 ipq806x（包括 Linksys EA8500）提供 FastPath Engine 选项；仅对使用 NSS ECM 的 qualcommax/ipq50xx 等目标隐藏相应选项，ipq806x 不在隐藏列表中。该应用已由提交 `01257a1a6` 从 openwrt-23.05 恢复，原 issue 使用 2023 年版本截图，问题已由当前代码解决，无需新增修改。

closed 2026-09-05 by coolsnowwolf