coolsnowwolf/lede #11879: sdio is not enabled for brcmfmac linux module on rongpin king3399

niej, 2024-02-07:
### 详细叙述

sdio is not enabled for brcmfmac linux module on rongpin king3399 platform. Only usb and pci interface is enabled. As a result, WIFI cannot be enabled on rongpin king3399 platform.
I guess there is the identical issue on other rockchip platform. SDIO interface is more common than USB and PCIe actually.

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

Rongpin king3399

### 详细日志

grep -r usb /lib/modules/5.15.98/brcmfmac.ko 
grep: /lib/modules/5.15.98/brcmfmac.ko: binary file matches

grep -r sdio /mnt/lib/modules/5.15.98/brcmfmac.ko 

No wifi interface is found with ifconfig -a command in runtime.

niej, 2024-02-07:
If SDIO interface in WiFi driver is enabled by default, lots of rebuild can be avoided.

coolsnowwolf, 2026-09-05:
当前 `package/kernel/mac80211/broadcom.mk` 已默认对 `TARGET_rockchip` 开启 `BRCMFMAC_SDIO`（提交 `6e97cb9f9528`），并在 brcmfmac 包配置中正确生成 SDIO 支持及 `kmod-mmc` 依赖。该修复早于本 issue，当前树不会再出现仅启用 USB/PCI 导致 SDIO 缺失的情况；设备仍需对应 SDIO 固件和 DTS/电源节点。issue 关闭。

closed 2026-09-05 by coolsnowwolf