coolsnowwolf/lede #11883: AP8220 QWRT 固件 adblock 插件的黑名单功能,不能正常使用

OvidioLee, 2024-02-10:
### 详细叙述

AP8220 QWRT 固件 adblock 插件的黑名单功能,不能正常使用

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

阿里云AP8220

### 详细日志

`rosettastone.com
root@QWRT:~# clear
root@QWRT:~# cat /etc/adblock/adblock.blacklist

rosettastone.com
root@QWRT:~# 
`

`
Sat Feb 10 08:20:18 2024 user.info adblock-4.1.5-8[26177]: adblock instance started ::: action: reload, priority: 0, pid: 26177
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_jsnup::: status: running, mode: normal mode, cnt: 0, notify: 0, notify_cnt: 0, notify_pid: -
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_uci  ::: config: dhcp, change:
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_uci  ::: config: firewall, change:
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_main ::: dns: dnsmasq, fetch_util: /usr/bin/wget (built-in), backup: 0, backup_mode: 0, dns_jail: 0, force_srt: 0, force_dns: 0, mem_total: 894, mem_free: 515, max_queue: 4
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_main ::: name: Adaway, enabled: 0
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_list ::: name: Adaway, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_main ::: name: AdGuardHome, enabled: 0
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_list ::: name: AdGuardHome, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_main ::: name: NoBitcoin, enabled: 0
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_list ::: name: NoBitcoin, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_main ::: name: Blacklist, enabled: 1
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_main ::: name: Disconnect, enabled: 0
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_list ::: name: Disconnect, mode: remove, cnt: 0, in_rc: 0, out_rc: 0
**Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_main ::: name: Blacklist, url: /etc/adblock/adblock.blacklist, rc: 1, log: /etc/adblock/adblock.blacklist: Scheme missing.**
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_list ::: name: overall, mode: merge, cnt: 0, in_rc: 0, out_rc: 0
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_hash ::: hash_util: /usr/bin/sha256sum, hash: e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855, out_rc: 1
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_hash ::: hash_util: /usr/bin/sha256sum, hash: e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855, out_rc: 0
Sat Feb 10 08:20:18 2024 user.debug adblock-4.1.5-8[26177]: f_jsnup::: status: enabled, mode: normal mode, cnt: 0, notify: 0, notify_cnt: 0, notify_pid: -
Sat Feb 10 08:20:18 2024 user.info adblock-4.1.5-8[26177]: blocklist with overall 0 domains loaded successfully (Aliyun AP8220, QWRT )
`

[image]
[image]
[image]
[image]

caxsn, 2024-02-20:
hello, sorry if off topic,
I have the same device as you,
where did you get the firmware for this device?
thanks

MarxD, 2024-06-24:
> hello, sorry if off topic, I have the same device as you, where did you get the firmware for this device? thanks

This is a paid firmware which made by coolsnowwolf , Only provide in paid chat group

coolsnowwolf, 2026-09-05:
已核对。日志来自 adblock 4.1.5-8，`Blacklist` 被当作 URL 传给 wget（`Scheme missing`），并且最终 hash 是空文件。当前 adblock 包的默认配置已明确将 blacklist 指向 `/etc/adblock/adblock.blacklist`，脚本在处理 blacklist 时先按本地文件读取再执行规则解析；README 也明确说明本地黑名单格式为每行一个域名。该历史版本行为已在当前包实现中修正，无需额外补丁，关闭为已完成。

closed 2026-09-05 by coolsnowwolf