coolsnowwolf/lede #11885: 手动添加Wiregurad节点输入本地IP爆粗

JilunSama, 2024-02-11:
### 详细叙述

在luci界面手动添加Wireguard节点时如果在<本地地址>一栏输入IP，保存会发生以下错误并无法保存。
[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

[x86-64]QWRT R24.1.24 / LuCI Master (git-24.014.66729-f45ed56)

### 详细日志

/usr/lib/lua/luci/dispatcher.lua:482: Failed to execute arcombine dispatcher target for entry '/admin/services/shadowsocksr/servers/cfg094a8f'.
The called action terminated with an exception:
/usr/lib/lua/luci/cbi.lua:165: Datatype error, bad token "cidr"
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:482: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:129: in function </usr/lib/lua/luci/dispatcher.lua:128>

kxdn3, 2024-02-13:
找Wiregurad作者

JilunSama, 2024-02-14:
> 找Wiregurad作者

这个是在ssrplus里添加wireguard节点时的问题

WYC-2020, 2024-02-18:
> > 找Wiregurad作者
> 
> 这个是在ssrplus里添加wireguard节点时的问题

改成这样
[image]
 或者直接删除 这个属性  然后 你保存 试试看功能是否正常，以为 17的luci根本就没有cidr这个玩意类型，不知道如何提交的

QYKing233, 2024-02-19:
楼上说的对，限制了输入类型，或者改为string，你还可以用passwall的sing-box的wireguard。

coolsnowwolf, 2026-09-05:
该报告的错误栈实际指向 `/admin/services/shadowsocksr/servers/...` 和旧版 Lua CBI 的 `Datatype error, bad token "cidr"`，并非 WireGuard 协议页面；使用的是 2024 年旧版 QWRT/LuCI。当前 LuCI 已移除旧 shadowsocksr CBI 应用，WireGuard 页面使用现行 `cidr6` 校验，未发现该错误路径。此问题无法在当前代码中复现，建议使用当前 passwall/WireGuard 页面重新配置；issue 关闭。

closed 2026-09-05 by coolsnowwolf