coolsnowwolf/lede #11887: rust x86版本编译体积过大,github action 编译out of space,有没有解决办法?

guhill1, 2024-02-13:
### 详细叙述

大神们,rust x86版本编译体积过大,build_dir/target-x86_64_musl/host/rustc-1.75.0-src/目录 达到13gb,github action 编译显示out of space,有没有解决办法?

本地编译是没问题的.

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86

### 详细日志

compile: writing output: write $WORK/b296/_pkg_.a: no space left on device

guhill1, 2024-02-14:
> 已经有很多解决方法了 #11480 #11665

谢谢,我已经加入jlumbroso/free-disk-space@main这个脚本,使用它清掉24g,磁盘空间使用30g,剩余44g,无奈编译需要48g左右.

liwenjie119, 2024-02-14:
> > 已经有很多解决方法了 #11480 #11665
> 
> 谢谢,我已经加入jlumbroso/free-disk-space@main这个脚本,使用它清掉24g,磁盘空间使用30g,剩余44g,无奈编译需要48g左右.去年一直到上个星期还是可以的,这个星期不知道什么突然体积变大,编译不过去了,估计大概率还是rust的原因.

你还是没仔细看，还有个脚本可以吧mnt的空间整合到一起，最高能用122g

guhill1, 2024-02-14:
> > > 已经有很多解决方法了 #11480 #11665
> > 
> > 
> > 谢谢,我已经加入jlumbroso/free-disk-space@main这个脚本,使用它清掉24g,磁盘空间使用30g,剩余44g,无奈编译需要48g左右.去年一直到上个星期还是可以的,这个星期不知道什么突然体积变大,编译不过去了,估计大概率还是rust的原因.
> 
> 你还是没仔细看，还有个脚本可以吧mnt的空间整合到一起，最高能用122g

好的,我再看看,谢谢

dzlea, 2024-02-14:
> > > > 已经有很多解决方法了 #11480 #11665
> > > 
> > > 
> > > 谢谢,我已经加入jlumbroso/free-disk-space@main这个脚本,使用它清掉24g,磁盘空间使用30g,剩余44g,无奈编译需要48g左右.去年一直到上个星期还是可以的,这个星期不知道什么突然体积变大,编译不过去了,估计大概率还是rust的原因.
> > 
> > 
> > 你还是没仔细看，还有个脚本可以吧mnt的空间整合到一起，最高能用122g
> 
> 好的,我再看看,谢谢

直接去抄我仓库的代码可以释放120+g空间出来

guhill1, 2024-02-14:
> > > > > 已经有很多解决方法了 #11480 #11665
> > > > 
> > > > 
> > > > 谢谢,我已经加入jlumbroso/free-disk-space@main这个脚本,使用它清掉24g,磁盘空间使用30g,剩余44g,无奈编译需要48g左右.去年一直到上个星期还是可以的,这个星期不知道什么突然体积变大,编译不过去了,估计大概率还是rust的原因.
> > > 
> > > 
> > > 你还是没仔细看，还有个脚本可以吧mnt的空间整合到一起，最高能用122g
> > 
> > 
> > 好的,我再看看,谢谢
> 
> 直接去抄我仓库的代码可以释放120+g空间出来

谢谢大佬,新年快乐! 已经抄到了😁👍

closed 2024-02-14 by guhill1

x2358473, 2024-07-08:
> #11665

> > > > > 已经有很多解决方法了 #11480 #11665
> > > > 
> > > > 
> > > > 谢谢,我已经加入jlumbroso/free-disk-space@main这个脚本,使用它清掉24g,磁盘空间使用30g,剩余44g,无奈编译需要48g左右.去年一直到上个星期还是可以的,这个星期不知道什么突然体积变大,编译不过去了,估计大概率还是rust的原因.
> > > 
> > > 
> > > 你还是没仔细看，还有个脚本可以吧mnt的空间整合到一起，最高能用122g
> > 
> > 
> > 好的,我再看看,谢谢
> 
> 直接去抄我仓库的代码可以释放120+g空间出来

牛逼，编译ext4的固件每次都是空间不足报错，这个直接122G可用，非常感谢。