coolsnowwolf/lede #11888: nginx+uwsgi 504 Bad Gateway Time-Out

gamilwcy, 2024-02-13:
### 详细叙述

# 描述
luci-ssl-nginx + uwsgi出现504错误，加了uwsgi_read_timeout 1800; 后出现502.

# 使用场景
- docker 拉取较大镜像的镜像，时常超过1分钟的时候出现，例如：onlyOffice/documentserver: latest
- 拉取较小的镜像例如: nginx 不会出现.
- 我使用秒表计时，做了多次验证，docker 刚好拉取镜像的60s的时候，就会出现这个现象.
- 如果我切换为uhttpd则没有出现这个问题.

# nginx配置如下 
    server_names_hash_bucket_size 512;
    client_header_buffer_size 32k;
    large_client_header_buffers 4 32k;
    client_max_body_size 8g;

    gzip on;
    gzip_vary on;
    gzip_proxied any;

    fastcgi_connect_timeout 300s;
    fastcgi_send_timeout 300s;
    fastcgi_read_timeout 300s;
    fastcgi_buffer_size 256k;
    fastcgi_buffers 16 256k;
    fastcgi_busy_buffers_size 512k;
    fastcgi_temp_file_write_size 512k;
            fastcgi_intercept_errors on;

    root /www;
  
    server { #see uci show 'nginx._lan'
	  listen 443 ssl default_server;
	  listen [::]:443 ssl default_server;
	  server_name _lan;
	  include restrict_locally;
	  include conf.d/*.locations;
	  ssl_certificate /etc/nginx/conf.d/_lan.crt;
	  ssl_certificate_key /etc/nginx/conf.d/_lan.key;
	  ssl_session_cache shared:SSL:32k;
	  ssl_session_timeout 64m;
	  access_log off; # logd openwrt;
     }
  
    server { #see uci show 'nginx._redirect2ssl'
	  listen 80;
	  listen [::]:80;
	  server_name _redirect2ssl;
	  return 302 https://$host$request_uri;
     }

    location /cgi-bin/luci {
		index  index.html;
		include uwsgi_params;
		uwsgi_param SERVER_ADDR $server_addr;
		uwsgi_modifier1 9;
		uwsgi_read_timeout 1800;
		uwsgi_send_timeout 300;
		proxy_read_timeout 75s;
		proxy_send_timeout 75s;
		proxy_connect_timeout 75s;
		uwsgi_socket_keepalive on;
		uwsgi_force_ranges on;
		uwsgi_connect_timeout 300s;
		uwsgi_ignore_client_abort on;
		uwsgi_next_upstream_timeout 120s;
		uwsgi_next_upstream_tries 10;
		uwsgi_max_temp_file_size 4096m;
		uwsgi_pass unix:////var/run/luci-webui.socket;
    }
    location ~ /cgi-bin/cgi-(backup|download|upload|exec) {
		include uwsgi_params;
		uwsgi_param SERVER_ADDR $server_addr;
		uwsgi_modifier1 9;
		uwsgi_read_timeout 1800;
		uwsgi_send_timeout 300;
		proxy_read_timeout 75;
		proxy_send_timeout 75s;
		proxy_connect_timeout 75s;
		uwsgi_socket_keepalive on;
		uwsgi_force_ranges on;
		uwsgi_connect_timeout 120s;
		uwsgi_ignore_client_abort on;
		uwsgi_next_upstream_timeout 120s;
		uwsgi_next_upstream_tries 10;
		uwsgi_max_temp_file_size 4096m;
		uwsgi_pass unix:////var/run/luci-cgi_io.socket;
    }

    location /luci-static {
		error_log stderr crit;
    }

    location /ubus {
        ubus_interpreter;
        ubus_socket_path /var/run/ubus/ubus.sock;
        ubus_parallel_req 2;
   }

# uwsgi 配置如下
- /etc/uwsgi/emperor.ini
```
[uwsgi]
strict = true
pidfile = /var/run/uwsgi.pid
emperor = /etc/uwsgi/vassals/*.ini
early-emperor = true
vacuum = true
emperor-on-demand-directory = /var/run/
emperor-required-heartbeat = 99
vassal-set = die-on-idle=true

- /etc/uwsgi/vassals/luci-webui.ini
```
[uwsgi]
strict = true
pidfile = /var/run/uwsgi.pid
emperor = /etc/uwsgi/vassals/*.ini
early-emperor = true
vacuum = true
emperor-on-demand-directory = /var/run/
emperor-required-heartbeat = 99
vassal-set = die-on-idle=true
root@OpenWrt:/etc/nginx# cat /etc/uwsgi/
emperor.ini  vassals/
root@OpenWrt:/etc/nginx# cat /etc/uwsgi/vassals/luci-
luci-cgi_io.ini  luci-webui.ini
root@OpenWrt:/etc/nginx# cat /etc/uwsgi/vassals/luci-webui.ini
[uwsgi]
strict = true
if-not-env = UWSGI_EMPEROR_FD
socket = /var/run/luci-webui.socket
chmod-socket = 666
cheap = true
end-if =
plugin = cgi
cgi-mode = true
cgi = /www/
chdir = /usr/lib/lua/luci/
buffer-size = 10000
reload-mercy = 8
max-requests = 2000
limit-as = 600000
reload-on-as = 256
reload-on-rss = 192
enable-threads = true
post-buffering = 8192
socket-timeout = 300
thunder-lock = true
plugin = syslog
logger = luci syslog:uwsgi-luci
; the regular expression leaves for successful de/activation only one line each:
log-route = luci ^(?!... Starting uWSGI |compiled with version: |os: Linux|nodename: |machine: |clock source: |pcre jit |detected number of CPU cores: |current working directory: |detected binary path: |uWSGI running as root, you can use |... WARNING: you are running uWSGI as root |chdir.. to |your processes number limit is |limiting address space of processes...|your process address space limit is |your memory page size is |detected max file descriptor number: |lock engine: |thunder lock: |uwsgi socket |your server socket listen backlog is limited to |your mercy for graceful operations on workers is |mapped .* bytes |... Operational MODE: |initialized CGI path: |... no app loaded. going in full dynamic mode ...|... uWSGI is running in multiple interpreter mode ...|spawned uWSGI worker |announcing my loyalty to the Emperor...|workers have been inactive for more than |SIGINT/SIGQUIT received...killing workers...|worker .* buried |goodbye to uWSGI.|...gracefully killing workers...|Gracefully killing worker|worker .* killed successfully)
disable-logging = true
req-logger = syslog:uwsgi-luci
log-format=%(method) %(uri) => return %(status) (%(rsize) bytes in %(msecs) ms)
threads = 3
processes = 3
cheaper-algo = spare
cheaper = 1
cheaper-initial = 1
cheaper-step = 1
master = true
idle = 360

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86虚拟机

### 详细日志

# nginx日志

```
2024/02/13 14:48:14 [error] 3132#0: *22 upstream prematurely closed connection while reading response header from 
upstream, client: 192.168.1.157, server: _lan, request: "POST /cgi-bin/luci/admin/docker/images HTTP/1.1", upstream: 
"uwsgi://unix:////var/run/luci-webui.socket:", host: "192.168.1.229", referrer: "https://192.168.1.229/cgi-
bin/luci/admin/docker/images"
```

# uwsgi 日志
```
/etc/uwsgi/vassals/luci-webui.ini is ready to accept requests
Tue Feb 13 14:47:23 2024 daemon.err uwsgi[31940]: Tue Feb 13 14:47:23 2024 - [emperor] vassal /etc/uwsgi/vassals/luci-webui.ini is now loyal
Tue Feb 13 14:48:14 2024 daemon.info uwsgi-luci: invalid CGI response !!!
Tue Feb 13 14:48:14 2024 daemon.err dockerd[13056]: time="2024-02-13T06:48:14.918256230Z" level=error msg="Not continuing with pull after error: context canceled"
```

WYC-2020, 2024-02-18:
proxy_connect_timeout  这个玩意加大看下

WYC-2020, 2024-02-18:
proxy_read_timeout还有这个  在看下你的uwsgi.ini 配置 是否有harakiri  这个选项 有的话 把这个调大 或这说命令行 启动的uwsgi 参数带有 harakiri  这个 把这个值调大

coolsnowwolf, 2026-09-05:
该问题使用的是 2024 年旧版配置，日志中的 `invalid CGI response` 发生在 Docker 拉取长耗时请求期间，属于后端请求被 uWSGI/CGI worker 中止，而非 nginx 语法错误。当前 packages 的 LuCI uWSGI 配置已将 `socket-timeout` 设为 120 秒，并与 OpenWrt 主线一致；nginx 的 `uwsgi_read_timeout` 不能覆盖 uWSGI 自身超时或应用/客户端中止。若需更长任务，应通过异步任务/轮询而不是保持 LuCI CGI 请求；当前代码无需修改，issue 关闭。

closed 2026-09-05 by coolsnowwolf