coolsnowwolf/lede #11889: 编译旧分支时出错

fynhnt, 2024-02-13:
### 详细叙述

切换旧分支，以 20230609 为例 ./scripts/feeds update -a 后提示以下错误

[image]

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

N5105

### 详细日志

ERROR: please fix feeds/packages/libs/afalg_engine/Makefile - see logs/feeds/packages/libs/afalg_engine/dump.txt for details
ERROR: please fix feeds/packages/libs/gost_engine/Makefile - see logs/feeds/packages/libs/gost_engine/dump.txt for details

fynhnt, 2024-02-13:
用 WinSCP 打开两个目录，并无 dump.txt 文件
[image]

caixin1218, 2024-02-15:
多运行几次
git pull 
./scripts/feeds update -a   
./scripts/feeds install -a

WYC-2020, 2024-02-18:
这种混合杂交出问题很正常，有的提交根本就不能兼容以前的 要用以前的 建议是所有的仓库都回滚到相同或者相似时间，当然如果有能力修的就当我没说

hzjnet, 2024-02-21:
> ### 详细叙述
> 切换旧分支，以 20230609 为例 ./scripts/feeds update -a 后提示以下错误
> 
> ![快照1](https://private-user-images.githubusercontent.com/121024319/304505719-54c03d16-6644-44d6-bcfa-853241ba1ebb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1NTc2ODYsIm5iZiI6MTcwODU1NzM4NiwicGF0aCI6Ii8xMjEwMjQzMTkvMzA0NTA1NzE5LTU0YzAzZDE2LTY2NDQtNDRkNi1iY2ZhLTg1MzI0MWJhMWViYi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDIyMVQyMzE2MjZaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kMzU0MTk1N2Q5ZTc4NDJhOTBiMjlkYTBiMDdhMzc5YTZkZmRhOTliMDA3ZWQxNmM0YjMzNjk2ZTliNzM5MDI4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.DOYQi6wRnPJMTCk5ROC6uPGuX90qt6a69bISV3DwGDg)
> 
> ### 重复 issue
> * [x]  没有类似的 issue
> 
> ### 具体型号
> N5105
> 
> ### 详细日志
> ERROR: please fix feeds/packages/libs/afalg_engine/Makefile - see logs/feeds/packages/libs/afalg_engine/dump.txt for details ERROR: please fix feeds/packages/libs/gost_engine/Makefile - see logs/feeds/packages/libs/gost_engine/dump.txt for details

看最后2行明显的提示啊：
see logs/feeds/packages/libs/afalg_engine/dump.txt

coolsnowwolf, 2026-09-05:
该问题是将 2023-06-09 的旧 LEDE 分支与当时之后不断演进的 packages feed 混用导致的 feed 兼容性错误：afalg_engine/gost_engine 的 Makefile 依赖新版 OpenSSL/构建框架，旧 buildroot 的 `scripts/feeds` 校验会报 “please fix ... Makefile”。旧分支应使用其对应日期的 feeds/packages 提交，不能直接更新到当前 master；当前 master 的相关包已按现行框架维护（afalg_engine 也已由上游移除）。这不是当前代码可回移的单包修复，建议锁定匹配 feed revision 后重编译，issue 关闭。

closed 2026-09-05 by coolsnowwolf