coolsnowwolf/lede #11891: 最新源码编译错误:openwrt/include/toplevel.mk:231：world] 错误 2

jiaopengzi, 2024-02-15:
### 详细叙述

大佬您好，

使用 sh 脚本编译 参考了: #11764 但是还是报错。

```sh
#!/bin/bash

# 更新软件包 & 安装依赖
sudo apt update -y
sudo apt full-upgrade -y
sudo apt install -y ack antlr3 asciidoc autoconf automake autopoint binutils bison build-essential \
    bzip2 ccache cmake cpio curl device-tree-compiler fastjar flex gawk gettext gcc-multilib g++-multilib \
    git gperf haveged help2man intltool libc6-dev-i386 libelf-dev libfuse-dev libglib2.0-dev libgmp3-dev \
    libltdl-dev libmpc-dev libmpfr-dev libncurses5-dev libncursesw5-dev libpython3-dev libreadline-dev \
    libssl-dev libtool lrzsz mkisofs msmtp ninja-build p7zip p7zip-full patch pkgconf python2.7 python3 \
    python3-pyelftools python3-setuptools qemu-utils rsync scons squashfs-tools subversion swig texinfo \
    uglifyjs upx-ucl unzip vim wget xmlto xxd zlib1g-dev llvm clang
sudo apt-get -y autoremove --purge
sudo apt-get clean

# 下载源码
git clone --depth 1 https://github.com/coolsnowwolf/lede -b master openwrt
cd openwrt
# 打印当前目录
pwd

# 更新软件包 & 安装依赖
./scripts/feeds update -a
./scripts/feeds install -a

# 向 .config 文件追加配置覆盖写入 自定义定制 - x86_64.config
# 删除原来的 .config 文件
sudo rm -rf .config
sudo touch .config
cat >>.config <<EOF
# x86_64.config
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_Generic=y
CONFIG_PACKAGE_grub2-efi=y
CONFIG_EFI_IMAGES=y
CONFIG_TARGET_IMAGES_GZIP=y
CONFIG_TARGET_KERNEL_PARTSIZE=16
CONFIG_TARGET_ROOTFS_PARTSIZE=300
# CONFIG_GRUB_CONSOLE is not set
CONFIG_PACKAGE_luci-app-docker=y
CONFIG_PACKAGE_luci-i18n-docker-zh-cn=y
# CONFIG_PACKAGE_luci-app-qbittorrent is not set
CONFIG_PACKAGE_luci-app-ssr-plus=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_Rust=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_V2ray_plugin=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Xray=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Trojan=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Redsocks2=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_NaiveProxy=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Kcptun=y
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=y
CONFIG_PACKAGE_luci-i18n-ssr-plus-zh-cn=y
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_luci-app-mwan3=y
CONFIG_PACKAGE_luci-app-mwan3helper=y
CONFIG_PACKAGE_luci-i18n-mwan3-zh-cn=y
CONFIG_PACKAGE_luci-i18n-mwan3helper-zh-cn=y
CONFIG_PACKAGE_luci-app-syncdial=y
CONFIG_PACKAGE_luci-app-ttyd=y
CONFIG_PACKAGE_luci-i18n-ttyd-zh-cn=y
CONFIG_PACKAGE_luci-app-webadmin=y
[... 249 lines cut ...]
Making install in tests
make[6]: Entering directory '/home/jpz/build-openwrt/openwrt/build_dir/host/patch-2.7.6/tests'
make[7]: Entering directory '/home/jpz/build-openwrt/openwrt/build_dir/host/patch-2.7.6/tests'
make[7]: Nothing to be done for 'install-exec-am'.
make[7]: Nothing to be done for 'install-data-am'.
make[7]: Leaving directory '/home/jpz/build-openwrt/openwrt/build_dir/host/patch-2.7.6/tests'
make[6]: Leaving directory '/home/jpz/build-openwrt/openwrt/build_dir/host/patch-2.7.6/tests'
make[6]: Entering directory '/home/jpz/build-openwrt/openwrt/build_dir/host/patch-2.7.6'
make[7]: Entering directory '/home/jpz/build-openwrt/openwrt/build_dir/host/patch-2.7.6'
make[7]: Nothing to be done for 'install-exec-am'.
 /usr/bin/mkdir -p '/home/jpz/build-openwrt/openwrt/staging_dir/host/share/man/man1'
 /home/jpz/build-openwrt/openwrt/staging_dir/host/bin/install -c -m 644 'patch.man' '/home/jpz/build-openwrt/openwrt/staging_dir/host/share/man/man1/patch.1'
make[7]: Leaving directory '/home/jpz/build-openwrt/openwrt/build_dir/host/patch-2.7.6'
make[6]: Leaving directory '/home/jpz/build-openwrt/openwrt/build_dir/host/patch-2.7.6'
make[5]: Leaving directory '/home/jpz/build-openwrt/openwrt/build_dir/host/patch-2.7.6'
make[4]: Leaving directory '/home/jpz/build-openwrt/openwrt/build_dir/host/patch-2.7.6'
mkdir -p /home/jpz/build-openwrt/openwrt/staging_dir/host/stamp
touch /home/jpz/build-openwrt/openwrt/build_dir/host/patch-2.7.6/.built
touch /home/jpz/build-openwrt/openwrt/staging_dir/host/stamp/.patch_installed
make[3]: Leaving directory '/home/jpz/build-openwrt/openwrt/tools/patch'
time: tools/patch/compile#23.71#10.03#24.78
make[2]: Leaving directory '/home/jpz/build-openwrt/openwrt'
make[1]: *** [tools/Makefile:217: /home/jpz/build-openwrt/openwrt/staging_dir/host/stamp/.tools_compile_nyyynyynnyyynynyyyyyyynyynnynyynnynnyyyyyyyyyyyyyyyyynynnyyyyyyyy] Error 2
make[1]: Leaving directory '/home/jpz/build-openwrt/openwrt'
make: *** [/home/jpz/build-openwrt/openwrt/include/toplevel.mk:231：world] 错误 2

closed 2024-02-16 by jiaopengzi

12123ads, 2024-07-20:
@jiaopengzi 大佬你好，请问这个问题您解决了吗

jiaopengzi, 2024-07-21:
> @jiaopengzi 大佬你好，请问这个问题您解决了吗

参考：https://github.com/jiaopengzi/build-openwrt

jiaopengzi, 2024-09-17:
> 能说明下详细的解决方案嘛，我也遇到了这个问题
> 

多半是没有全程梯子，可以看下我楼上我写的的脚本。