coolsnowwolf/lede #11896: 自编译的openwrt 每天晚上 4-5点都会自动重启.

jt8823, 2024-02-18:
### 详细叙述

主机型号 | QEMU Standard PC (i440FX + PIIX - Intel(R) Celeron(R) J4025 CPU @ 2.00GHz : 2C2T (CpuMark : 19723.957057 Scores)
CPU 信息 | 1996.802 MHz
固件版本 | OpenWrt R24.2.2 / LuCI Master (git-24.019.37088-cdf2b69)
内核版本 | 6.1.77

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

J4025

### 详细日志

`[    0.000000] Linux version 6.1.77 (jt8823@ZHOU-PC) (x86_64-openwrt-linux-musl-gcc (OpenWrt GCC 11.3.0 r6422-801b8fdb8) 11.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP PREEMPT_DYNAMIC Mon Feb 12 14:54:20 2024
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=ca45ebf8-8e3a-0fb9-b5da-abb352a4c002 rootwait console=tty0 console=ttyS0,115200n8 noinitrd swiotlb=65536
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009fbff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009fc00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000f0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007ffdffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007ffe0000-0x000000007fffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000feffc000-0x00000000feffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffc0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: QEMU Standard PC (i440FX + PIIX, 1996), BIOS Virtualization-2.5.1-11303-200428-0-gfff70e7 04/01/2014
[    0.000000] Hypervisor detected: KVM
[    0.000000] kvm-clock: Using msrs 4b564d01 and 4b564d00
[    0.000001] kvm-clock: using sched offset of 726284702039861 cycles
[    0.000004] clocksource: kvm-clock: mask: 0xffffffffffffffff max_cycles: 0x1cd42e4dffb, max_idle_ns: 881590591483 ns
[    0.000008] tsc: Detected 1996.802 MHz processor
[    0.000879] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000884] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000889] last_pfn = 0x7ffe0 max_arch_pfn = 0x400000000
[    0.000926] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WP  UC- WT  
[    0.008705] found SMP MP-table at [mem 0x000f5b80-0x000f5b8f]
[    0.008744] Using GB pages for direct mapping
[    0.008986] ACPI: Early table checksum verification disabled
[    0.008993] ACPI: RSDP 0x00000000000F5990 000014 (v00 BOCHS )
[    0.009012] ACPI: RSDT 0x000000007FFE159A 000030 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.009020] ACPI: FACP 0x000000007FFE147E 000074 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.009028] ACPI: DSDT 0x000000007FFE0040 00143E (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.009032] ACPI: FACS 0x000000007FFE0000 000040
[    0.009036] ACPI: APIC 0x000000007FFE14F2 000080 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.009040] ACPI: WAET 0x000000007FFE1572 000028 (v01 BOCHS  BXPC     00000001 BXPC 00000001)
[    0.009044] ACPI: Reserving FACP table memory at [mem 0x7ffe147e-0x7ffe14f1]
[    0.009046] ACPI: Reserving DSDT table memory at [mem 0x7ffe0040-0x7ffe147d]
[    0.009047] ACPI: Reserving FACS table memory at [mem 0x7ffe0000-0x7ffe003f]
[    0.009048] ACPI: Reserving APIC table memory at [mem 0x7ffe14f2-0x7ffe1571]
[    0.009049] ACPI: Reserving WAET table memory at [mem 0x7ffe1572-0x7ffe1599]
[    0.009140] Zone ranges:
[    0.009142]   DMA      [mem 0x0000000000001000-0x0000000000ffffff]
[    0.009144]   DMA32    [mem 0x0000000001000000-0x000000007ffdffff]
[    0.009146]   Normal   empty
[    0.009147] Movable zone start for each node
[    0.009148] Early memory node ranges
[    0.009149]   node   0: [mem 0x0000000000001000-0x000000000009efff]
[    0.009150]   node   0: [mem 0x0000000000100000-0x000000007ffdffff]
[    0.009152] Initmem setup node 0 [mem 0x0000000000001000-0x000000007ffdffff]
[    0.009166] On node 0, zone DMA: 1 pages in unavailable ranges
[    0.009303] On node 0, zone DMA: 97 pages in unavailable ranges
[    0.025899] On node 0, zone DMA32: 32 pages in unavailable ranges
[    0.026417] ACPI: PM-Timer IO Port: 0x608
[    0.026437] ACPI: LAPIC_NMI (acpi_id[0xff] dfl dfl lint[0x1])
[    0.026478] IOAPIC[0]: apic_id 0, version 17, address 0xfec00000, GSI 0-23
[    0.026483] ACPI: INT_SRC_OVR (bus 0 bus_irq 0 global_irq 2 dfl dfl)
[... 412 lines cut ...]
[   16.032485] usbcore: registered new interface driver ums-alauda
[   16.034889] usbcore: registered new interface driver ums-cypress
[   16.039255] usbcore: registered new interface driver ums-datafab
[   16.041532] usbcore: registered new interface driver ums-freecom
[   16.044160] usbcore: registered new interface driver ums-isd200
[   16.048746] usbcore: registered new interface driver ums-jumpshot
[   16.051502] usbcore: registered new interface driver ums-karma
[   16.053765] usbcore: registered new interface driver ums-sddr09
[   16.056037] usbcore: registered new interface driver ums-sddr55
[   16.058201] usbcore: registered new interface driver ums-usbat
[   16.079812] xt_time: kernel timezone is -0000
[   16.126632] PPP generic driver version 2.4.2
[   16.130327] PPP MPPE Compression module registered
[   16.133372] NET: Registered PF_PPPOX protocol family
[   16.409211] [drm] amdgpu kernel modesetting enabled.
[   16.451237] kmodloader: done loading kernel modules from /etc/modules.d/*
[   20.414104] 8021q: adding VLAN 0 to HW filter on device eth1
[   20.418150] br-lan: port 1(eth1) entered blocking state
[   20.418865] br-lan: port 1(eth1) entered disabled state
[   20.419722] device eth1 entered promiscuous mode
[   20.425277] br-lan: port 1(eth1) entered blocking state
[   20.427004] br-lan: port 1(eth1) entered forwarding state
[   20.432343] 8021q: adding VLAN 0 to HW filter on device eth0
[   25.692559] pppoe-wan: renamed from ppp0
`

WYC-2020, 2024-02-18:
固定时间的话 看看这个时间段有没啥定时任务呗，我编译5.4内核 都稳定一年了哈哈

jt8823, 2024-02-18:
> 固定时间的话 看看这个时间段有没啥定时任务呗，我编译5.4内核 都稳定一年了哈哈

我在GPT里面问了一下,    定时任务是正常的一些任务呀.   而且   也没有这个时间段的任务!

WYC-2020, 2024-02-18:
> > 固定时间的话 看看这个时间段有没啥定时任务呗，我编译5.4内核 都稳定一年了哈哈
> 
> 我在GPT里面问了一下, 定时任务是正常的一些任务呀. 而且 也没有这个时间段的任务!

问啥gpt，那个gpt弱智的很， 你都是固定时间 那就吧 所有定时任务先清空 把没必要的软件关闭 在观察不就知道了

jt8823, 2024-02-18:
> > > 固定时间的话 看看这个时间段有没啥定时任务呗，我编译5.4内核 都稳定一年了哈哈
> > 
> > 
> > 我在GPT里面问了一下, 定时任务是正常的一些任务呀. 而且 也没有这个时间段的任务!
> 
> 问啥gpt，那个gpt弱智的很， 你都是固定时间 那就吧 所有定时任务先清空 把没必要的软件关闭 在观察不就知道了

我试试看，我一共就装了3个插件，。 而且和上一版本编译所使用的插件一摸一样，上一版本挺好的。

fxxccp, 2024-02-24:
固定时间死机的话，删除/etc/crontab/root里面的cpumark.sh即可解决，或者在编译固件时不勾选coremark。

jt8823, 2024-02-24:
> 固定时间死机的话，删除/etc/crontab/root里面的cpumark.sh即可解决，或者在编译固件时不勾选coremark。

这个不是首次安装后,就运行一次的跑分吗?  后面他就不会自动运行了吧

我是看运行时间,大概就是晚上4 5点左右会自动重启.

jt8823, 2024-02-24:
> 固定时间死机的话，删除/etc/crontab/root里面的cpumark.sh即可解决，或者在编译固件时不勾选coremark。

一开始我也怀疑是cpumark的问题,然后我手动运行过一次这个,也没重启或者死机呀

coolsnowwolf, 2026-09-05:
该报告只有一次 2024 年旧版 QEMU/KVM 启动日志，没有“4–5 点自动重启”发生时刻的前后日志、watchdog/panic/oops、宿主机任务或 cron 信息。现有内容显示的是正常启动过程，无法将定时重启归因于 LEDE 内核或软件；QEMU RTC/宿主机快照、计划任务和电源管理均可能造成此现象。请在当前版本提供重启前后串口日志、`logread -f`/pstore 及宿主机 QEMU 日志后再分析。

closed 2026-09-05 by coolsnowwolf