coolsnowwolf/lede #11897: 内核6.1.75后云编译皆报错

tonylee2022, 2024-02-18:
### 详细叙述

卡在Configuring luci-proto-ncm.，修改config文件也无效，内核6.1.75编译正常。

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

x86_64

### 详细日志

Configuring luci-i18n-nlbwmon-zh-cn.
Configuring luci-proto-ncm.
make[2]: Leaving directory '/home/runner/work/HB_OpenWrt/HB_OpenWrt/openwrt'
make[1]: *** [package/Makefile:111: /home/runner/work/HB_OpenWrt/HB_OpenWrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/runner/work/HB_OpenWrt/HB_OpenWrt/openwrt'
make: *** [/home/runner/work/HB_OpenWrt/HB_OpenWrt/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

WYC-2020, 2024-02-18:
算不出

tonylee2022, 2024-02-18:
编译git checkout 299db83defeaf405069096ec65137fadc955aa27  #11853开始出错。#11830正常
Configuring luci-i18n-nlbwmon-zh-cn.
Configuring luci-proto-ncm.
 * check_data_file_clashes: Package iperf3-ssl wants to install file /home/runner/work/HB_OpenWrt/HB_OpenWrt/openwrt/build_dir/target-x86_64_musl/root-x86/usr/bin/iperf3
	But that file is already provided by package  * iperf3
 * opkg_install_cmd: Cannot install package iperf3-ssl.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/runner/work/HB_OpenWrt/HB_OpenWrt/openwrt'
make[1]: *** [package/Makefile:111: /home/runner/work/HB_OpenWrt/HB_OpenWrt/openwrt/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/runner/work/HB_OpenWrt/HB_OpenWrt/openwrt'
make: *** [/home/runner/work/HB_OpenWrt/HB_OpenWrt/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

tonylee2022, 2024-02-18:
这个提交增加了iperf3-ssl 
[target/linux/x86/Makefile](https://github.com/coolsnowwolf/lede/commit/8a89f464cbceeb1c87292cdd3f3381da2c01608d#diff-63cfaeebbcf45a5e79bffbb47a27a0e26ef33872b7fe44913dc89a1f9137b086)
@@ -18,11 +18,10 @@ KERNELNAME:=bzImage
include $(INCLUDE_DIR)/target.mk

DEFAULT_PACKAGES += partx-utils mkf2fs e2fsprogs kmod-button-hotplug kmod-usb-hid kmod-mmc kmod-sdhci kmod-fs-f2fs cfdisk usbutils pciutils \
kmod-alx kmod-e1000e kmod-igb kmod-igc kmod-igbvf kmod-iavf kmod-bnx2x kmod-pcnet32 kmod-tulip kmod-via-velocity kmod-vmxnet3 kmod-i40e kmod-i40evf kmod-r8125 kmod-8139cp kmod-8139too kmod-tg3 \
htop lm-sensors iperf3 autocore-x86 automount autosamba luci-app-adbyby-plus luci-app-ipsec-vpnd luci-proto-bonding luci-app-diskman \
luci-app-unblockmusic luci-app-zerotier luci-app-xlnetacc ddns-scripts_aliyun ddns-scripts_dnspod ca-bundle luci-app-wireguard luci-app-ttyd \
kmod-sound-hda-core kmod-sound-hda-codec-realtek kmod-sound-hda-codec-via kmod-sound-via82xx kmod-sound-hda-intel kmod-sound-hda-codec-hdmi kmod-sound-i8x0 kmod-usb-audio \
kmod-usb-net kmod-usb-net-asix-ax88179 kmod-usb-net-rtl8150 kmod-usb-net-rtl8152-vendor kmod-usb-net-aqc111 kmod-mlx4-core kmod-mlx5-core kmod-drm-i915 kmod-drm-amdgpu
kmod-alx kmod-e1000e kmod-igb kmod-igc kmod-igbvf kmod-iavf kmod-bnx2x kmod-pcnet32 kmod-tulip kmod-via-velocity kmod-vmxnet3 kmod-i40e kmod-i40evf \
kmod-sound-hda-core kmod-sound-hda-codec-realtek kmod-sound-hda-codec-via kmod-sound-via82xx kmod-sound-hda-intel kmod-sound-hda-codec-hdmi kmod-sound-i8x0 \
kmod-usb-net kmod-usb-net-asix-ax88179 kmod-usb-net-rtl8150 kmod-usb-net-aqc111 kmod-mlx4-core kmod-mlx5-core kmod-drm-i915 kmod-drm-amdgpu kmod-usb-audio \
kmod-usb-net-rtl8152-vendor kmod-r8125 kmod-r8168 kmod-8139cp kmod-8139too kmod-tg3 htop lm-sensors iperf3-ssl autocore-x86 automount autosamba ca-bundle

$(eval $(call BuildTarget))

WYC-2020, 2024-02-19:
这个问题我已经说过了我已经在某个issue说过 自己翻阅一下

WYC-2020, 2024-02-19:
https://github.com/coolsnowwolf/lede/issues/11838

closed 2024-11-04 by aiamadeus