coolsnowwolf/lede #11899: 如何关闭端口回流功能

ishunsei, 2024-02-19:
### 详细叙述

我需要使用公网ip访问局域网设备,并且需要走公网,所以希望关闭Nat回环;
现在的症状是如果我关闭"**启用 NAT 环回**"选项,局域网设备就无法使用公网ip访问局域网的服务器;
要做到使用公网ip访问,并且需要走公网,要怎么实现呢?
网上查到这是端口回流问题,但大多都是启用环回来解决或是强制映射域名到局域网ip,有没有让他正常走外网的方法呢?

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

J4215

### 详细日志

无

JayXus, 2024-02-20:
关闭NAT回流后：内网访问 公网IP:xxx **不通**，公网访问公网IP:xxx **通**
开启NAT回流后：内外网访问 公网IP:xxx **都能通**
https://www.right.com.cn/forum/forum.php?mod=redirect&goto=findpost&ptid=8203412&pid=16664932

ishunsei, 2024-02-21:
> 关闭NAT回流后：内网访问 公网IP：xxx **不**通，公网访问公网IP：xxx 通 开启NAT回流后：内外网访问 公网IP：xxx **都能通** https://www.right.com.cn/forum/forum.php?mod=redirect&goto=findpost&ptid=8203412&pid=16664932

我查看了大量的资料,确定了这是端口回流问题,目前我想知道是否能实现让他走外网

coolsnowwolf, 2026-09-05:
“NAT 环回/端口回流”关闭后，内网客户端访问本机公网地址不会再被 firewall 重写到内网服务器，这是该选项的预期语义，不能同时要求关闭 hairpin NAT 又保持同一公网地址的内网访问。若必须使用公网域名，可使用 split DNS（内网解析到内网地址、外网解析到公网地址），或让客户端经独立外部网络/VPN 出口真正绕行；当前 LuCI/firewall4 选项行为正常，无需代码修复。

closed 2026-09-05 by coolsnowwolf