coolsnowwolf/lede #11905: ZN-M2 刷入后无法启动

liyiranlab, 2024-02-20:
### 详细叙述

1c57996
今天编译的

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

ZN-M2

### 详细日志

无法启动没有日志

Glunefish, 2024-02-20:
你是怎么解决acl编译的？我编译老是acl编译出错

liyiranlab, 2024-02-21:
重新拉源码

liyiranlab, 2024-02-21:
哭，一觉起来zn-m2 设备已经删了。

WYC-2020, 2024-02-21:
哭晕在厕所哈哈

Glunefish, 2024-02-21:
啥情况？

liyiranlab, 2024-03-01:
lean大佬你好，[CMIOT AX18]编译成功了，没有M2选项，刷机后没有无线，哭。

Drliao2023, 2024-03-01:
AX5 JDC生成固件后，factory和sysupgrade的都刷不进去，uboot用的是面包多上买的

FreshKai, 2024-03-12:
> AX5 JDC生成固件后，factory和sysupgrade的都刷不进去，uboot用的是面包多上买的

#11971 红米AX5 JDC编译成功后生成的固件无法刷入机器

开发者称

> IPQ6 的开源并没有到可以日用的地步

所以需要等后续更新

Drliao2023, 2024-03-12:
收到，谢谢~

coolsnowwolf, 2026-09-05:
当前 LEDE 已包含 `zn_m2` 目标及 `ipq6018-m2.dts`。你提供的串口日志实际已经进入 Linux，随后停在 `Waiting for root device /dev/ubiblock0_1`，这不是内核未启动，而是刷入的镜像/UBI 分区布局与设备启动参数不匹配（或刷错 factory/sysupgrade 方式）。请使用目标专用 `sysupgrade`/factory 镜像并保留完整分区布局，必要时附 NAND 分区和 U-Boot 环境核对。当前无可由 package 修复的代码问题，该旧 issue 现关闭。

closed 2026-09-05 by coolsnowwolf