coolsnowwolf/lede #11907: 请问一下,编译的时候,如何指定openwrt版本,和内核版本呀!!!

jt8823, 2024-02-20:
### 详细叙述

请问一下,编译的时候,如何指定openwrt版本,和内核版本呀!!!

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

X86

### 详细日志

请问一下,编译的时候,如何指定openwrt版本,和内核版本呀!!!

breeze303, 2024-02-21:
> 相同需求。需要5.10内核的版本，不然MTK闭源驱动不兼容

[image]
这里

pangfeng161, 2024-02-21:
同问云编译的话，比如编译5.15版本如何操作，那位大神指点下

WYC-2020, 2024-02-21:
另外补充一下 如果linux 内核原生就支持的平台 比如x86  这个只要你不是用了什么特定的需要给内核打补丁的 插件或者什么模块之类的，基本上 x86是最好适配的，只需要改下 然后加上内核的config 很大程度都是拷贝上一个版本的 config 然后进行编译调试  有八成能过

hpx502766238, 2024-02-21:
> ![image](https://private-user-images.githubusercontent.com/122870904/306464711-94717d19-f21e-424e-878a-68a2f1b67a38.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]
我改
include/kernel-version.mk，在文件开头指定 Kernelpatchver:5.10 是不是更好？和改 makefile 有什么区别？

hpx502766238, 2024-02-21:
> 先看 ![image](https://private-user-images.githubusercontent.com/34025388/306476110-c716f788-d57c-487f-857e-6c8f84675527.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg1MDEyMTUsIm5iZiI6MTcwODUwMDkxNSwicGF0aCI6Ii8zNDAyNTM4OC8zMDY0NzYxMTAtYzcxNmY3ODgtZDU3Yy00ODdmLTg1N2UtNmM4Zjg0Njc1NTI3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIxVDA3MzUxNVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPThkNjdjNjY2YTg5NjU0MjBlYTNjOWMxOTU2MzEyNmU2YTcxNzkyZWZiMDI1YTcyNTM5NzM3ZTM1Y2YzNTlmMjEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.fZXS0WkhFZ3w6SyXbqIFY-JAgx8YTw9OzLNo-U_qU_4) 这里是否有你的 内核 如果有 那就简单 自己 手工make menuconfig 然后选好 自己内核 然后吧.config 保存 然后 action编译的时候 把这个 config 拷贝回去就行了 然后编译就完事 如果没有的话 自己改这里内核，同时自己确定 你这个平台是否适配了这个内核，有时候l会把老内核适配过的所有patch删除的所以有的机型只是单单改这里是不行的，如果压根就没适配过这个内核建议你放弃，因为你能这样问我就有理由确定你没这个能力进行新内核的适配工作哈哈

还有就是，是否适配某个内核，是不是得看有没有对应的 patch-5.10 hack-5.10 config-5.10?如果没有就从上一个 git 版本拷贝过来来适配，再编译，不知道这么理解有没有问题

WYC-2020, 2024-02-21:
> 还有就是，是否适配某个内核，是不是得看有没有对应的 patch-5.10 hack-5.10 config-5.10?如果没有就从上一个 git 版本拷贝过来来适配，再编译，不知道这么理解有没有问题

没有 肯定 的自己适配 某些patch了 比如你要适配5.15  这个时候 是不可以直接拷贝5.10的来的 有的补丁会失败，内核原生支持的平台 很大部分 都不需要打补丁的 这个补丁都是要么为了适配某些驱动而特意打的 可有可无 如果不是原生的 那就需要按需 看那些补丁需要 反正这个适配过程 比较麻烦 自己看 l大屌 6.6的适配过程你就知道，基本新内核适配都是以原生支持 开始的  因为这样改动最小也最简单

hpx502766238, 2024-02-21:
> 没有 肯定 的自己适配 某些patch了 比如你要适配5.15 这个时候 是不可以直接拷贝5.10的来的 有的补丁会失败，内核原生支持的平台 很大部分 都不需要打补丁的 这个补丁都是要么为了适配某些驱动而特意打的 可有可无 如果不是原生的 那就需要按需 看那些补丁需要 反正这个适配过程 比较麻烦 自己看 l大屌 6.6的适配过程你就知道，基本新内核适配都是以原生支持 开始的 [...]

那像我这种不会做适配的，最好的办法就是找到仓库用老内核的 commit 直接用对吧

ywt114, 2024-02-21:
> 同问云编译的话，比如编译5.15版本如何操作，那位大神指点下

sed -i "s/KERNEL_PATCHVER:=*.*/KERNEL_PATCHVER:=5.15/g" lede/target/linux/x86/Makefile
sed -i "s/KERNEL_TESTING_PATCHVER:=*.*/KERNEL_TESTING_PATCHVER:=5.15/g" lede/target/linux/x86/Makefile

ywt114, 2024-02-22:
[image]

hztbhtbh, 2024-02-25:
下面网站是我小白刚刚出的一期视频指定编译内核版本的openwrt，第一次录制，大家可以看看，顺便给我点个订阅点个赞，谢谢
[https://youtu.be/aVZFQYj4rrA](url)

RainyLave, 2024-06-08:
by using this, the IPKs works well on my device:
```make V=s ac_year2038_required=no LINUX_VERMAGIC:=abe015ee882e4d05089595c35422df69 ./package/kernel/linux/{clean, compile}```

coolsnowwolf, 2026-09-05:
当前源码中的版本选择路径已经明确：每个 target 的 target/linux/<target>/Makefile 设置默认 KERNEL_PATCHVER；x86 当前默认是 6.12。可在 make menuconfig 选择 target 后修改对应 target Makefile，或在云编译中使用保存的 .config/自定义补丁；但目标内核必须同时存在 include/kernel-<版本>、target/linux/generic/config-<版本> 及该 target 对应的 config、hack/patch/files 目录。仅把 KERNEL_PATCHVER 改成 5.10/5.15 并不会自动完成驱动和补丁适配，未适配版本需要逐项移植并修复。该 issue 属于使用说明，现关闭。

closed 2026-09-05 by coolsnowwolf