coolsnowwolf/lede #11908: 20220716分支编译失败，batman-adv、upx[host]报错

hpx502766238, 2024-02-21:
### 详细叙述

需要使用老5.10内核，故克隆了20220716分支，但编译失败
batman-adv编译失败
upx[host]编译失败

### 重复 issue

- [X] 没有类似的 issue

### 具体型号

XiaomiRedmiAX6S

### 详细日志

make[4]: Entering directory '/home/mrhpx/openwrt-lean/lede20220716/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.127'
  CC [M]  /home/mrhpx/openwrt-lean/lede20220716/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/batman-adv-2023.3/net/batman-adv/multicast.o
/home/mrhpx/openwrt-lean/lede20220716/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/batman-adv-2023.3/net/batman-adv/multicast.c: In function 'batadv_mcast_mla_rtr_flags_bridge_get':
/home/mrhpx/openwrt-lean/lede20220716/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/batman-adv-2023.3/net/batman-adv/multicast.c:211:7: error: implicit declaration of function 'br_multicast_has_router_adjacent'; did you mean 'br_multicast_has_querier_adjacent'? [-Werror=implicit-function-declaration]
  if (!br_multicast_has_router_adjacent(dev, ETH_P_IP))
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       br_multicast_has_querier_adjacent
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:280: /home/mrhpx/openwrt-lean/lede20220716/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/batman-adv-2023.3/net/batman-adv/multicast.o] Error 1
make[4]: *** [Makefile:1822: /home/mrhpx/openwrt-lean/lede20220716/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/batman-adv-2023.3/net/batman-adv] Error 2
make[4]: Leaving directory '/home/mrhpx/openwrt-lean/lede20220716/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/linux-5.10.127'
make[3]: *** [Makefile:104: /home/mrhpx/openwrt-lean/lede20220716/build_dir/target-aarch64_cortex-a53_musl/linux-mediatek_mt7622/batman-adv-2023.3/.built] Error 2
make[3]: Leaving directory '/home/mrhpx/openwrt-lean/lede20220716/feeds/routing/batman-adv'
time: package/feeds/routing/batman-adv/compile#0.81#0.14#1.06
    ERROR: package/feeds/routing/batman-adv failed to build.
make[2]: *** [package/Makefile:116: package/feeds/routing/batman-adv/compile] Error 1
make[2]: Leaving directory '/home/mrhpx/openwrt-lean/lede20220716'
make[1]: *** [package/Makefile:110: /home/mrhpx/openwrt-lean/lede20220716/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/mrhpx/openwrt-lean/lede20220716'
make: *** [/home/mrhpx/openwrt-lean/lede20220716/include/toplevel.mk:230: world] Error 2

zow2023, 2024-07-04:
> 编译成功了，找到了暂时解决方法： 1.batman-adv，将feed routing改为[commit 5437d2c91fd9f15e06fbea46677abb529ed3547c](https://github.com/coolsnowwolf/routing/tree/5437d2c91fd9f15e06fbea46677abb529ed3547c)，不再报错 2.package/lean/upx，从[20230609 commit](https://github.com/coolsnowwolf/lede/tree/20230609/package/lean/upx)中找到upx替换，不再报错

你好  我用你的仓库编译LINKSYS E8450的固件 改了：src-git routing https://github.com/coolsnowwolf/routing.git^5437d2c91fd9f15e06fbea46677abb529ed3547c
还是一样的报错哦，。

coolsnowwolf, 2026-09-05:
已核对。该报告针对 20220716 分支的历史组合（batman-adv 2023.3 与旧内核 5.10.127），评论已给出可行解决方案：routing feed 固定到 5437d2c91fd9f15e06fbea46677abb529ed3547c，并将 upx 替换为 20230609 版本。当前主线 routing feed 已继续更新，LEDE 内置 upx 为 4.2.4；该历史分支构建问题已有明确解决路径，不再对当前主线构成缺陷，关闭为已完成。

closed 2026-09-05 by coolsnowwolf